grobasoz/zigbee-firmware #1: EZSP NCP firmware request

MattWestb, 2020-07-10:
I trying getting EZSP NCP up and running on Ikeas ICC-A-1 module (You finding them in most Ikea Tradfri products).

mtx512 have done one first set of firmware with standalone bootloader (UART XMODEM Bootloader) and standalone bootloader with external storage and NCP.

The both bootloaders are working also applying them on Ikeas original firmware.
But the NCP its crashing hard (cant access SWD without booting in bootloader).

Hi have done one test app that working as expected and running  on the MCU.

We are stuck between hardware and software parts that we don't have not enough experience off.
Likely its a combination of the external flash and the 2 serial busses used for UART and flash (USARTx) that making it crashing.

Can you trying compiling one NCP and also if possible bootloader with one new hardware config ??

MCU: EFR32MG1P132F256GM32
PB15: RX
PB14: TX
PA0: Force bootloader boot

NCP UART SW (Software Flow Control)

Here its the main steps for doing it: https://github.com/Koenkk/zigbee-herdsman/issues/168#issuecomment-651288482

And my ICC-A-1 project git: https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module

Thanks in advance !!

Mattias W

grobasoz, 2020-07-10:
I have added [firmware ](https://github.com/grobasoz/zigbee-firmware/blob/master/NCP_USW_MG1P132F256-115k2-V676.ebl) for NCP requested above. Unfortunately I can't test as I don't have the required hardware.

MattWestb, 2020-07-10:
Thank you very much!
I trying it out and coming back with feedback 2 you :-)

MattWestb, 2020-07-10:
I have making a full erase of internal flash and flashing bootloader but the s37 file only have the second one so i was flashing one from mx512 first that having both but the wrong RX and TX pins and its working OK.
Was trying installing the NCP gbl file from bootloader but its not accepted from the bootloader.
Flashing the NCP s37 file with Blackmagick  probe (GDB) was working OK. 
Rebooting the module and getting this from Minicom -H in Lubuntu:
1a c2 02 8b c2 8a 7e  
Its looks like an NCP its trying talking to the outside the module !!!
Suld trying out with Z3GatewayHost.

Is it OK putting the files on github and referring to you ?

Then 6.7.6.0 its only basic support implanted but its moving fast forward and very soon being full featured very soon. But i think it can being very handy with one 6.6.4.0 version that its backwards compatible with zigpy / bellows - HA and OpenHab.

Can you also making one 6.6.4.0 version for the HA community ?

Thanks in advance and best regards.

Mattias W

grobasoz, 2020-07-11:
[Bootloader](https://github.com/grobasoz/zigbee-firmware/blob/master/BTL_STD_S1_256-COM_PB14-PB15-PA0.s37) added as per above specification.

MattWestb, 2020-07-11:
Thank you very much.
I trying it out and reporting back.

Is it also possible getting one NCP in 6.6.4.0 version ??

grobasoz, 2020-07-11:
I have added [NCP-V664](https://github.com/grobasoz/zigbee-firmware/blob/master/NCP_USW_115K2_S1_F256_664_PB14-PB15.s37) file to repository.
I have also changed the NCP with version 676 stack and removed the old file.

MattWestb, 2020-07-11:
Thanks one more time more the much !!
For the moment its only one big problem is testing that all working together on the "Billy EZSP".
Looks that the southern winter its making all very creative also in the early evening ;-)
I keeping you informed if i getting problem but i think i have more then i need for the moment.

Best regards from 35°C Vienna.

MattWestb, 2020-07-11:
By the way.
The first NCP gecko bootloader file "NCP_USW_MG1P132F256-115k2-V676.ebl" was having wrong file extension (ebl) sud have being gbl.
Have trying renaming it but the bootloader dont accepting it. Its for the moment no problem then still must updating the bootloader thru SWD.
Fast test of the new files:
Bootloader flashing OK and booting OK
NCP 6.6.4.0 flashing OK and booting OK and getting 1a c2 02 8b c2 8a 7e  from the comport.

Have a nice evening and great weekend !!

MW

closed 2020-07-21 by grobasoz

MattWestb, 2020-07-22:
Thanks one more time and little updated feedback for you.
The NCP 6.6.4.0 its up and running on standard zigpy / bellow = ZHA and real zigbee bindings its working but more intense testing its needed.
The NCP 6.7.4.0 with tasmota its loaded OK and the NCP its being initiated OK but its only in alpha stage so not possible testing so much. 
One user and i have trying the 6.7.4.0 on zigpy / bellow with the first v8 framing patch and having little problems but it's also WIP.
The next large thing its if Z2M also implementing support for the EZSP but its not up to my.

Hopefully it's being lesser "hardware storms" in the Tasmota Sonoff Zigbee Bridge thred then we have getting NCPs as signed OTA files and more software cocking can being done.

Mvh MW

MattWestb, 2020-08-14:
For no publick (github_nick)@gmail....

Mvh  MW