grobasoz/zigbee-firmware #2: [REQUEST] EmberZNet 6.8.0 NCP application firmware for EFR32 MG2

Hedda, 2020-09-15:
@mtx512 OK to request or suggest that release precompiled EmberZNet 6.8 NCP application firmware as optional downloads?

Silicon Labs EmberZNet SDK version 6.8.0 (6.8.0.2) was released this summer so it is new but not quite cutting-edge any more.

Mostly interesting in testing this on the Sonoff ZBBridge (Sonoff Zigbee Bridge) by ITead with Tasmota zbbridge as it is versatile.

Preferably also as an OTA image as well that can be flashed with Tasmota itself:

https://github.com/arendst/Tasmota/tree/development/tools/fw_zbbridge

Sonoff ZBBridge when flashed with Tasmota zbbridge and custom EmberZNet firmware firmware can be used in two ways:

**Sonoff ZBBridge as a remote pass-through WiFi bridge adapter to be used by other software like Home Assistant ZHA:**

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

**Zigbee2Tasmota (Z2T) is Tasmota's own native Zigbee to MQTT solution which is lightweight but not very user-friendly:**

https://tasmota.github.io/docs/Zigbee/

# Notes

Please note that release notes say that EmberZNet PRO stack version 6.8.x requires latest Simplicity Studio 5 (SSv5) which is new.

Might be nice for developers and users to have the option to upgrade to the latest and greatest firmware available from SiLabs ;)

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.8.0.2.pdf

https://www.silabs.com/documents/public/quick-start-guides/qsg106-efr32-zigbee-pro.pdf

Major new Zigbee feature in EmberZNet 6.8 and later is support for concurrent multiple PANs (multi-PAN) on one coordinator:

https://www.silabs.com/documents/public/application-notes/an724-multi-network.pdf

There is also support for "Dynamic Multiprotocol Development with Bluetooth and Zigbee" which might not be as interesting.

https://www.silabs.com/documents/public/application-notes/an1133-dynamic-multiprotocol-bluetooth-zigbee.pdf

Keywords: Silabs NCP EZSP Ember ZNet 6.8 Stack EFR32 EFR32MG EFR32MG1 EFR32MG2 Sonoff ZBBridge IKEA icc-a-1

Hedda, 2020-12-21:
FYI, Silicon Labs has now also released Zigbee EmberZNet 6.9.0.0 SDK as well:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.9.0.0.pdf

Note! EmberZNet 6.9.0.0 removes support for EM35x devices based devices.

grobasoz, 2020-12-22:
@Hedda - Thanks for the request and information. At this stage I am not 100% happy with the operation of the NCP code in later stack releases. I haven't managed to get a reliable NCP working on the Sonoff ZBBridge product with either 6.8.x or 6.9.0.
News about EM35x not being supported is also a little "disconcerting" as I still have customers with many thousands of EM35x devices installed (as well as EM250 - unsupported for many years now).
Simplicity Studio 5 is also a little "buggy" (I have a few support issues posted with Silabs) so at this stage any of my development code/systems are based on Simplicity Studio 4 and EmberZNet 6.7.6.
Regards,
Gary.

MattWestb, 2021-01-13:
I think Silabs is going making the 6.7.X.X as one "golden release" of EZSP after implanting more bug fixes for EM35X and first gen EFR32 (with 256kb flash and lesser). The last releases is having very serious bug fixes that is not implanted in 6.8 and 6.9. For all hardware above the 6.7 is and going to being the main brache and the multi pan and so on is making no sense for "normal" NCP user cases.

One better solution would having being implanting "sniffing" on the same channel at the same time the coordinator is running (not possible doing sniffing then the is running) and skipping the multi pan thighs.

Hedda, 2021-01-13:
> at this stage any of my development code/systems are based on Simplicity Studio 4 and EmberZNet 6.7.6

@grobasoz  FYI, apparently there is a critical bug fixed in Emberznet 6.7.7 and higher from SiLabs so suggest you upgrade:

https://community.openhab.org/t/zigbee-ikea-and-other-efr32mg1p-based-products-occasionally-crashing/109296/6

I also learned that the upcoming EFR32MG21 USB stick from ITead will come pre-flashed with EmberZNet 6.7.8 version firmware.

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.7.8.0.pdf

Zigbee EmberZNet SDK 6.7.8.0 GA released October 28, 2020

MattWestb, 2021-01-13:
@Hedda The bug can crashing all IKEA routers (very likely ols ZLL and new ZB3 ones but perhaps not the the new LED drive and CWS3 that is have the OTA firmware released for 2 days ago) and new Philips HUE tuya and so on that is using silabs chips.

The pinpointing of it done and verifying in deCONZ git by the ZHA devs (no named ones but some on . . ) with sending parent announcement from bellows and crashing the zigbee stack in the routers.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-739313960

I recommending using 6.7.8.0 is its possible then its around 10 nasty bugs fixes in the 6.7.X for both router, end devices and coordinators is all systems then is possible.
Sadly we cant forcing manufacturers upgrading there firmware but some is shure doing it after seeing its helping the system stability and not getting negative PR.

I think Gary can pointing out most of the fixed bugs as hi have "personal experience of them" and is very happy seeing them on silabs fixed list.

grobasoz, 2021-01-14:
@Hedda - I have added ZBBridge V6.7.8 and 6.9.0 to the repository. I have tested the NCP section but not the Tasmota interface as I'm concentrating on EFR32 - Zigbee2MQTT development.
Regards, Gary.

MattWestb, 2021-01-14:
Thanks Gary I have pinged Stephan so hi can testing it on his "open" bridge.
The i hope hi can getting one signed file from the factory for the more normal users.

Hedda, 2021-01-14:
> I'm concentrating on EFR32 - Zigbee2MQTT development.

Have you found any Zigbee2MQTT developers of willing to work on EZSP adapter and driver support for zigbee-herdsman yet?

https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts

https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter

grobasoz, 2021-01-14:
> Have you found any Zigbee2MQTT developers ...

No - I developed my own Zigbee2MQTT driver some time ago for my Home Assistant installation but never really got it ready for production - side tracked with Ethernet development ;) I am currently working on the Zigbee Green Power integration at the moment. My Green Power Devices (see GFR_PDT_Switch) and Philips Hue Switch work OK with my Coordinator but for some reason my GP devices don't message Zigbee2MQTT correctly as Zigbee2MQTT uses Group based messaging. I'll figure it out eventually :)

[image]

MattWestb, 2021-01-18:
By the way "someone" have cooking one nice ["**Tube0013 Zigbee Bridge ESP32 PRO Pure Black**"](https://github.com/zigpy/zigpy/discussions/586#discussioncomment-289078) !! and running ESPHome on it :-))

Hedda, 2021-01-18:
@grobasoz Might also be nice idea to offer a Zigbee router firmware as an option, similar to request in https://github.com/tube0013/zb_ebyte_eth/issues/1

It would be very nice to have a few EFR32MG21 based Zigbee routers as Zigbee network mesh extender (a.ka. Zigbee signal repeater)

That would be an excuse for some of us to buy several devices too ;)

grobasoz, 2021-01-18:
@Hedda - It's on my list 😎 Actually just a USB-Zigbee in **Router** mode! But right now I have to work on Thread based firmware for a customer 😒

closed 2021-02-18 by grobasoz

Hedda, 2021-02-18:
> @Hedda - It's on my list 😎 Actually just a USB-Zigbee in **Router** mode! But right now I have to work on Thread based firmware for a customer 😒

@grobasoz FYI, @tube0013 revealed a couple of days ago that he is working on a Zigbee Router based on an a Texas Instruments CC2652P (+20dBa power amplified) module as open source hardware->  https://github.com/zigpy/zigpy/discussions/664

I suggested he make a EmberZNet Zigbee Router variant with that design as well but he answered that he currently has no plans to do so himself at this time. 

> > I don't think I'm going to look at an efr32mg equivalent to this router design though, the module availability and options are not abundant on that side.
> 
> @tube0013 FYI, if you are looking to buy modules as surface mounted components to make and sell boards then know that MGM210P (and MGM210L) RF modules by Silicon Labs that contains EFR32MG21 and are also available.
> 
> https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules
> 
> https://www.silabs.com/documents/public/data-sheets/mgm210l-datasheet.pdf
> 
> https://www.silabs.com/documents/public/data-sheets/mgm210p-datasheet.pdf
> 
> MGM210PA32JIA and MGM210PB32JIA are both is a +20 dBm TX pre-certified module based module with both built-in antenna and RF Pin for optional external antenna.
> 
> https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules/device.mgm210pa32jia
> 
> https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules/device.mgm210pb32jia
> 
> The differnce between those two is that MGM210PB32JIA also have support for features Silicon Labs' enhanced security option, "Secure Vault" https://www.silabs.com/security
> 
> MGM210PA32JIA and MGM210PB32JIA looks to cost about $11 (US) dollars each when not buying in bulk
> 
> https://octopart.com/click/track?ai=1872&sig=02b5980&sid=459&ppid=108099587&vpid=615228293&ct=offers
> 
> https://octopart.com/click/track?ai=1872&sig=045aafa&sid=2401&ppid=108099587&vpid=618359838&ct=offers
> 
> https://octopart.com/click/track?c=1&ai=1872&sig=0881c5f&sid=459&ppid=112348737&vpid=667261090&ct=offers
> 
> https://octopart.com/click/track?ai=1872&sig=09e2041&sid=819&ppid=112348737&vpid=692111326&ct=offers