grobasoz/zigbee-firmware #3: [REQUEST] Change UART GPIO pins to PA0 PA1

Harwesta, 2020-11-28:
Hi friend.
Thank you for your work.
I have a request to make additional assemblies of your firmware with modified ports for UART with TX:PA0, RX:PA1 and bootloader's GPIO activation pin PD15.
And please change file format to HEX, but too many flashers use only HEX or ELF formats, to get it like this:
BTL_STD_S1_256-COM_PA0-PA1-PD15.hex
NCP_USW_115K2_S1_F256_676_PA0-PA1.hex

This module (EFR32MG1B232F256GM48) used in popular Wifi routers by SerComm (PCB ID: CQR00B-1.6) and this GPIO pins is use for CPU interconnect & placed out from EMI shield, see photo.
[image]
[image]

Thanks

grobasoz, 2020-11-29:
@Harwesta - unfortunately I don't have access to EmberZNet V676 stack on my computer - we are now at V678.

Harwesta, 2020-11-30:
@grobasoz No problems, newer version - better version :)
The most important thing is the coincidence of the UART ports with the board layout.

grobasoz, 2020-11-30:
I have added the code but did not test it - I don't have the part available.

Harwesta, 2020-11-30:
I flash bootloader, is fine.

`$ ./eblink -I stlink,dr -S silabs-auto -P ../scripts/ -F erase,verify,run,file=BTL_STD_S1_256-COM_PA0-PA1-PD15.hex

EBlink version 3.8-[57] by Gerard Zagema

Interface USB# : 2821040032124353354B4E00
Interface type : STlink/V2
STlink connect : Hot plugging
Target voltage : 3.17V
Interface speed: 4000KHz
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Active break (level 2)
Silabs device  : EFR32MG1B232 - Mighty Gecko Gen1
Detected FLASH 256 Kbyte - Page 2048 bytes
Detected RAM   32 Kbyte
Reset: system
Flash: -- chip erased --

HEX file: BTL_STD_S1_256-COM_PA0-PA1-PD15.hex

Checking current flash content  
Flashing sector  0, address: 0x00000000, length: 0x0020
Flash verify  done
Flasher starts target
Reset: system
`

I try a flash main firmware, but really flashing only 8 sectors :(
Normally firmwale length is 80...95 sectors

`$ ./eblink -I stlink,dr -S silabs-auto -P ../scripts/ -F verify,file=NCP_USW_115K2_S1_F256_678_PA0-PA1.hex

EBlink version 3.8-[57] by Gerard Zagema

Interface USB# : 2821040032124353354B4E00
Interface type : STlink/V2
STlink connect : Hot plugging
Target voltage : 3.17V
Interface speed: 4000KHz
Target detected: Cortex-M4 (r0p1) with FPv4_SP
HW breakpoints : 6
HW watchpoints : 4
Fault unwind   : Active break (level 2)
Silabs device  : EFR32MG1B232 - Mighty Gecko Gen1
Detected FLASH 256 Kbyte - Page 2048 bytes
Detected RAM   32 Kbyte
Reset: system

HEX file: NCP_USW_115K2_S1_F256_678_PA0-PA1.hex

Checking current flash content  
Flashing sector  8, address: 0x00004000, length: 0x0010
Flash verify  done
`

grobasoz, 2020-11-30:
@Harwesta - I have just tested the code on an old Thunderboard Sense (BRD4160A - EFR32MG1P132F256GM48) and it works fine.

Harwesta, 2020-11-30:
One question: new .hex file - is a Intel-HEX format?

grobasoz, 2020-12-01:
> One question: new .hex file - is a Intel-HEX format?

Yes - as generated by Simplicity Studio - and verified with Segger J-Flash.

Harwesta, 2020-12-01:
I don't understand what the problem is, but if you take your's old **.s37** files and convert them to **.hex**, then they are flashin normally.
The new **.hex** files that you made at my request are not loaded and not convert to .s37 format.
Converter _srecord-1.63_ say me _"garbage in line 1. No file data founded"_ 

I learned to use EBLink to accept Motorola s37 format, can you make another set of format files in .s37?
like as 
BTL_STD_S1_256-COM_PB14-PB15-PA0.srec (or .s37)
NCP_USW_115K2_S1_F256_678_PA0-PA1.srec (or .s37)

Tnx.

Harwesta, 2020-12-06:
Problem solved:
I used the latest version of Simplicity Commander, converted your .hex files to .37:
```
commander convert BTL_STD_S1_256-COM_PA0-PA1-PD15.hex --outfile BTL_STD_S1_256-COM_PA0-PA1-PD15.s37
commander convert NCP_USW_115K2_S1_F256_678_PA0-PA1.hex --outfile NCP_USW_115K2_S1_F256_678_PA0-PA1.s37
```
Renamed them to .srec, and flashed them with the EBLink:
```
eblink.exe -I stlink,dr -S silabs-auto -P ../scripts/ -F erase,verify,run,file=BTL_STD_S1_256-COM_PA0-PA1-PD15.srec
eblink.exe -I stlink,dr -S silabs-auto -P ../scripts/ -F verify,run,file=NCP_USW_115K2_S1_F256_678_PA0-PA1.srec
```
[converted_files.zip](https://github.com/grobasoz/zigbee-firmware/files/5648581/converted_files.zip)

closed 2020-12-07 by Harwesta

EmBitz, 2024-05-31:
Only the changed sectors are re-flashed with EBlink so if your new hex file has sectors equal to the old flash content then these sectors are skipped. You can increase the verbose level of EBlink to watch this more closely.