grobasoz/zigbee-firmware #4: EZSP NCP firmware update request.

MattWestb, 2021-01-13:
Hi gary hope you have one good time down under and you is on the beach with one tequila in you hands !!
Up in the north is around ZERO °C and snow in the air but nice sunny.

We have hunting problems in ZHA that IKEA controller devices is draining batteries and especially on Sonoff zigbee bridge and have pinpointing the largest problem to one bug in EZSP that is used in the coordinator. Silabs have fixing it in the last 6.7.8.0 release and Stephan is informed and very likely hi is posing one request for one updated firmware for Sonoff zigbee bridge.

Its looks like the bug is more likely being  triggered on second gens EFR32 but its have happening on the first gen to.

Ther for i asking you kindly if you can "cooking" one 6.7.8.0 for the "IKEA Billy platform" as you have successfully have done before.

HW config:

MCU: EFR32MG1P132F256GM32
PB15: RX
PB14: TX
PA0: Force bootloader boot

NCP UART SW (Software Flow Control)

Your cocked bootloader is working OK so no need for new one if you dont have other opinion.
Please s37 and GBL if its possible.

Thanks in advance and all the best for you in the new year !!!
And keep safe from (Great white and Wire) SHARKS and CORONA (beer) things !

Mvh  Mattias

grobasoz, 2021-01-14:
@MattWestb - Hello and Welcome to 2021 :) 
I have added 6.9.0 version - I will try build 6.7.8 version later today (30°C).
Please test the version and let me know if they work - I don't have the same devices as you to test fully...
Pretty safe here - thanks! No freezing weather, sharks, beer or Covid :)
Regards, Gary.

MattWestb, 2021-01-14:
Thanks @grobasoz and the same for you :-)

I have downloading them and trying flashing my Billy that is not in production (the wireshark sniffing one) for testing (its not so urgent for my then i dont trigger the bug so frequently).

I informing you if its not working OK but im very sure you have cooking it well as normal !!!

Today its new snow outside but corona lockdown so not possible going on icebear hunt on the streets of vienna :-((
Did you see the pictures from Madrid in the weekend with siberian huskies sledges and ski runners on the street ?

One more time great thanks from my and Stephan H for help with NCP firmware for our zigbee bridges wi cant living without help from down under.

grobasoz, 2021-01-14:
@MattWestb - De Nada :) 
Yes we saw weather in Spain and rest EU - too cold for me :p I was in Trondheim in Jan a few years ago - no fun!
Stay Safe up North!
Gary.

MattWestb, 2021-01-14:
Sorry but I have 2 questions that i think you are the right man to answering.

1. Is it possible cooking one 6.7.8.0 NCP firmware for the EM35X chips that is discontinued or is it already deleted in Gecko SDK Suite 2.7 ?
 (Or what is the latest version of  NCP possible to cooking for those chips ?)
2. Do you have or knowing some that have the compiler and knowledge for doing it (if its possible) ?

The question is coming up in ZHA then some users of old hardware also have observed the battery drains that the second gens EFR32 is often getting.
I can also being that its not existing or not being triggered on older chips then they have different CPU architecture / tool chang / different bugs.

The community have trying collecting all firmware for those sticks and the newest EmberZNet is 6.6.5 what i can see from the last updated ones: https://github.com/walthowd/husbzb-firmware (I think its the latest Silabs was precooked in the SDK).

Thanks in advance and dont forget securing your tequila on the beach for the invading penguins !

grobasoz, 2021-01-15:
@MattWestb - I have added EmberZNet NCP Version[ **6.7.8 EM357 NCP**](https://github.com/grobasoz/zigbee-firmware/tree/master/EM357) with 57k6 (Standard) and 115k2 baud rates. 
These are with the standard TXD and RXD pins for EmberZNet EM35x (Dev0680). 
For other boards, I would need the Serial Port pins, Bootloader Pin and Power Amplifier enable (and Active High or Active Low).
I have a CEL MeshConnect, ETRX3-HR-LRS-USB and EM3585/EM3587/EM3588 to test with but this work takes many hours and I have "real work" to do first :) (So no Tequila!)
Regards, Gary.

MattWestb, 2021-01-15:
Thanks @grobasoz for the time and efforts you have made !!
And congratulations to you new nice red car ! :-)

Dont do any testing for the moment i asking the "hardware hackers" to testing it so you dont need to do that.

If they is getting problems or more request its one later thing (problem).
The most important was the Sonoff Zigbee Bridge and its being good cooked but i have not getting feedback from the Tasmota community but it should working (the same with my Billy) !

Thanks one more time and hope you can getting little time for earning money for your Tequila collection and later enjoining it !!

Mvh  Mattias

MattWestb, 2021-01-15:
Greeting from the north from 2 happy Billy EZSP !!
```
2021-01-15 15:10:01 INFO (MainThread) [bellows.zigbee.application] EZSP Radio manufacturer: IKEA of Sweden
2021-01-15 15:10:01 INFO (MainThread) [bellows.zigbee.application] EZSP Radio board name: Billy EZSP by MW
2021-01-15 15:10:01 INFO (MainThread) [bellows.zigbee.application] EmberZNet version: 6.7.8.0 build 373
```
I was doing wrong the upgrading the NCP and changed the aktive in my production system with my test / sniffing one without thinking and was losing the network and have repaired 40 devices and more must being done :-((

All is looking great and one more large thanks for help !!

I closing this issue and opening one new if needed :-)

closed 2021-01-15 by MattWestb

MattWestb, 2021-01-28:
Hi Gary.
Tasmota have starting using the firmware and ilts looks the user can loading it without problems and no new problems is fund.

But one user have dring new batteries in 2 days on 3 devices.
I have recommending repairing the devices so the NCP is setting up the upll control new and the false is being corrected.
But hi still getting battery draining of IKEA remotes.
But som IKEA remotes little longer away from the coordinator is woking normal.
I think they is using one router for communicating and not the coordinator.
Stepnan H is also have his IKEA remote connected to the coordinator.

Now the quest for you !!
Have you experienced of EFR32 second gen NCPs having problem setting up pull control on direct connect end devices ?

deCONZ have seeing it with smartthings and IKEA devices (both EFR32 devices) and is resulting the end device is staying in short pull interval because long pull setting is wrong and the device is drawing the battery.

I have trying getting the "problem user" pairing one of the problem remotes with one CorBee coordinator for eliminating one faulty device (i have returning on E1743 remote to IKEA that was eating batteries for one year and i have around 10 of them).

@Adminiuga thinking the most problems is EFR32 second gen (sonoff zigbee bridge) 90% and the rest 10%  is "noise" in the air and hardware problems (with device and bad batteries) and i agree with him. Ikea have delivered many thousands of there remotes (all controlling devices) and many is in our systems (Tasmota, Z2M, ZHA and deCOZ) so its not one general bad hardware we is seeing.

The bug fix in 6.7.8.0 shall fixing the problem setting up pull control in EZSP but it can being more bugs that is not being found yet.

It can also being that tasmota is doing the configuration of end device in one wrong way and still is getting the problems with the fixed firmware as they also having the same problems with Philips HUE dimmer switch.

I dont have any EFR32 second gen for testing if i can triggering the problem and sniffing the setup of the device. And IKEA in austria dont have the new GU10 CWS with the new EFR32 second gen module and its closed because of the lockdown.

I asking you because you have very much hard and software experience that not many in the community is having and if you knowing or have some ideas that can helping solving the problem please sharing it.

Have one great time and thanks for all help you have given !!

Mvh Mattias

grobasoz, 2021-01-29:
Hi Mattias, 
Thanks for the information!
I have been watching the problems and will look into the possible causes when I have some spare time. I will need to buy a few Tradfri products to test though... I generally create all my own devices and "emulate" other (like Hue and IKEA lights), so don't have a bag of "test products".  
At this stage work comes first - I have to earn some $ for living :D
Stay Safe! Gary.

MattWestb, 2021-01-29:
Thanks Gary !
I trying getting more user that can pinpointing the problem and trying doing test with them if its possible to finding the logic behind wath is happening.

The syndrome with always creating own devices, i think your doctor have one very nice name for :-)) 
But im in the ground the same with electronic then is possible.

And you is in the first place and if you have some time and can spending some of it i being very glad but still money for living is prio one !!!

Take care and be safe.

Mvh Mattias

MattWestb, 2021-01-29:
Stephan H is considering that the EZSP 6.7.8.0 have solving the battery draining problems in tasmota (hi still having users that saying the opposite but its should not my problem) and ZHA is digging little more on the problematic on setting up the pull control. 

Its dont make sense for the moment that you is running around and buying "ugly, expensing, crappy , bad behaving swedish" zigbee devices.

I keep one and a half eye for more info is coming or if its have stabilised.

MW