grobasoz/zigbee-firmware #5: 115200 verses 57600 bit rate speed for Silicon Labs EZSP on EFR32MG1 and EFR32MG2?

Hedda, 2021-01-21:
What are the pros and cons with 115200 verses 57600 bit rate speed for Silabs EZSP serial protocol on EFR32MG1/EFR32MG2?

kirovilya brought up the question in https://github.com/Koenkk/zigbee-herdsman/issues/168 for [EByte-E180-Z120B](https://www.cnx-software.com/2020/04/27/ebyte-e180-zg120b-tb-zigbee-3-0-evaluation-board-features-silicon-labs-efr32mg1b-zigbee-thread-soc/) and [SM-011 V1.0](https://github.com/zigpy/zigpy/discussions/586) modules.

PS: The upcoming Sonoff Zigbee 3.0 dongle from ITead will ship with firmware config for 115200 -> https://fb.watch/386X5ET50f/

Hedda, 2021-01-21:
Note! This is cross-posted at https://github.com/zha-ng/EZSP-Firmware/issues/6

MattWestb, 2021-01-21:
The EM35X was having 115200 with hardware flow control and 57600 with software flow control as standard (probably hardware limitations) and its inharaged to the EFR32 thats looks woking OK with 115200 (alos ZHA is having 57600 as standard for EZSP NCPs) and most host systems is having it for companilits reasons.

Hedda, 2021-01-22:
Update bellows README.md for zigpy / ZHA with more info about flow control speeds in PR https://github.com/zigpy/bellows/pull/391

_**Note that there are generally two image versions of the Silabs EmberZNet NCP firmware in use. One version operates at a baud rate of 115200 with RTS/CTS flow control (i.e. hardware flow control), the other operates at a lower baud rate of 57600 with XON/XOFF flow control (i.e. software flow control). If you are flashing firmware image to your own EM35x or EFR32 adapter then it is advisable to use a version with hardware flow control, Many available commercial dongles (e.g. Nortek HUSBZB-1) seem to ship pre-flashed with a firmware that uses the lower speed and software flow control.**_

Paraphrased from https://www.openhab.org/addons/bindings/zigbee/#ember-ezsp-ncp-coordinator

MattWestb, 2021-01-22:
I dont agree with the recommendation !
Silabs have one bug case if having massive transactions in large systems and using high speed and sw flow control it can being problems with the buffers in EZSP.

But for normal use 115200 SW is working very well and its better than the lower with HW then buffers is emptied faster and not halting the coordinators "normal" work.
If having long cables and unstable environment the lower speed and HW is recommended but its not the case then its some cm between the NCP and the host com port pins.

And i have long com cables and no nice environment for My Billy on the Pi3:
[image]
Pi3B+ ESP32 CAM and Billy EZSP also hosting HA and many ESPHome devices dont using ETH but 5Ghz WiFi so all is against it but still working well :-))

Gary please don't kill my for the bad hardware implantation !!!!

MattWestb, 2021-01-22:
I think (= not knowing) that the comports on EFR32 chips is using real hardware UART with DMA transfer so its having no problems with timing of the communication if its lagging and the CPU only need reading / writing to the software buffers as long not killing the main thread its running.

Hedda, 2021-01-22:
I did not base that on empirical tests, instead I just stole the recommendation from https://www.openhab.org/addons/bindings/zigbee/#ember-ezsp-ncp-coordinator

I believe that @cdjackson from @zsmartsystems wrote that original recommendation for EM358, (and he is also the author of https://github.com/zsmartsystems/com.zsmartsystems.zigbee and https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer plus maintainer of [OpenHAB Zigbee Binding](https://www.openhab.org/addons/bindings/zigbee/)).

And I assumed that if EM358 has no issues with 115200 hardware flow control then EFR32 should have no issues either?

> i have long com cables and no nice environment for My Billy on the Pi3:

OMG, use shorter wires/cables! ;) But is your setup common (e.g. the norm) or the extreme exception? I would think most other people more common setups consist of a bought USB dongle/stick or a Shield such as for example [Elelabs Zigbee USB Adapter ](https://elelabs.com/products/elelabs-zigbee-shield.html)or [Elelabs Zigbee Raspberry Pi Shield](https://elelabs.com/products/elelabs-zigbee-shield.html)?

MattWestb, 2021-01-22:
I dont knowing how the hardware part of the EM35X is working and the EM368X have multi protocol versions (with Z-Wave) that can making it difficult getting the communication working stabile and need the hardware flow control then the NCP(s) is not have time "talking to the host" all the time.

ZHA have dropped the HW flow control so its not possible using their firmware.
And its looks no EFR332 have problems with 115200 with software flow control.

I think its more interesting if EM35X family is stable with 115200 SW than converting all new firmware (i think gary have make 95% of all adapter working with new ones if they is working OK on all hardware but waiting for verification on that) to it and make 115200 sw default in  ZHA.

Then testing and working with longer cables its OK but doing one PCB its being much shorter and more away from other signals sources that can making interference.

MattWestb, 2021-01-22:
I  think the owner of this git and @walthowd can awsering if Em35X devices is working satbile with 115200 sw or need 57600 with sw (then HW is not working with ZHA)  and the rest is speculation without experience.

closed 2021-02-18 by grobasoz