grobasoz/zigbee-firmware #6: Request Zigbee Router mode firmware for EFR32MG1

Harwesta, 2021-01-28:
Hi Gary.
Nice to see you again & thank you for your work.

Can you cooking a Zigbee router mode firmware for EFR32MG1 (gen1) ?
With UART 115200 baud SW
GPIO TX:PA0, RX:PA1
bootloader GPIO activation pin PD15.
SDK 6.7.8

Thanks.

grobasoz, 2021-01-28:
@Harwesta  - Probably a good idea to give me the chip number? Thanks Gary.

Harwesta, 2021-01-29:
@grobasoz 
Oh, sorry...
This is EFR32MG1B232GG

grobasoz, 2021-01-30:
@Harwesta - I have added MG1B232 based NCP firmware to the repository. Regards Gary.

Harwesta, 2021-01-30:
@grobasoz 
Thanks you 🥇

FeelTheLemon, 2025-05-21:
Hi @grobasoz. Thank you for NCP firmware.
Is it possible to build standalone router firmware for this chip?

grobasoz, 2025-05-22:
@texvek - Sorry - this issue was closed so I wasn't notified of your request.
@FeelTheLemon - What are your pin settings for the router?

FeelTheLemon, 2025-05-22:
@grobasoz 
Pins are the same 
> UART 115200 baud SW
GPIO TX:PA0, RX:PA1
bootloader GPIO activation pin PD15

PD15 is only used in bootloader as i understand, so it can also be used as a pairing button.

grobasoz, 2025-05-22:
@FeelTheLemon - Thank you. 
I will use PD15 for turning pairing on for 3 minutes if joined to a network, and it can also be used to leave a network by holding for ~5 seconds. If not joined the router should try join as power is applied. 

Are there any LEDs available?
I have a router for Sonoff ZBDongle and it uses the LED for signaling status. 
OFF No power/Off.
100mS flash every 1 Second - On and Joined.
500mS toggle every 1 Second - On and NOT Joined.
900mS flash every 1 Second - Join Permitted.

I will try to develop the firmware for the MG1B tomorrow. Do you know how much flash it is? 512 or 256?

FeelTheLemon, 2025-05-22:
@grobasoz
That would be nice logic.

No, there are no LEDs directly connected to the chip, but there are 2 more unused pins easily available out of metal shielding PB12, PB13.
I think we could use one, to make small customization for LED possible. 

Flash is 256 KB.

Thank you!

FeelTheLemon, 2025-05-27:
@grobasoz 
It works great!

I was unable to flash it via bootloader ([this one](https://github.com/grobasoz/zigbee-firmware/blob/master/EFR32%20Series%201/BTL_STD_S1_256-COM_PA0-PA1-PD15.hex)) for some reason, but with SWD it went fine.

I'm using ZHA instead of Z2M and the device has joined network on boot, i've successfully paired few end devices to it, network leave command via uart also works.
After power loss it remembered network which is great.

So everything works! Thank you very much!