grobasoz/zigbee-firmware #9: [DISCUSSION] EFR32 BTL_INT (BTL INT) "application" bootloader with support for updates via OTA (Over-The-Air) and BLE interface

Hedda, 2021-02-19:
_Originally posted by @grobasoz in https://github.com/grobasoz/zigbee-firmware/issues/8#issuecomment-781671868_

There are also a couple bootloader options... 
1. The "standalone" bootloader  (I call BTL_STD) like EZSP uses and allows XModem uploads.
2. The "application" bootloader (I call BTL_INT) that uses the device application and can support over the air (OTA) updates.

I prefer the latter (BTL_INT) as it means the firmware can be uploaded to flash without interfering with the functioning of the Zigbee device and supports OTA updates from a gateway - it also allows for me to implement a BLE based flashing interface so mobile phones can be used for OTA updates.

Hedda, 2021-02-19:
Would having such OTA ability in the bootloader also mean that EFR32 Zigbee Coordinator firmware and Zigbee Router firmware could be upgraded via the OTA capabilities in zigpy and zigbee2mqtt as long as code OTA providers for them?

For zigpy that would be by extending https://github.com/zigpy/zigpy/tree/dev/zigpy/ota

By the way, there is a long discussion about zigpy OTA support and OTA providers in https://github.com/zigpy/zigpy/issues/154

grobasoz, 2021-02-19:
@Hedda - yes, OTA is a standard ZCL command - if implemented...

closed 2021-03-08 by grobasoz