grobasoz/zigbee-firmware #10: Deep Teck quests / talk with Gary

MattWestb, 2021-03-12:
## Hi Gary !

I have some very strange observation then sleeping end device is direct connected to one coordinator and is pulling its parent (the coordinator) or trying sending one attribute report or commands to it and the coordinator is ignoring its children but is responding to other childrens requests.

See https://github.com/zigpy/zigpy/discussions/604#discussioncomment-470496

Its looks like EZSP coordinator its some time lazy and not responding on received frames and also not replaying with ack in the IEEE 802.15.4 layer.

I think its some  of the "personality" of the EZSP stack, then IKEA lights is famous for not responding on multiple commands being sent to them (Like power on, dimmer up and changing color in fast pace).

I was using your latest EZSP 6.5.1.0 and its looks being worse then the 6.7.8.0 that have the "NCP fix"so 6.7.8.0 is defiantly being helping our systems. The test above is with the better 6.7.8.0.

If having one router in the near the sleeping end device can jumping to it if the coordinator is "lazy" and the device is not leaving. If not having any routers is very likely the end device is doing some beacon requests and if dot finding any router it is leaving the network until being waked up by pressing one button.

I cant see if some frames is corrupted from end device or the coordinator then EZSP sniffer is not sending them to the host app.

Is it possible using the PIT interface for see that and wot is needed for using PIT sniffing  with EZSP devices ?
Is one original dev kit needed or is it possible with one "Billy" ?
Is PIT working without licence in Simplicity Studio ?

ZHA is configuring long pull with 24 (6 seconds) that is to short (and only draining batteries and dont do any good) and i is trying the "IKEA standard" 1160 (4.88 min) and its  working much better and its much unlikely triggering the leaving then the coordinator is having one "lazy period".

No priority then you have your work and socials life in the first row but i welcome your experience and knowledge.

Thanks in advance and all the best !!

Mvh Mattias

grobasoz, 2021-03-13:
Hi @MattWestb - Always nice to hear from you in the North :D I will look into this problem - it doesn't seem to be solved with the new EZSP firmware and ZHA? Does the problem exist with Tasmota-EZSP based Gateways?
From the link you supplied, the battery on the IKEA 5 button remote is draining?  I don't have any of these and IKEA don't deliver to my area.  Does the problem occur with any other IKEA Tradfri (battery) products? 

I don't use ZHA or Tasmota so will need to build a dedicated system for ZHA (Home Assistant) EZSP.

Regarding PTI (Packet Trace Interface). You need (for Zigbee, add PTIClk for BLE), two port pins, PTISync (or PTIFrame) and PTIData. These are normally connected to the Silabs WSTK and Simplicity Studio for tracing in the Network Analyzer. Unfortunately I don't have time to see if there is a way to use PTI without the WSTK.

The license is normally only for development of products using the EmberZNet stacks - I think you can use the Network Analyzer without a license - if you can find a way to get the PTI packets into Simplicity Studio. For that, you normally need a WSTK board that supports PTI, the cheapest is currently the $99 SLWSTK6021A. That also gives you access to the EmberZNet Zigbee stack :) (Considering the first Ember Zigbee stack cost me US$3000+ in 2005, $99 is a good deal :D)

On a side note, I see the tube0013 Zigbee-Ethernet has been released? It looks great. I stopped development as it seemed from comments the price of US$30 was too high? These are US$52? Maybe I make my own and sell for US$50 :D I have about 1000 Zigbee modules in stock. But I don't like ESP32 for reliability so will use another device for Ethernet. 
 
Also if you want me to send you some Zigbee hardware to 'play with', send me your shipping address - I'll send you "Easter Bunny Present'.

Regards,
Gary.

grobasoz, 2021-03-13:
My version of Zigbee-Ethernet :) Not in production :(
[image]

I will make another using WT32-ETH01 - like Hedda suggests :)

MattWestb, 2021-03-13:
Thanks for replay and its always great getting information from down under :-)()()(

IKEA is using EZSP 5.7.2.0 in there GW and the 2 ZLL device that not have getting ZB3 (The old motion sensor and the old RGBW bulb) all other have getting ZB3 and some is using 6.2.3.1 but the most 6.3.1.1 (most controlling devices) but we have not getting the version of GLB OTA files only from EBL files.

If looking in Silabs release note its around 10 bugs and most is fixed in the 6.7.7.0 that can hitting the controlling devices and also some (parent accounts is locking the stack until power reset) that can hitting the router devices.

We cant do so much with the devices but learning wot is happening and perhaps finding work around for getting it working better.
For Sonoff and also EM35X and EFR32 first and second gen EFR 6.7.8.0 is helping then no user dint have "super draining" but its can triggering other bugs in the device firmware

I dont have any EFR32 second gen Coordinator (One LDL LED stripe controller but its not getting one new  firmware) but then i was testing tasmota for one year ago i was having one E1743 (  2 button remote) that was draining in some days and was doing it more time = tasmota is not setting up the pull OK and later the devs was stopping battery reporting of IKEA controllers.
But later also Philips HUE remotes start draining batteries. But  tasmota devs is saying iits being OK with your 6.7.8.0 and for 2 weeks ago Sonoff was releasing one "fixed" 6.7.8.0 then they was having some bad data setting in there signed firmware (I was thinking they have taking your sonoff firmware and signing it and releasing it as one new factory one !!).

I can verifying that 6.7.8.0 is replaying much better then 6.5.1.0 from frames revived from its children both end device and routers. And Silabs cant do to large / drastically changes in the stack without braking many things. So i think 6.7.8.0 was one great thing for all EM35X and EFR32 devices !!!
And i thanking you very much for doing all possible and impossible firmware for our devices !

The last part i is looking on is trying catching the last smaller parts that user having problem with and its looks being one problem if the coordinator is "busy" and children is trying talking to it and can leaving if not getting replay in time and is not having routers around. Many Tasmota user only having one coordinator and  some sensors and no or very few routers  (perhaps not in range for the controller) in the network and getting problem then the devices cat jumping to one router is the coordinator is having one lazy period.

From last night with 3 remote connected to the coordinator and with one extra router for jumping 2 if having problem:
```
No.	Time	Source	Destination	Protocol	Length	Info
2614	23:23:46,608467	0xe16f	0x0000	IEEE 802.15.4	76	Data Request
2615	23:23:46,611033	N/A	N/A	IEEE 802.15.4	69	Ack
2634	23:25:54,722036	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2635	23:25:54,724237	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2636	23:25:54,725927	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2637	23:25:54,727529	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2638	23:25:55,760770	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2639	23:25:55,762887	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2640	23:25:55,776560	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2641	23:25:55,779061	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2642	23:25:56,807883	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2643	23:25:56,809756	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2644	23:25:56,811375	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2645	23:25:56,813052	0xbc1b	0x0000	IEEE 802.15.4	76	Data Request
2646	23:25:56,814609	N/A	Broadcast	IEEE 802.15.4	74	Beacon Request
2647	23:25:56,839047	0x5c1f	N/A	ZigBee	92	Beacon, Src: 0x5c1f, EPID: e5:f5:25:15:55:65:b5:05
2648	23:25:56,963770	0xbc1b	0x5c1f	ZigBee	111	Rejoin Request, Device: 0xbc1b
2649	23:25:56,965542	N/A	N/A	IEEE 802.15.4	69	Ack
2650	23:25:57,000323	0x5c1f	0x0000	ZigBee	119	Route Record, Dst: 0x0000
2651	23:25:57,002567	N/A	N/A	IEEE 802.15.4	69	Ack
2652	23:25:57,005107	0x5c1f	0x0000	ZigBee	119	Route Record, Dst: 0x0000
2653	23:25:57,006692	N/A	N/A	IEEE 802.15.4	69	Ack
2654	23:25:57,037932	0x5c1f	0x0000	ZigBee	132	APS: Command
2655	23:25:57,039728	N/A	N/A	IEEE 802.15.4	69	Ack
2656	23:25:57,041522	N/A	N/A	IEEE 802.15.4	69	Ack
2657	23:25:57,149386	0xbc1b	0x5c1f	IEEE 802.15.4	76	Data Request
2658	23:25:57,151558	N/A	N/A	IEEE 802.15.4	69	Ack
2659	23:25:57,195327	0x5c1f	0xbc1b	ZigBee	121	Rejoin Response, New Address: 0xbc1b
2660	23:25:57,197169	N/A	N/A	IEEE 802.15.4	69	Ack
2661	23:25:57,199498	0xbc1b	Broadcast	ZigBee ZDP	129	Device Announcement, Nwk Addr: 0xbc1b, Ext Addr: SiliconL_ff:fe:8e:00:31
2662	23:25:57,201078	N/A	N/A	IEEE 802.15.4	69	Ack
2663	23:25:57,204797	0xbc1b	0x5c1f	ZigBee	120	End Device Timeout Request
2664	23:25:57,233305	N/A	N/A	IEEE 802.15.4	69	Ack
2665	23:25:57,235735	0xbc1b	Broadcast	ZigBee ZDP	129	Device Announcement, Nwk Addr: 0xbc1b, Ext Addr: SiliconL_ff:fe:8e:00:31
2666	23:25:57,237607	0xbc1b	0x5c1f	IEEE 802.15.4	76	Data Request
2667	23:25:57,239356	N/A	N/A	IEEE 802.15.4	69	Ack
2668	23:25:57,241542	0x5c1f	0xbc1b	ZigBee	120	End Device Timeout Response, Success
2669	23:25:57,258142	N/A	N/A	IEEE 802.15.4	69	Ack
2670	23:25:57,260588	0xbc1b	Broadcast	ZigBee ZDP	129	Device Announcement, Nwk Addr: 0xbc1b, Ext Addr: SiliconL_ff:fe:8e:00:31
2671	23:25:58,201270	0xbc1b	0x5c1f	IEEE 802.15.4	76	Data Request
[... cut ...] W (LIDL one) was stop working and its not possible login in as root (Its still running somthing but i thing tuya have locking the firmware as it was running  tuya app and have detecting changed files.

I think i letting the "battery daring problems" as it not bringing any more good things for normal users and moving on with more interesting things . . .

I must running but i mailing you my address :-))

All the best and hope the Easter kangaroo is bringing you much nice things ;-))

Mvh  Mattias

MattWestb, 2021-03-31:
Hi Gary i trying getting over the first step in Simplicity studio 4 and was getting problems with long file names and was finding the problems in the silabs community but no solution for it so i was deleting SS4 and reinstalled it on `d:/ss4` instead of the long silabs default and its start working compiling bootloader and Z3light demo and also flashing on the EFR32MG22 board.
And i can using the PTI for network analyze after changing the channel of  the Z3Light :-)) 
And i was able changing the RF channel thru CLI at the first try THAT is not so bad for one beginner !!

I need reading little more so can putting in on real working key in the demo so can joining on real network but its not having so high priority but is on the list.

I have one more wild thinking of the never ending problems with Sonoff ZBB.
I have reading that all pre 2 gen EFR32 chips was calibrated and the hardware config was burned in the chips but with 2 gen is it not done and shall being done in the board / device manufacturing process.

I have not finding the silabs paper that was finding it in but i shall looking more if i can finding it.

Back to my "old problem" with Sonoff ZBB they have released updated EZSP 6.7.9.0 with "updated configuration parameters" and then my warning bells is start sounding, wot was wrong in the first case ?

The hardware config burned in the chip is calibrating the RF parameters and shall being done OK for FCC and other complaint certifications.
But  is it also  calibrating the clock frequency and other parameters that can doing that  the radio is not working OK ? I must reading more to finding it out.

My theory is that the SM-011 module that Sonoff ZBB is using and the new USB-Zigbee stick is not calibrated and can having problems with radio is not correct configurated  and the sonoff devs is trying compensating it with tuning some chip parameters.
Devices with good made modules like the new IKEA that is using on silabs module is being calibrated from the factory and is not having this problems (but is expensive).

And then sleeping end devices is coming back from sleeping and is trying pulling its parent and its parents radio is "off tuned" the sleeping end device is thinking is being left  alone (the network is gone) ans start network scans and draining there batteries.

This scenario is fitting in the pattern that we is seeing from user having this problem and its working OK if its parent is one router and not the coordinator.

One more think with Sonoff ZBB and USB stick they is 20 dBm chips but the RF tuning is made for 0 or 6 dBm then looking on the board and comparing with silabs references setups documents, that can also making the same problem with sleeping end devices having problem keeping the radio pipe open to it parent.

My conclusion is that my "Billy EZSP" (with Gary FW) is not so power full as the Sonoff but is working much better  in the real life !!!

Its only "loud brainstorming" from my side but i think you have much experience and perhaps can bringing little  light in the "dark SonOFF" times ;-)

No need answer if you not have time or no having any wild out of the box thinking.

Thanks for letting my spamming your git !!!

Mattias

grobasoz, 2021-03-31:
@MattWestb - Excellent news you have a Silicon Labs development kit! Welcome to the "wonderful world of Zigbee development" :D

Did you perhaps buy one of [these](https://au.mouser.com/ProductDetail/Silicon-Labs/SLSDA001A?qs=P0I6NS6hZyTk%252B%252BlMhaP2eQ%3D%3D) as well? Very useful for custom designs. I put the connector on all my radio boards.

It seems you have a great start on the basic device development. It may be an idea to build your own Zigbee Switch to test with the IKEA battery draining problem. Another option is to use the development tool as a Zigbee Packet Sniffer to see what is happening "Over the Air"...

Have Fun! Gary.

MattWestb, 2021-04-01:
The firmware cooker of SonOff ZBB and USB stick is "recommending" cooking firmware with "HFXO CTUNE value to 128" = they have not calibrating the chip then manufacturing the SM-011 and USB stick and must patching the firmware for getting it working OK with other devices then all devices is miss tuned !!!

So that is explaining way IKEA remotes and other sleeping end devices is having so large problems and draining there batteries then must re sending frames that the coordinator cant receiving and its working well then is being  connected to one router.

My tension is some time scaring my then its saying that one have digging down one dog some where and i can feeling it and its can being scaring but also god to listening to.

MattWestb, 2021-04-01:
One deep T question:
Where shall i putting the Zigbee alliance master link key 9 in SS4 for making it possible pairing the Z3Light with on ZB3 coordinator ?

I have reading it before but i have not finding it in 2 days of searching.
I think the Silabs community is no liking it being written in its forum :-((

grobasoz, 2021-04-03:
@MattWestb - Sorry for the delay - holidays here :) The key is added by default in Zigbee 3 so you don't need to add it.
CTUNE should be set during manufacturing by the board supplier. Regards, Gary.

MattWestb, 2021-04-03:
Thanks for repay !!

Then i can trying adding my flashed Z3Light to my test network :-))
I was thinking (but not testing) i was only using the ZA test key and Silabs private keys.
And its only needed doing patching if like implanting touch link paring (the very secret ZLL master key) that its little out of scope for the moment.

I have reading more and the bad thing is looks like tuya is having doing the same mistake as SonOff but i must reading my not updated tuya ZBGW for verifying it but it shall not being any problem its have SWO pins.
The new Sonoff USB-Zgbee 3 stick is the same: not calibrated and have putting the offset in the firmware so the cheese is making shortcuts in the hardware production that Silab is not recommending for getting cheaper device on the market :-((

deCONZ is also doing strange things (devised dont getting IEEE 802.15.4 ack and re sending frames oft that can being miss tuned radio or lazy NCP) but also not confirmed then i dont have one god signal analyzer.

Have one great Easter holidays and all the best for you !!

Dont eating TOOO much eggs then its not so god in our age but is tasting great ;-)

MattWestb, 2021-04-08:
Hi Gary !!

I have trying doing all mistake you can do in SS4.

Then have getting on project working i have copy it and pasting it and SS4 is making one new one but all is pointing the original so editing the "original project" and if deleting  the "old" files the new is not working then some is depending of the original ones.
My latest creation: Z3Light with Z3L Control Bridge.
[image]
```
Z3LightSoc03>Z3LightSoc03>info
MFG String: 
AppBuilder MFG Code: 0x1002
node [(>)086BD7FFFEC3901A] chan [15] pwr [3]
panID [0xD337] nodeID [0x06B0] xpan [0x(>)E5F525155565B505]
parentID [0x5550] parentRssi [-44]
stack ver. [6.7.9 GA build 405]
nodeType [0x02]
Security level [05]
network state [02] Buffs: 75 / 75
Ep cnt: 4
ep 1 [endpoint enabled, device enabled] nwk [0] profile [0x0104] devId [0x0101] ver [0x01]
    in (server) cluster: 0x0000 (Basic)
    in (server) cluster: 0x0003 (Identify)
    in (server) cluster: 0x0004 (Groups)
    in (server) cluster: 0x0005 (Scenes)
    in (server) cluster: 0x0006 (On/off)
    in (server) cluster: 0x0008 (Level Control)
ep 2 [endpoint enabled, device enabled] nwk [0] profile [0x0104] devId [0x010D] ver [0x01]
    in (server) cluster: 0x0000 (Basic)
    in (server) cluster: 0x0003 (Identify)
    in (server) cluster: 0x0004 (Groups)
    in (server) cluster: 0x0005 (Scenes)
    in (server) cluster: 0x0006 (On/off)
    in (server) cluster: 0x0008 (Level Control)
    in (server) cluster: 0x0300 (Color Control)
    in (server) cluster: 0x1000 (ZLL Commissioning)
ep 3 [endpoint enabled, device enabled] nwk [0] profile [0x0104] devId [0x0840] ver [0x01]
    in (server) cluster: 0x0000 (Basic)
    out(client) cluster: 0x0003 (Identify)
    in (server) cluster: 0x0003 (Identify)
    out(client) cluster: 0x0004 (Groups)
    out(client) cluster: 0x0005 (Scenes)
    out(client) cluster: 0x0006 (On/off)
    out(client) cluster: 0x0008 (Level Control)
    out(client) cluster: 0x0300 (Color Control)
ep 242 [endpoint enabled, device enabled] nwk [0] profile [0xA1E0] devId [0x0061] ver [0x01]
    out(client) cluster: 0x0021 (Green Power)
Nwk cnt: 1
nwk 0 [Primary (pro)]
  nodeType [0x02]
  securityProfile [0x05]

Z3LightSoc03>zcl on-off toggle

Msg: clus 0x0006, cmd 0x02, len 3
buffer: 01 38 02 

Z3LightSoc03>send_multicast 2 3

Z3LightSoc03>
T00000000:RX len 3, ep FF, clus 0x0006 (On/off) FC 01 seq 41 cmd 02 payload[]
On/Off set value: 01 02
Toggle on/off from 00 to 01
Event: move from 1 to 51 (diff +1)
Z3LightSoc03>
T00000000:RX len 7, ep FF, clus 0x0006 (On/off) FC 08 seq 46 cmd 0A payload[00 00 10 01 ]

T00000000:RX len 5, ep 01, clus 0x0006 (On/off) FC 18 seq 46 cmd 0B payload[0A 00 ]
ERROR: tx 66, Profile: HA (0x0104), Cluster: 0x0008, 7 bytes, ZCL Global Cmd ID: 10
ERROR: tx 66, Profile: HA (0x0104), Cluster: 0x0008, 7 bytes, ZCL Global Cmd ID: 10
```
Paired with my ZHA test and the EP 1 is added to one light group in (2) ZHA that is controlled by one IKEA remote and I sending toggle command with CLI from EP3  (Control bridge) to group 2 and all lights is toggling nicely and reporting all that is happening in the group.

My thinking is modding one IKEA 2 button On/Off dimmer switch so the 2 normal pins  is on/off and dimming up/down and 2 extra is start / stop sending on with off time 60 seconds ever 50 seconds for over riding motion sensor that is sending with on with off every 60 seconds but not than i taking on shower or alternative 10 / 20 min on with off that is stopping then using then normal on/off.

I was frying one more (number 3) IKEA module then erasing flash and flashing wrong boot loader and cant recovering it so must playing with the EFR322MG22 radio boards that is possible to recover if flashing them wrongly.

And i cant going to IKEA then its hard lockdown but i like baying the new remote (E2001) with the new E27 CWS (RGBW) in one family pack. The cheapest is baying on on/off switch and one E27 WW pack for 9.99€ = 5€ for one EFR32MG1P devices that can being flashed.

Network analyzer with PTI is working very well but need learning little more how to doing it well.

I think many deCONZ and all Sonoff ZBB and the new USB Zigbee 3 stick have problem with having miss tuned radios but i have not finding how to proving it.
With deCCONZ is devices doing many re sending then its not getting 802.15.4 Ack and also devices is not reacting on receiving from it that can being miss tuned radio or only bad coded software and / or hardware.

Have one great time and keep safe and do your cooking well !!

Mvh Mattias

grobasoz, 2021-04-08:
Hi Mattias, You are doing very well with your SS4 development! Some developers take a long time to even create a working project. You have created firmware, joined a network and can send /receive messages! That is great!

I find it easiest to re-create new projects from the beginning each time. I have found that doing a copy/import of a project can cause some issues. In my case, I have my own "zigbee application" folder that I add to each project after it is created.
You can then add these files and any "DEFINES" into your project in the "Includes" Tab. In the case below, the callbacks are handled in the grCallbacks.c and Zigbee messages in the grMessageHandler.c. The defines determine the type of device (COO, RTR, SED etc). Also you can add your own "tokens" for storage in EEProm (ie zbTokens.h).
[image]

I have not found that the CTUNE setting (unless very bad) will have much of an effect on the general operation of the radio. Normally it is antenna issues (hardware) or power settings(software) - assuming you have a working project.

"Key" to using Network Analyzer is setting the security key :) (Window->Preferences->Network Analyzer->Decoding->Security Keys). You can get the Key from using CLI command "keys print" and copy the "NWK Key".

Unfortunately no Zigbee for me - another big project on the go :(
Have fun!
Gary.

MattWestb, 2021-04-19:
Hi Gary :-))

I have making the classical 2 steps forward and 3 back the last days.

I was getting one EZSP 6.9.2.0 bootloader and NCP made for My billys and then my laptop was blocking the graphic (dark screen because shared memory is being swapped out and locking the system) and i was doing on hard power down. 
(was running my test ZHA in docker on it to so it was  perhaps little tight with resources on on Pentium Silver laptop)
Then it was possible starting SS5 but it was thinking on instance was running.
Deleting the workspace folder and trying adding new projects and is locked in appbuilder running 100% but not doing any thing. 
Deleting and reinstalling SS5 . . . . . . . . . . . .
2 days later i was getting SS4 working doing on bootloader but no NCP. After doing on Z3Light it was possible also doing NCP !!
Installing SS5 and doing one bootloader = OK Doing on Z3Light for billy its building but complaining of to less space on flash. Then its possible building NCP for Billy !!!

The god thing is that the WSTK have special commands in the J-Link adapter for recover broken chips and connecting hardware reset it was possible recover 3 broken IKEA modules so now i have 4 working one after soldering the 10 pads/pins to my standard connector :-))

Then our friend Hedda was insisting getting one HW flow control EZSP for Sonoff USB stick for using with OpenHAB (that only using HW) that dont have the hardware connection to the USB chip but hi dont like to understand that its not working.
So I was doing on EFR32MG21 with HW flow control and Sonoff pinning plus the 2 extra pins for hi :-Z 

I was having one cable and adapter for TI CC-2531 that is the same pinning and is converting to 2.5 mm pins that i using with the WSTK that was coming with the mini adapter and its working great for SWD flashing and debricking chips.
[image]

I trying putting the 2 PTI pins and the 3 SWD in HW configurator and generating and building but i dont getting the network analyzer working with the external PTI and also the vcom is also only data coming from the board that is on the socket on the kit but the SWD is switching to out then i selecting it. Do i need doing more in the firmware for getting it working or is it only putting the right pins in HW config ?

I have getting Billy running on EZSP 6.9.2.0 after "finding" the dc-dc trick :-)))
Do you have more tricks that can being good to knowing then playing around with SS ?
IAR is little to expensive for only playing around so its no alternative for the moment.
I dont have so mush space on my system SSD so i have the the SS5 installed on one A2 SD card and have not installed Bluetooth only Ember ZNet SDK then i dont have so much space or i must baying on larger M2 SSD for getting little more speed but its coming later if all is going well.

All the best for you and have one great time with or without Zigbee problems all the day and nights.

Mvh  Mattias

grobasoz, 2021-04-20:
Hi Mattias, 
Sorry to hear of your troubles. Yes SS4 and SS5 can be very CPU hungry :p I now run a 4GHz i9 as my i7 laptop struggled
:( 
[image]

You can speed up the software by disabling indexing - I do this and use VSCode for actual code development. 
[image]

I also run many Silabs workspaces - one for **my code** and the others for the Silabs "**updates**". When a new release comes out I install it onto a new workspace - then do a diff (WinMerge :)) and migrate any changes to my code. I do this as there are some "issues" with the Silabs code that I "fixed" and don't want them in my code. Here is an example from my old drive :)
[image]
Another tip is to try keep your code far away from Silabs code if possible. So if you do some changes to a part of the SDK, keep them in your own project file. For example. The Silabs Battery Monitor plugin doesn't work with Series 2 chips so I create my own code for that and replace all the Silabs code in the SDK. Normally an update will overwrite my code :(

As you know you don't need IAR with EmberZNet any more - unless you want to do Zigbee+BLE? 
The DC-DC "trick" is that Series 21 EFR32 don't use it? 
Sadly there are many "tricks" in the development that you learn as you go. Most answers are on the Silabs Zigbee Forum (just do a search). Also you can ask and you will meet YK who is very good at Zigbee :)
One "trick" is to remember that Series 1 bootloaders use the "bootloadername_**combined**" file. Series 2 just use the standard bootloader file.

With PTI - you just need to enable the Packet Trace pins then use Network Analyzer to check the packet data. Remember to add the network key ;)
VCom Enable is just needed to enable the serial port interface to the plugin board on the WSTK. In your case, you use external board so VCom isn't used. I don't use the plugin boards so always disable it in the hardware configurator. NB. Set the GPIO pin to "Disabled" first!
[image]
You can use SWO on the SWD connection if you don't have access to a USART. See [this](https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/12/03/setting_up_serialou-fJAC) for information. You can set the SWO pin using the "GPIO" setting in the hardware configurator.
You can also use the same hardware configuration for all your projects. I have a specific version for my Zigbee modules that I just use in my different projects. You just select Custom Hardware Configurator in the HAL tab of SS.
[image]

Regarding debugging - I see you have the Debug Adapter - that is great! Minimal requirement is SWDIO, SWCLK and GND. You can then add RESET if you want, SWO and PTI_D, PTI_F. 

One last "trick". In the ZCL Clusters tab, Home Automation profile (Endpoint 1), General Cluster, Basic Cluster Id, set the...
Manufacturer Name to **IKEA of Sweden**, 
ModelID to **TRADFRI bulb E27 CWS opal 600lm**
Power Source to **0x01**
That way when your product joins a Zigbee network it will look like an IKEA light :D
[image]

Since "Billy" is based on an IKEA light you shouldn't be breaking any rules :D

As usual - good luck with your dev fun!
Gary.

MattWestb, 2021-04-25:
Thanks Gary !!
I have uninstalled on old msys32 than i have seen that SS  is using modules for it and its can being bad if getting  on version that SS is not likening and doing some more moving of data from the internal SSD to the SD-card so i have space installing SS 4 and 5 in the internal SSD and its not very fast but now its working well (no locking for infinite time).
My convertebal have only on 1/5 of you CPU logical processors and  the 1/3 pf its seeped so cant comparing with your MONSTER but its working doing most things i trying :-))

VUART is working great i was also getting it working in bootloader mode (i dont knowing how but after erase chip and flashing bootloader i was having it in console 0 and it was reacting on 1 for starting download process.

PTI i is not getting working on external boards and i dont knowing what i doing wrong or perhaps is on hardware failure in the KIT / Adapter / cable. 
One HWConf from My control bridge: [custom_efr32mg1p132f256im32.hwconf.txt](https://github.com/grobasoz/zigbee-firmware/files/6371715/custom_efr32mg1p132f256im32.hwconf.txt)
Then putting in the 2 PTI pins its being showed OK but if looking in the GPIO table its looks like its not being changed and is also having some default setting left. 
Shall i using some other pins / port for the PTI ?
In WSTK config > PTI i changing the port from 0 to 1 (that i think is the external mini connector) and keeping all the other parameters as for the internal (0) and looking in the network analyzer but i dont getting ans data there :-((

I have "patching" one LIDL / tuya LED Stripe controller with on MG21 chip (the same as Sonoff is using but mush better made and it was not locked so i have dumping the flash) and its looks  being very stable then running as coordinator for one IKEA remote so i very sure the large problems with IKEA battery draining was being fixed with EZSP 6.7.9.0 for all devices but Sonoff is having so bad hardware that is miss tuned and cant  keeping one stable communication with sleeping end devices but is working OK with routers that  can reseeding and having the radio fine tuned all the time.

One thing then using EZSP to do normal sniffing is that EZSP 6.7.9.0 is parsing all IEEE 802.15.2 transaction and the 6.9.2.0 is only parsing "data request" (SED pulling its parent for data) and assumed Ack to it self (All Acks is without only PAN-ID and only sequence number) so its not possible seeing if Ack is failing in one transaction that i like to see so i is  using 6.7.9.0 for doing sniffing with wireshark.
I think its have to do that MG22 is having its own CPU for the radio and its not bringing much sending the low level ack to the stack but it can being very interesting then hunting radio problems.

The EFR32MG1P is little problem loading sample code on then its dont have so much flash then like have CLI and debug printing in console but i have doing on Control bridge from on "minimal example" and can sending commands from it to groups and individual devices and also receiving commands if have binding clusters to the group i like getting information from.

By the way the Easter Bunny must having problem getting all the way to the north but i still waiting for it :-))

Have fun, taking care and have great experience with you systems you is making down under !!

Thanks for feedback and hints.

Mvh  Mattias

MattWestb, 2021-05-10:
Long time no see (or more correct never seen) Gary !

I was nearly giving up getting external PTI working on the WSTK but its working but in one very disappointment way.
If power the WSTK off and removing the radio board on it and power it on its enabling external PTI and v-com.
The enabling signal is coming from the board controller and is common for both in/outputs.
The board controller have 2 PTI lanes and the debug board is hardware fixed to 0 and the onboard radio board is default using the same PTI channel but can also using number 1 is looking on the schematics.
If i understand the hardware design it shall being possible using both internal PTI on channel 1 and the external on channel 0 if the firmware was allowing it and software switching witch or both the Network analyzer is using.
The v-com is little more tricky that is using the same enabling signal as the external PTI but if the radio board is selecting no v-com the UART signal is routed to the external header and it shall being possible using the mini simple connector for v-com also if the radio board on the WSTK is being used with PTI.

Do you have some  experience with PTI thru  mini simplicity connector ?

I was cooking one  router firmware for Billy module that working in real but its little tricky using on short cable for button and grounding it on the module shield then its detecting then i touching it with fingers but  the principle its working OK and end devices is doing pull control check ins and i was adding the pull control cluster in the firmware but no GP proxy.

I was positing it for Headda but hi looks only liking MG21 devices for the moment but one simple router on one MG21 with 1024 kB flash and 96 kB ram is more then over kill im my eyes.

I was getting 2 new "IKEA family" packs (new 4 button remotes with CWS3 bulb both with original Silabs MG21 modules) and IKEA have not protecting the firmware or the bootloader on the module only signing the OTA files as they have done on the MG1 devices = full BONGO for DIY  things :-))
The remote N2 (E2000/2001) looks little empty inside:
[image]
Main flash and data dumped !!!
And the CWS3 (Color White Spectrum = RGB + CT) is having getting color loop that was missing in the MG1 devices. 

IKEA is releasing WW3 (Warm White) and WS3 (White Spectrum = CT) (3 is the generation 3 = MG21 code name Zingo) all with original Silabs modules so its good that Silabs have open 2 new facility's for manufacturing the MG21 modules the last half year then IKEA is needing some millions of them for getting there new products on the market all over the world (some of the devices is over 3 years finished hardware if looking on FCC papers so i think it was one very tuff making on deal with Silabs (in the end for Silaabs but selling very much modules to one customer that is member of Zigbee alliance board)).
What do you IKEA is paying for one original Silabs MG21 module 1.5 $ ?

Back to Simplicity Studio 4 and 5.
Its looks it dont like running it self or SDK on external media in windows (or perhaps need HPFS as file system and not exFAT as i is having on all my external medium).

If having SS 4 and 5 on disk C (one internal M2 SSD) with short path (Ex SSv4) its working OK but is not so fast and is not crashing then doing crazy things.
I was trying installing older SDK on external disks but its crashing all the time.
That is sad then in the old SDKs have more good examples that i can using for doing experiments.

I was finding in SS5 its one EZSP with GP support.
Its have 2 more parameters and some extra commands implanted.
Its also using NVM3 instead for SSIM2 for key storage and its possible compiling for MG1 devices if reducing the NVM3 thaat is default stetted very high in SS5 examples.
I have asking ZHA devs testing compiling it for MG1 devices and defining "standard" parameters that is working well with old MG1 devices and perhaps also EM35X coordinators if its coming so long with GP enabled NCP on old devices.

One last picture for you: How to abusing Sirlabs original MG21 modules
[image]
The new IKEA remote hocked up and dumped the MG21 module :-))

All the best for you and take care and stay safe !!

Mvh  Mattias

MattWestb, 2021-06-19:
### Spamming time !!!

Have upgraded Si5 and have finding the first bug :-))
Then first doing the apps and restarting Si5 and then installing six ten all looks OK and no more updates.
Making one new project and the new GCC 10 is preferred  end selected.
Building the project and its using GCC 7.2 :-((
Looking in tool changes the GCC 10 is preferred but not installed.
Looking in installed component and no updates and looking on tool changes and the GCC 10 is not installed. Installed and its working OK.

New bootloader for MG22:
```
Gecko Bootloader v1.12.00
1. upload gbl
2. run
3. ebl info
BL > 
```
And one six ten Zigbee Control Bridge:
```
stack ver. [6.10.0 GA build 169]
```
Its one interesting fix not in EZSP 6.10.0.0  but in Gecko Platform 3.2.0.0 GA.
- 697483 Fixed bug affecting CTUNE for HFXO on Series 2 devices.

That can being somthing for the "Sonoff ZBB problem" that tasmota, ZHA and Z2M is having in combination with the bad hardware they is using.

Next up for six ten NCP is Billy and also my tuya / LIDL module with MG21 and testing how its working !!

I think its expected with some new undocumented futures (bugs) is shoeing up then changing tool change but hopefully Silabs is catching and fixing them fast.
EZSP 6.9.2.0 was running pity OK and i hope six ten is being better also for old devices that is not longer supported in the new SDK.

Hope all is well with you in the dark and cold down under here is around 35°C and no ice bears and i have getting the first covid vaccine (Moderna) so very likely its being holy day in September in southern Spain :-)))

All the best from the north and take care and being safe !!

Mattias

PS: You have business with some Swedish companies like IAR AB . Assa - Abloy AB (they was in the same concern and is having my old VD from Securitad teckink AB) but you dont have one good Swedish car like Volvo V90 !!!!

grobasoz, 2021-06-27:
Hello Mattias, 
Always good to hear from you in the north :) Sorry for a long delay in replying - we have "winter" holidays in the south now so I get to look after annoying teenage children :(
I also have new products to work on for customers (always demanding of contractors!) and now am working with Wi-SUN development. Then we have to get going with Matter products... lots of learning.

Something changed with Zigbee2MQTT so I had to do some modifications to my Zigbee Coordinators that support Z2M - but all working now :)

Finally my stock of Zigbee modules has still not arrived. I am sure the supplier sold them to the highest bidder or biggest company :( Small developers seem to be "forgotten" :( So still no Zigbee stock to send you - sorry :(

I may need to get some IKEA battery products as it seems the battery issue has not been solved? A strange thing happened on my systems (HA/Z2M and Alexa), all IKEA lights lost their connections? My lights all still worked (and so did Sonoff and Xiaomi parts) but IKEA = gone? Had to re-connect from factory settings?

Sounds like you are having great fun "cooking" new firmware in the North. Let me know if you need any help! 
Stay safe, Southern Gary :D

PS - I did some work with Assa Abloy a long time ago - even went to visit them in Norway :) Volvo too expensive in Australia :( Was looking at XC40 last year but Kia Sportage was 1/2 the price and 7 years warranty!!!

MattWestb, 2021-06-27:
Hi Gary and very nice hearing from you in the dark cold south with out the sauna days and nights here in the north !

Your first large problem i cant helping you with then im not so patient and better letting other having problems with children ;-)))

I like the concept of Matter have on IPV6 network and one Zigbee application over it then its opening large possibility only problem getting the security and redundancy working OK and still getting it user friendly.
I hope the community is jumping on the train but its needs more cheep devices for trigger the demand for it.

You stock problem i understanding the 2 of promoted Zigbee alliance companies is vacuum cleaning the market for the moment, tuya and IKEA have demands for mega shipments for getting there new products on the market.

And its one global sink of semiconductor for the moment. Volvo Cars, Volvo Trucks and Scania trucks is halting there production of lorry and cars because they not getting needed semiconductors.
I hope you can getting your order delivered so you can doing little hardware work and not only leaning new system and how to getting them working (and hopefully not forgetting how the old classic Zigbee was working) !!

IKEA router devices is hitted of one bug in the Zigbee stack that if fixed in 6.7.7.0 (parent announce can locking the buffet and must being repowered) and they is still responding on broadcast but not unicast and cant sending status to the coordinator. Reddit is writing they is digging in it. I think IKEA like going with EZSP 6.10.0.0 for fixing many bugs that is in there products but the 256k flash (in the old module) is making it nearly impossible getting the ZB3 and app in the flash.
The bug is hitting all EFR32 devices so also tuya, new Philips HUE, Sonoff  and so on shall having  the same problem.

Xiaomi (NXP) and old Sonoff (CC-2530) is not having the "IKEA Router bug" but newer Sonoff (router devices) can being triggered of repowered devices sending parent announce (circuit breaker on off for 1/3 devices = 50% devices is crashed one ZHA user have tested).

If you IKEA routers is coming back after repower then its the "IKEA bug" if not then its some other magic that is happening. Losing network parameters or security frame counter corrupt ?

EZSP 6.10.0.0 (or more correct SDK 3.2) is fixing one CTune problems in MG21 devices that can being the last problem with the "Sonoff" problem or is it only very bad hardware they have made / using so no need for you baying old IKEA remotes that you dont have any use off !!
The problem for my is that killing one battery in 48 hours = doing OTA in loop all the time so somthing is very strange but its looks blocking the coordinator having direct children (parameter to the stack) is forcing them using router looks working OK.
I is using one LIDL LED strip controller (tuya RGBWW) with MG21 and one NCP with GP and its very stable with 5 different IKEA remotes and some routers for testing.

SDK 3.2 is little more buggy then the old 3.1 and 2.7. If dong one change in HAL and saving, generating and compiling its oft not working and complaining of missing rules so need doing project clean build and its working.

The new XC40 is little small but looks very nice but not for my $ :((
But Volvo is having one high time of selling cars so somthing they must have doing right or is BMW and AUDI have making somthing very wrong the last years (its the large competitors in the segment).

Thanks for feed back and its always great getting new for the south and hope your projects its going well so you can doing some $ and not only living on sun and water (without bred) !!

All the best and take care !

Northerner Mattias

MattWestb, 2021-12-14:
Hey MAN how did you cooking the MG22 NCP ??
I have not finding any NCP in SS5 for my MG22 modules :-(((

I was thinking cooking one EZSP 6.7.10.0 and testing little and see if its working OK but i think its no great use case for it and better running on the 6.7.8.0 that is stable in production systems.
Also 6.10.X is possible but i think its little heavy for old MG1 chips.

Andy thinking ?

Great greetings from the north !!

grobasoz, 2021-12-14:
Hi Vienna - Cold enough yet? 
We are hot and wet in Queensland now :) No Covid either but internal borders opening so expect it soon :(

Yes, took a while to work out how to build MG22 NCP and test on RPi - as mentioned on the forum, create a project for MGM22 and then change it to EFR32MG22 manually in the project file.
EmberZNet 6.10.3 has quite a few bug fixes and seems Ok so far so I have started upgrading to it on github for people wanting to "play". I should put a disclaimer in the Readme :p
I haven't built for EFR32MG1x parts yet... takes time to connect a test system for each chip...
Stay Safe - Merry Christmas :) Gary in the South.

MattWestb, 2022-03-12:
Hey Gary hope you dont have getting wet feats the last weeks !!

In the north we have problem with Tsar PT-in making the life bad :-(

I have start looking on the Silabs RCP and have compiling one working for tuya MG21 module and also one EZSP 7.0.1.0 that looks working OK.
[image]
But i have problem making one for Billy EZSP.
Wath i knowing is the IKEA module needing DC-DC bypass in the hardware setting for working OK and i have finding it on GSDK4 but i think its not working as intended (many parts is not includes / installed / stetted up in the demo apps in GSDK4).
Also you have customize the `Address Table Size = 100` that looks not being one standard option in GSDK4 and need being added / installed for changing the parameters.
Do you have some thinking how getting MG1P devices working with 7 or have Silabs braking  the support for the old chips ?
Also how to twerking the parameters so i can getting it working OK in ZHA.
I was liking six-ten and the way GUI was looking but i have not getting all in place with version 4 ;-/
Was getting all my modules working with cooked six ten also the tuya ZBGW version that you was cooking for out ZBGW :-)))

I have getting the tuya module ruing with RCP firmware in ZHA and adding one new addon  so using "software" EZSP and also have one Open Thread boarder router up and running at the same time on the module but dont have any more modules i can installing OT routers on so yesterday i was to IKEA and baying 2 very cheep lights (9€) with the original Zilabs MG21 module and have getting the lights disassemble easily and shall doing the de-soldering and dumping the firmware and soldering my cables so i can using it as one original Silabs module but without the extra flash but all pads is like Silabs reference board and can using all function also PIT if connecting with my WSTK. 
The "code name" is IKEA Markus" = on very nice working share that many programmer is using in office or at home.

Hope all is well with you then the summer is more or less over down there and you have getting more EFR32 modules so you can doing little Zigbee / BT / Thread and Matter work for your customers and keep getting more orders for your products and service !!!

Here is the last cold week but the sun is warming very nice and soon is it summer here !!

All the best winches and the best for our great firmware cooker down under !!

Mvh  Mattias

grobasoz, 2022-03-20:
Hi Mattias, 
Thanks for the message and sorry for the delay - I have been helping my 2nd child into University and a new apartment, difficult as moved to a new city and there have been floods in our state! Roads were cut for many days! Sad to hear of trouble up north - some crazy people around :(
Good to see you started cooking now :) Great progress - soon I can retire :D 
I have started on V7 but too many things to fix in the release so I can't use it for my customers - they get grumpy if you send them broken code! Also as you say quite tricky to navigate...
I have some more EFR32 modules now so can make some more Zigbee products. As I told Hedda many have been sold locally for industrial use. Unfortunately EFR32 chips are available at Mouser but expensive ($$$)
The main thing is getting stable code on the new releases. GSDK4 is very new - some issues, EmberZNetV 6.10.3 also issues - so I stick with EmberZNet 6.8.01 for now.
Stay safe! Enjoy the new spring and summer :) Regards Gary in the South!

MattWestb, 2022-04-02:
Thanks for replaying and its great herring for you with or without wet fetes !!
Then reading little more its looks like Gecko SDK 4 is only supporting HW flow control but its one large mess than many key function is not in place or not working.
I was doing one more try with FR32MG1P RCP for my Billy and only doing config in pin tool for the com and then DC-DC bypass  in software setting and it was not working ;-(((
Looking little more and was seeing that all 10 soldered cables to the connection pin bar was OK but  the 11th bootloader pin was under the cables and grounded of the RF shield so the chip was booting in bootloader ever time and after putting it on the other side i is getting in ZHA:
```
2022-04-01 16:09:25 INFO (MainThread) [bellows.zigbee.application] EZSP Radio manufacturer: 
2022-04-01 16:09:25 INFO (MainThread) [bellows.zigbee.application] EZSP Radio board name: 
2022-04-01 16:09:25 INFO (MainThread) [bellows.zigbee.application] EmberZNet version: 7.0.2.0 build 0
```
So now Billy is running RCP 7.0.2.0 with Zigbeed and OTBR on the host !!!
Also the OTBR is up and running with one tuya / LIDL MG21 module with RCP firmware have joining the Open Thread network.
[image]

The EFR32MG1P is 100% unsupported but not deleted in GSDK 4 / EZSP 7 then its one 256K device.
But then running RCP its no problems as long Silabs is not breaking the RCP firmware that is working OK (EFR32MG1P is still one reference design in Silabs OT git code with all the radio code in place) and the updated for OT and Zigbeed is made in the general software in the host system and hopefully not being broken in the near future.

The address table i was copy from your settings for MG1P and is in your readmy `NB Route Table Size = 32, Address Table Size = 100` and it was possible setting in EZSP 6.10.x but i dont finding it in 7.X but in the end i think its useless running EZSP 7.x on this old chips / SOC then better running RCP and host Zigbeed if like playing with them or staying on EZSP 6.10.X or lower.

The tuya ZBGW you was cocking one firmware with HW flow control but did you putting in some more parameters like DC-DC by pass for getting it working (ts working great for our users) ?
I like trying making one RCP firmware for it and hoping getting support for network attached RCP (the devs is trying getting it implanted).

Now its only getting the IKEA Markus (Silabs MG210L module) working with RCP firmware but i have getting it working OK with EZSP 6.10.

All the best for you and keep your head cool and your feats dry then the winter is not coming yet down there :-)

Mvh Mattias

MattWestb, 2022-04-10:
## Dam IKEA !!!
Was trying making one RCP for tuya ZBGW that is having one tuya TYZS4 module with EFR32MG1B232F256GM48-D chip and in GSDK 4 is have **ZERO** Zigbee and OT example !!
Was trying using the old IKEA module chip EFR32MG1P132F256IM32 that i have getting up and running but that chip is not  having the pins for HW flow control so its one no go :-(((.

Do you knowing if its one way "porting" the chip hardware config so can compiling one RCP for it ?
Normally its only making one RCP-UART project with the chip and in pin tool config the pins for UART0 and setting up flow control (Billy soft ZBGW hard) and also Dc-Dc by pass in software components and compiling (and not forgetting having one bootloader on the chip).

I think this is "little" political then IKEA **is** the leader of the European Matter group and if they is needing somthing for there system im very sure Silabs is doing it for them and they is baying _some_ chips and modules from Silabs.

IKEA have also one new Zigbee Hub in the pipeline (was seen in CSA certs but was deleted / hidden and no FCC papers is being seen) so i think they is still using the old chip as RCP and is upgrading the Zigbee Controller Bridge to Zigbee 3 (the current is ZLL and EZSP 5.7.2.X) and EZSP 4.X and still going for without coordinator and using touch link for forming the network
.
The RCP is using around 150 Kb flash and i cant see any good using one MG21 chip then its only over kill and is not  bringing and good things only higher price.
Also it can being way i cant getting the MGM210LA22JNF2 module RCP working then its only fast ported and not tested of any "gold customer" that need it in there system that is using it as normal OT devices and not as boarder router in there systems.

Also IKEA have "adjusting" the price for most  Zigbee products with 10-30 % up and i think they is writing all Russian activity's off (i think Moscow or Saint Petersburg was one of the larges store in the wold and they is / was having many there) but "Tsar Vladimir I" is taking all that hi like around here.

Enough of bad things and now the summer is coming and you is getting one nice cold period without the bad tropical nights that is being one killing pain in the AS....

Ans i must  doing more work on my balcony for the Easter for getting it ready for the Ester bunny and the summer :-)))

All the best from the north and have one great Easter down under !!!

Mvh Mattias

MattWestb, 2023-02-11:
Hi Gary.

I was reading that the EFR32MX24 radio can reviving on more channels in some cases.
EFR32MG21 RCP with multi PAN can only using the same for Thread and Zigbee.
I have not looking on the spec but is it technical possible with MG24 hardware tuning 2 radio channels at the same time and piping the frames to the hots / the SOC APP ?

Thanks in advance !!

MW

closed 2024-08-12 by grobasoz