grobasoz/zigbee-firmware #12: Request Zigbee Router mode firmware for EFR32MG1P232GG

texvek, 2021-04-27:
Hello Gary.
We saw issue with request: https://github.com/grobasoz/zigbee-firmware/issues/6

Can you cooking a Zigbee router mode firmware for  P series - Perfomance Silicon Labs EFR32MG1PxxxF256( Marked as EFR32MG1P232GG) ?

WE need  UART 115200 baud SW
GPIO TX:PA0, RX:PA1 
bootloader GPIO activation pin PB11.
[image]

SDK 6.7.8 or other

Thank you so much!

grobasoz, 2021-04-27:
@texvek - Can you confirm what you mean by "Zigbee router mode firmware for P series" please?
Do you want Silicon Labs EZSP with the pinout above? **See Attached.**
or
Do you want a Standalone Zigbee "Router"? 

Also I don't have EFR32MG1P to test.
Gary.
[NCP_USW_115k2_F256_678_PA0-PA1-PB11.zip](https://github.com/grobasoz/zigbee-firmware/files/6382831/NCP_USW_115k2_F256_678_PA0-PA1-PB11.zip)

texvek, 2021-04-27:
i can send to you EFR32MG1P for test.

MattWestb, 2021-04-27:
Do you have one working SWD probe then you can downloading Simplicity commander ([Download page for standalone install](https://www.silabs.com/mcu/programming-options)) and and reading the chip model.
Sillabs have released some chips with strange writing (that is not possible encoding with there matrix) that cant being identified (mostly on there own modules) and its more secure getting it right then having the "real chip" model.

The EFR32MG1P is generic and only radio bands and flash size is different but is normally working OK if not braking the flash limits on the chip.

One example of tuya /LIDL ZBGW:
[image]

MattWestb, 2021-04-27:
So its one EByte-E180-Z120B mudule little  stripped: https://fccid.io/2ALPH-E180ZG120B

And ZHA have doing EZSP firmware for it https://github.com/zha-ng/EZSP-Firmware/tree/master/EByte-E180-Z120B and I think Gary already have doing one before (for my x- country man H).

texvek, 2021-04-27:
> So its one EByte-E180-Z120B mudule little stripped: https://fccid.io/2ALPH-E180ZG120B
> 
> And ZHA have doing EZSP firmware for it https://github.com/zha-ng/EZSP-Firmware/tree/master/EByte-E180-Z120B and I think Gary already have doing one before (for my x- country man H).
> 
> Edit. The chip is one efr32mg1p232f256gm48 from FCC Manual.

Great information. Will try tomorrow this

grobasoz, 2021-04-27:
@texvek - No need to send me any hardware - just test the firmware I attached using your hardware?
@MattWestb - Yes, programming with the the SWD works BUT won't work using XModem serial upload. The chip ID is checked for a match.

texvek, 2021-04-29:
Hi all
1) I try to use ebyte FW with esp module with tasmota project 
EByte-E180-Z120B
Log:
00:00:16.769 ZIG: Resetting EZSP device
00:00:20.266 RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
00:00:25.879 RSL: RESULT = {"ZbState":{"Status":55,"Version":"6.7.8.0","Protocol":8,"Stack":2}}
00:00:25.882 RSL: RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
00:00:33.995 RSL: RESULT = {"ZbState":{"Status":56,"IEEEAddr":"0x90FD9FFFFE672E4F","ShortAddr":"0x0000","DeviceType":1}}
00:00:34.104 ZIG: Subscribe to group 0 'ZbListen0 0'
00:00:34.160 RSL: RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
00:00:34.161 ZIG: Zigbee started
00:00:34.162 ZIG: No Zigbee device information
00:00:34.162 ZIG: No Zigbee device data

Total- not work :)

texvek, 2021-04-29:
2) FW from Gary 
Ncp 

00:00:00.001 HDW: ESP8266EX
00:00:00.026 CFG: Loaded from flash at FB, Count 4
00:00:00.029 QPC: Count 1
00:00:00.033 SNS: Hardware Serial
00:00:00.035 Project tasmota Tasmota Version 9.4.0.2(zbbridge)-2_7_4_9(2021-04-26T15:43:11)
00:00:00.035 ZIG: Resetting EZSP device
00:00:00.186 WIF: WifiManager active for 3 minutes
00:00:00.692 HTP: Web server active on tasmota_DF11F1-4593 with IP address 192.168.4.1
00:00:06.961 QPC: Reset
00:00:16.862 ZIG: Resetting EZSP device
00:00:21.512 RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
00:00:21.566 RSL: RESULT = {"ZbState":{"Status":55,"Version":"6.7.8.0","Protocol":8,"Stack":2}}
00:00:21.567 RSL: RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
00:00:21.678 WIF: Connecting to AP HUAWEI-X343S as tasmota_DF11F1-4593 ...
00:00:22.119 RSL: RESULT = {"ZbState":{"Status":56,"IEEEAddr":"0x90FD9FFFFE672E4F","ShortAddr":"0x0000","DeviceType":1}}
00:00:22.133 ZIG: Subscribe to group 0 'ZbListen0 0'
00:00:22.139 RSL: RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
00:00:22.140 ZIG: Zigbee started
00:00:22.140 ZIG: No Zigbee device information
00:00:22.141 ZIG: No Zigbee device data

total - not work with tasmota

Also I try to use this module with both fw in home assistance. Not work , in log “Error while starting zigbee-herdsman. Failed to start zigbee. 

Thank you for support. Can you push me in needed directions for upgrade my skills? ;)

MattWestb, 2021-04-29:
```
00:00:34.161 ZIG: Zigbee started
``` 
= Tasmota ZBB have starting = OK
```
00:00:34.162 ZIG: No Zigbee device information
00:00:34.162 ZIG: No Zigbee device data
```
You have no device paired to the coordinator so it cant finding any devices.

Trying  pairing on device and look what tassmota is liking it or not.

texvek, 2021-04-29:
11:25:23.152 ZIG: Zigbee started
11:25:23.153 ZIG: No Zigbee device information
11:25:23.154 ZIG: No Zigbee device data
11:25:33.078 RSL: RESULT = {"Reset":"Reset and Restarting"}

It mean that esp chip resetting to default settings. Ok. Thanks 
Will try again

closed 2021-05-03 by texvek