grobasoz/zigbee-firmware #13: EFR32 Series1 firmware details

sharathbk1912, 2021-07-21:
Hi,

I was looking to flash E180-ZG120B-TB.

The firmware given here contains 2 version combined.s37 and .hex.
I want to flash using openocd and hence not sure if openocd today supports .s37 format .
What is the exact difference between the 2 firmware and do we have a combined version in .hex format ? 
This would simplify the firmware flashing through openocd 

Details on hardware are as below :
https://modkam.ru/?p=1889

Thanks and Regards
Sharath

grobasoz, 2021-07-21:
Hi Sharjah,
You may need to convert .s37 to .hex manually. I seem to recall someone detailing the steps on the internet for OpenOCD. Gary

sharathbk1912, 2021-07-22:
Thanks Gary for the reply,

I was able to covert to s19 format and flash using openocd using rpi 
Thanks for the firmware and support 

Just for sake of curiosity what is the difference between the combined and regular firmware required for 1st upload .

Regards
Sharath

grobasoz, 2021-07-22:
On Series 1 devices, three bootloader images are generated into the build directory: a main bootloader, a main bootloader with CRC32
checksum, and a combined first stage and main bootloader with CRC32 checksum. The main bootloader image is called <projectname>.s37, the main bootloader with CRC32 checksum is called <projectname>-crc.s37, while the combined first stage image +
main bootloader image with a CRC32 checksum is called <projectname>-combined.s37.
See  [https://www.silabs.com/documents/public/user-guides/ug266-gecko-bootloader-user-guide.pdf](url)

sharathbk1912, 2021-07-22:
Thanks gary!

closed 2021-07-22 by sharathbk1912

grobasoz, 2021-07-22:
You are welcome :)

sharathbk1912, 2021-07-22:
Gary one last clarification 

I am using this NCP_UHW_MG1B232_678_PA0-PA1-PB11_PA5-PA4.gbl (from https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32%20Series%201/EFR32MG1B-256k/NCP). I understand this is a EZSP firmware for zigbee **coordinator** .

Is there also a **router** firmware for the same ? 

Thanks 
Sharath B K

grobasoz, 2021-07-22:
The firmware is NCP (Network Co-Processor) and is a slave device. The application that controls the NCP determines whether it is a Coordinator or Router.

sharathbk1912, 2021-07-22:
Okay Got it! 
This is quite different from from ti cc series. 

So to act as a router , we shall need a controller(eg ESP) communicating to this . 

Thanks and Regards
Sharath B K

grobasoz, 2021-07-22:
TI uses ZNP that has a higher level ZigBee implementation supporting coordinators, rout3rs and 3nd devices… the control is managed by that host but once settings etc have been done, the ZNP takes over.

NCP requires constant control by the host.

Will Silabs development kits,  you can also create standalone coordinators, routers and end devices… eg gateways, lights and switches.

Regards,
Gary.