grobasoz/zigbee-firmware #15: [REQUEST] em3587 Bitronvideo BV 2010/10

NilsOF, 2021-11-18:
Actually, the Bitronvideo BV 2010/10 is already working using the first commit for etrx3587-lr.
Unlike etrx3587-lr, the Bitronvideo BV 2010/10 uses PC6 for PA/LNA while etrx3587-lr uses PC5.

#14 is what led me to try the em3587 firmware.

IOW, I used https://github.com/grobasoz/zigbee-firmware/raw/af7c35ea8d580152eb9853af1d3fab91bef3b5d4/EM3587/NCP_USW_EM3587-LR_678-115k2.ebl
and it works on my Bitronvideo BV 2010/10

Hedda, 2021-12-01:
Nice! For reference, did you manage to upgrade your Bitron Video/Smabit BV AV2010/10 USB-Stick using the Silabs Simplicity Studio, or did you use the docker image and scripts by walthowd, or perhaps Elelabs Firmware Update Utility?

https://github.com/walthowd/husbzb-firmware

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/

Firmware for Bitron Video/Smabit BV AV2010/10 USB-Stick has also been asked for and directly or indirectly discussed here:

https://github.com/zigpy/bellows/issues/348

https://github.com/Koenkk/zigbee-herdsman/issues/441

gromgsxr, 2021-12-15:
I managed to upgrade my BV AV2010/10 USB-Stick with the above linked f/w today i used the walthowd docker image and a bit of editing the npc.py

Hedda, 2021-12-16:
> I managed to upgrade my BV AV2010/10 USB-Stick with the above linked f/w today i used the walthowd docker image and a bit of editing the npc.py

Great! Maybe also report this and change to issue tracker on walthowd repo -> https://github.com/walthowd/husbzb-firmware

NilsOF, 2021-12-18:
@Hedda I used the integrated updater in openHAB

Hedda, 2022-01-03:
FYI, EmberZNet 6.7.10 for EM3587 has now also been made available by grobasoz

https://github.com/grobasoz/zigbee-firmware/blob/master/EM3587/NCP_USW_EM3587_6710-LR.ebl

EmberZNet 6.7.10 has been confirmed working by Elix-g in a discussion about flashing here:

https://github.com/walthowd/husbzb-firmware/issues/38

Again,  this requires defining a baud rate of 115200 instead of 57600 which I believe is the baud rate on the default firmware.

richard-gravy, 2022-03-12:
Just wanted to write that I was having a weird problem using ZHA, where the stick would pick up a device for about 1 minute before losing all communications, making it useless for automation.  Came here and found your instructions, updated the firmware to 6.7.10 and now all works perfectly! Cheers

gromgsxr, 2022-03-12:
That's the exact reason I upgraded mine glad it helped someone out

grobasoz, 2022-03-13:
Thanks for the feedback... much appreciated :)

closed 2022-09-15 by grobasoz