grobasoz/zigbee-firmware #16: [REQUEST] EmberZNet 6.10 / 6.10.x.x Zigbee Coordinator firmware images?

Hedda, 2021-11-25:
Silicon Labs EmberZNet 6.10.3.0 GA should really be mature and stable enough now as only contain bug-fixes since 6.10.0.0 GA.

Silabs EmberZNet Zigbee 6.10.0.0 was released on the 16th June of 2021 and newer EmberZNet 6.10 updates since then:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.0.0.pdf

EmberZNet 6.10.2.0 was released on the 8th of September of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.2.0.pdf

EmberZNet 6.10.1.0 was released on the 21st June of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.1.0.pdf

EmberZNet 6.10.3.0 was released on the 13th October of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.3.0.pdf

FYI, Elelabs has now released EmberZNet v6.10 firmware for their EFR32 based adapters (based on EFR32MG12):

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/blob/master/data/EFR32MG13/ELE_MG13_zb_ncp_115200_610_211112.gbl

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/31f8e9c09d612734edbe41619240162fe595ecc5

### Added

- 6.10 Zigbee Firmware for ELU013/ELR023 (just keep up with the latest EmberZnet SDK)

Their firmware update tool has also been updated with a few new features:

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/721019cb542b273159372357a30251330e7c8d51

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/e9b3fe1dd73b46fef1629a7a54ca0a257c1388ae

### Changed

- `ele_update` parameters changed. Now the function only updates to latest version of available Zigbee or Thread firmware
- `probe` and `restart` functions can now detect Thread adapters as well

Hedda, 2021-11-25:
> FYI, Elelabs has now released EmberZNet v6.10 firmware for their EFR32 based adapters (based on EFR32MG12):
> 
> https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/blob/master/data/EFR32MG13/ELE_MG13_zb_ncp_115200_610_211112.gbl
> 
> [Elelabs/elelabs-zigbee-ezsp-utility@31f8e9c](https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/31f8e9c09d612734edbe41619240162fe595ecc5)

FYI, Sonoff has now also released Silabs EmberZNet 6.10 (v6.10.3) firmware for ITead Zigbee 3.0 USB Dongle (EFR32MG21 based):

[https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle](https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle)

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/ncp-uart-sw_v6.10.3_115200.gbl

xsp1989 also announced the good news is that a router firmware "will be released soon" for the same USB dongle here:

[https://github.com/xsp1989/zigbeeFirmware/issues/18](https://github.com/xsp1989/zigbeeFirmware/issues/18)

grobasoz, 2021-11-25:
@Hedda - Thanks for the updates :) 
EM35x and EM358x aren't supported in 6.10 so I'll concentrate on EFR32? 
Seems Elelabs and xsp1989 have their modules covered so what devices should be targeted?

Hedda, 2021-11-26:
> EM35x and EM358x aren't supported in 6.10 so I'll concentrate on EFR32? 
> Seems Elelabs and xsp1989 have their modules covered so what devices should be targeted?

@grobasoz Did you not have plans before to make and sell your own EFR32MG21 based USB adapter/dongle/stick? ...if you are still considering that maybe even base it on the latest generation chip, which I believe is the EFR32MG23?

A lot of people are still buying and using EM35x and EM358x based adapters like the [Nortek GoControl QuickStick Combo Model HUSBZB-1 (Z-Wave & Zigbee Ember 3581 USB Adapter)](https://www.nortekcontrol.com/products/2gig/husbzb-1-gocontrol-quickstick-combo/) and the [Bitron Video/Smabit BV AV2010/10 USB-Stick (a.k.a. ZigBee Funkstick)](https://bv.smabit.eu/index.php/smart-home-produkte/zb-funkstick/) because they are the first USB adapters that show up in a quick search on Amaazon and are also still the only Silabs USB adapters which are widly available and on paper works out-of-the-box with [Home Assistant's ZHA](https://www.home-assistant.io/integrations/zha) and [OpenHAB's Zigbee Binding](https://www.openhab.org/addons/bindings/zigbee/), at least when considering that many users that are new to Zigbee seems to be start by looking for a USB-adapter that they techically do not need to upgrade the firmware on then buying their first Zigbee adapter. It is usually after they bought the adapter that they find out that is based on an old chip and/or that they will need to upgrade the firmware on it.

Therefor you should know that there is still a demand for a good EFR32MG21 based USB adapter/dongle/stick similar to [ITead Zigbee 3.0 USB Dongle Model 9888010100045](https://www.itead.cc/zigbee-3-0-usb-dongle.html) but properly engineered and tuned (maybe with a ceramic chip antenna) + with a good looking enclosure and packaged in a professionally made retail box so can be nicly displayed to be sold in brick-and-mortar stores as well as online, (instead of just shipping a it like an evaluation board in anti-static bag like ITead does with its EFR32MG21 based USB donge). ITead did a much better job on all that with its new TI CC2652P based USB-dongle but with that not being Silabs based it is not compatible with [OpenHAB's Zigbee Binding](https://www.openhab.org/addons/bindings/zigbee/) and other Zigbee implementations which are only compatible with Silabs chips (or work better with Silabs chips).

Quick solution could be to just make and sell a USB-adapter with a Silabs MGM210P radio module on the board and add a case.

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules

Anyway, I believe most commonly used by Home Assistant community are the ones listed on ZHA list of knows working adapters:

https://www.home-assistant.io/integrations/zha#known-working-zigbee-radio-modules

I also believe most of those either directly link or indirectly refer to the script and repo by @walthowd for upgrading firmware:

https://github.com/walthowd/husbzb-firmware

So the most commonly EFR32 adapters used for Zigbee today are probably:

* DIY gateway projects/products based on Silabs MGM210P (or MGM210L) radio modules, like [Tube's EFR32 Gateway](https://www.tubeszb.com/product/efr32_gw_2/6?cp=true&sa=false&sbp=false&q=false&category_id=2).
  * Firmware should be available from DIY projects like https://github.com/tube0013/tube_gateways ?
* The "latest" [Elelabs Zigbee USB Adapter model ELU013](https://elelabs.com/products/elelabs-usb-adapter.html), also sold as [POPP ZB-Stick](https://www.popp.eu/zb-stick/) in Europe (EFR32MG12 based?).
  * Official firmware available via https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/
  * Unofficial firmware available via https://github.com/zha-ng/EZSP-Firmware
* [Elelabs Zigbee Raspberry Pi Shield / HAT](https://elelabs.com/products/elelabs-zigbee-shield.html), also sold as [POPP ZB-Shield](https://www.popp.eu/zb-shield/) in Europe (EFR32MG12 based?).
  * Official firmware available via https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/
  * Unofficial firmware available via https://github.com/zha-ng/EZSP-Firmware
* [ITead Zigbee 3.0 USB Dongle (EFR32MG21) Model 9888010100045](https://www.itead.cc/zigbee-3-0-usb-dongle.html)
   * Semi-official firmware available from xsp1989 (believbe to be engineer from @CoolKit-Technologies who ITead contracts?)
* [ITead Sonoff ZBBridge](https://www.itead.cc/smart-home/sonoff-zbbridge.html) based on CoolKit "SM-011 V1.0" radio module (EFR32MG21) but require signed firmware.
  * Signed firmware available at https://github.com/arendst/Tasmota/blob/development/tools/fw_SonoffZigbeeBridge_ezsp
* Other hacked Chinese USB-sticks and gateways that are also based CoolKit "SM-011 V1.0" radio module (EFR32MG21), like [ZB-GW03](https://github.com/zigpy/zigpy/discussions/819) sold under different brands.
  * [https://github.com/zigpy/zigpy/discussions/586](https://github.com/zigpy/zigpy/discussions/586)
  * [https://github.com/zigpy/zigpy/discussions/819](https://github.com/zigpy/zigpy/discussions/819)
* EByte-E180-Z120B (uncommon since have to be flashed via JTAG but still used by some DIY projects as its is cheap).
  * Unofficial firmware available via https://github.com/zha-ng/EZSP-Firmware/

grobasoz, 2021-11-29:
@Hedda - thanks again for a detailed and comprehensive reply to my query.

I made 400 USB-Zigbee based on EFR32MG21 and they sold almost immediately. I did chat to Elelabs about a possible joint venture but nothing came of that. Unfortunately it's difficult to source stock of EFR32 chips and radio modules at the moment so another round of production is not on the cards... 

EM35X, EM358X - I will keep updating the repository where possible.
EFR32MGXX - I will keep updating the repository as time permits.

closed 2022-09-17 by grobasoz