grobasoz/zigbee-firmware #17: [REQUEST] EmberZNet 6.7.10.0 (6.7.10.x) Zigbee Coordinator firmware images?

Hedda, 2021-11-26:
FYI; Silicon Labs has released Silabs Zigbee EmberZNet SDK 6.7.10.0 GA bugfix release:

https://cn.silabs.com/documents/public/release-notes/emberznet-release-notes-6.7.10.0.pdf

grobasoz, 2021-11-28:
Thanks @Hedda - seems like a few important issues have been resolved in this release :) I've been using it for a while and will start updating this repository as time permits. Gary.

Hedda, 2021-12-20:
@grobasoz While don't have an EM3581 based adapter myself I like would to request prioritized a new 6.7.x firmware update for it.

The reason is a lot of people got Nortek HUSBZB-1 combo stick and upgrade via -> https://github.com/walthowd/husbzb-firmware

Copy of the same request opened here -> https://github.com/walthowd/husbzb-firmware/issues/36 (the latest release for EM3581 is still EmberZNet 6.7.8)

I did note however that you have a @walthowd folder with specified settings so guess that might need separate releases for him?

https://github.com/grobasoz/zigbee-firmware/tree/master/EM3581/%40walthowd

Specific Settings for this folder

CONFIG_APS_UNICAST_MESSAGE_COUNT: 20
CONFIG_MAX_END_DEVICE_CHILDREN: 32
CONFIG_SOURCE_ROUTE_TABLE_SIZE: 200
CONFIG_ROUTE_TABLE_SIZE: 16
CONFIG_ADDRESS_TABLE_SIZE: 16

grobasoz, 2021-12-21:
I have updated the EM3581 versions including walthowd's specific release. PC5 is used for TX Active and PC6 for nTX Active.

sreknob, 2021-12-22:
Thanks a lot Gary for working on this!
I've tried the generic 6710 and the 115K2-W variant without success so far on my HUSBZB-1.
I get the banner back when starting upload.

```
root@677a98369618:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB1 115200 True False 
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "6.7.8-373"}]}
root@677a98369618:/tmp/silabs# ./ncp_new.py flash -p /dev/ttyUSB1 -f NCP_USW_EM3581_6710-115k2-W.ebl
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image... 
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\n' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'S' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'e' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'i' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b' ' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'u' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'p' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
```

From my recollection, there we comments about this being a [bootloader issue last time at 6.7.8](https://github.com/walthowd/husbzb-firmware/issues/12#issuecomment-763027823). Not sure how it was solved.

I haven't tried minicom directly yet...

grobasoz, 2021-12-23:
I still don't have an EM3581 to test on but the EM3581 code can be flashed into an EM3585 using an Ember ISA3 and works fine. The serial bootloader (eg that ncp uses) checks the chip ID so can't be uploaded to an incorrect chip.

sreknob, 2021-12-23:
Thanks Gary. Last time there was the same issue with HUSBZB-1 I believe you fixed it using an older compiler.
I'll see if anyone else can get it to work or I'll try XMODEM but that also had the same result before.

grobasoz, 2021-12-23:
@sreknob  - Thanks for testing - apologies for it not working. I am confused however. I have tried flashing EM3581 code to my EM3587 test setup and it works there as well. The development system is set to the older IAR compiler so that shouldn't be the issue. I'll see if I can source a  HUSBZB-1 for testing.

grobasoz, 2022-01-07:
@sreknob - Apologies for the delays getting to this - I was waiting for a HUSBZB-1 to arrive - and sadly I still can't find stock. I have added a new version to the repository recompiled on the latest IAR Compiler to see if that works. If you get a chance please test on your equipment. Thanks Gary.

sreknob, 2022-01-10:
Hey Gary!
Thanks so much for continuing to work on this. Old defunct hardware is always hard to source :-)

Anyway, whatever it is with your compiler, you have once again found the magic sauce!
Uploaded without problems to my HUSBZB-1.
Just fired Home Assistant up and it's chugging along like usual!

```
pi@raspberrypi:~/husbzb-firmware $ ./ncp-115k2.py flash -p /dev/ttyUSB1 -f NCP_USW_EM3581_6710.ebl 
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image... 
Finished!
Rebooting NCP...
pi@raspberrypi:~/husbzb-firmware $ ./ncp-115k2.py scan
Connecting to.. /dev/ttyUSB1 115200 True False 
Connecting to.. /dev/ttyUSB0 115200 True False 
{"ports": [{"stackVersion": "6.7.10-423", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}, {"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB0", "vid": "10C4"}, {"deviceType": "unknown", "port": "/dev/ttyAMA0"}]}
```

This is great!!

I know that you compiled specific zigbee settings for @walthowd for his repo. I'm not sure what the zigbee settings were on [this latest version](https://github.com/grobasoz/zigbee-firmware/blob/master/EM3581/NCP_USW_EM3581_6710.ebl) but it flashed and is operating at 115K2 no problem on my Home Assistant instance currently.

You're my hero 🥇 

Blake

grobasoz, 2022-01-10:
Thanks for testing and replying Blake - much appreciated :) 
I have recompiled the @walthowd versions but as you mentioned my standard version should run OK with his settings.
Gary.

Mazotis, 2022-01-15:
Tried flashing the latest version on my HUSBZB-1 and I get the same result as @sreknob earlier
`./ncp.py flash -p /dev/ttyUSB1 -f NCP_USW_EM3581_6710.ebl
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image... 
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\n' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'S' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'e' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'i' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b' ' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'u' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'p' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
`
Tried with the W variant as well, and tried flashing 6.7.8 115k baud rate before (I was on 57k), no success. Was the file modified since the last comments ?

grobasoz, 2022-01-15:
@Mazotis - Thanks for the update - I was worried I had broken something again...

closed 2022-01-15 by grobasoz

flufftronix, 2022-06-12:
I was still having these failed upload problems with three HUSBZB-1's that hadn't had their firmware altered since purchase (I think they were running a 5.x firmware?). What ended up working for me, in case anyone else ends up here, was first flashing to 115k2:

1. Use [walthrowd's updater](https://github.com/walthowd/husbzb-firmware) to flash the most recent 115k2 firmware in the repo, ncp-uart-sw-6.7.8.ebl. On each device for me this required [`xonxoff=True` to be commented out in the ncp.py file](https://github.com/walthowd/husbzb-firmware/issues/33). After the firmware is flashed and the device reboots, run `./ncp.py scan` if you'd like; it should hang as it's set up for 57k on a HUSBZB-1 which we're no longer operating at.
2. Edit ncp.py and uncomment the `xonxoff=True` line. Also, as the device is now running at 115k instead of 57k6, comment out line 35 and add `CEL_BAUD = 57600`.
3. Now run `./ncp.py scan`, and the device should show up.
4. Flash firmware from this repo onto the device. I went with [NCP_USW_EM3581_6710-115k2-W.ebl](https://github.com/grobasoz/zigbee-firmware/blob/master/EM3581/%40walthowd/NCP_USW_EM3581_6710-115k2-W.ebl).

And that did it for me. This was in order to run Zigbee2MQTT, which just needed a couple lines added to the config [as indicated in their documentation](https://www.zigbee2mqtt.io/guide/adapters/#not-recommended):

```
serial:
  port: /dev/ttyUSB1
  adapter: ezsp
```

..in Z2M's configuration.yaml.

cvocvo, 2022-09-24:
I've been attempting to update my HUSBZB-1 and have the same errors as noted above. I tried all 3 6.7.10 images (regular + walthowd 56k + walthowd 115k) and have the same issue. I am doing this on Windows, however, I can flash the 6.7.8 firmware just fine.

As noted above and over here: https://github.com/walthowd/husbzb-firmware/issues/36#issuecomment-1009170141 there does appear to be some success doing this, but I can't get it to work.

Any ideas?

Here's my log / error data:
```
PS C:\temp\husbzb-firmware> python .\ncp.py flash -p COM3 -f "C:\temp\zigbee_6.7.10\NCP_USW_EM3581_6710.ebl"
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\n' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'S' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'e' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'i' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b' ' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'u' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'p' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
PS C:\temp\husbzb-firmware> python .\ncp.py flash -p COM3 -f "C:\temp\zigbee_6.7.10\NCP_USW_EM3581_6710-57k6-W.ebl"
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\n' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'S' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'e' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'i' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b' ' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'u' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'p' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
PS C:\temp\husbzb-firmware> python .\ncp.py flash -p COM3 -f "C:\temp\zigbee_6.7.10\NCP_USW_EM3581_6710-115k2-W.ebl"
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
[... 10 lines cut ...]
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
PS C:\temp\husbzb-firmware> python .\ncp.py scan
Connecting to.. COM3 57600 True False
Exception: <class 'IndexError'> COM1
{"ports": [{"port": "COM3", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "6.7.8-373"}, {"port": "COM1", "deviceType": "unknown"}]}
PS C:\temp\husbzb-firmware> python .\ncp.py flash -p COM3 -f "C:\temp\husbzb-firmware\ncp-uart-sw-6.7.8.ebl"
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
Finished!
Rebooting NCP...
PS C:\temp\husbzb-firmware> python .\ncp.py scan
Connecting to.. COM3 57600 True False
Exception: <class 'IndexError'> COM1
{"ports": [{"port": "COM3", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "6.7.8-373"}, {"port": "COM1", "deviceType": "unknown"}]}
PS C:\temp\husbzb-firmware>
```
[image]

grobasoz, 2022-09-25:
@cvocvo - Sorry I am away for a few days on site - not able to assist you at this stage.

localboast, 2024-04-26:
Man I tried everything to get the upgrade to work with my Nortek husbzb-1. I am able to flash Walt's version to 6.7.8-115k but I'm stuck here. I modified the ncp.py file for 115200 in both spots but no matter what 6.7.10 version in @grobasoz repo I try to load I get the xmodem errors reported above. Bummer.

Happy that other folks were able to get it going, wish I could say the same. I read through the relevant topics here on GitHub numerous times, and think I tried it all (including 57600 variants).