grobasoz/zigbee-firmware #18: Question about ZBBridge firmware 6.10.3

r-vit, 2021-12-07:
I tried to flash NCP_USW_115k2_ZBBridge_6103.gbl firmware using standard Tasmota firmware update interface, but every time the process failed. At the same time flashing the older version from https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp worked well. 

Is there some trick to flash 6.10.3 to ZBBridge?

MattWestb, 2021-12-07:
Sonoff ZBB is needing one signed firmware or its not flashing the file on the device so only original firmware is working from SonOff (if not have flashing it withit  with SWD).

r-vit, 2021-12-07:
Ah, ok. I do have a J-Link adapter, so I guess it will be the only way to flash the newer firmware.
Thank you.

MattWestb, 2021-12-07:
SonOff have released little newer signed firmware but not six ten https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011 but perhaps its coming one, you never knowing ;-)

r-vit, 2021-12-07:
Oh, nice! Thank you, I'll wait for the official one before trying to upgrade it manually. 
Closing the issue now, thanks again for your help.

closed 2021-12-07 by r-vit