grobasoz/zigbee-firmware #20: [REQUEST] EmberZNet 7.0.0.0 (7.0.0.x) Zigbee Coordinator firmware images?

Hedda, 2022-01-07:
FYI, Silicon Labs have now reeleased Silicon Labs Zigbee EmberZNet SDK 7.0.0.0 (and Gecko SDK 4.0.0)

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.0.0.0.pdf

http://docs.silabs.com/zigbee/7.0

Also first time Silicon Labs posted Gecko SDK Suite to publicly available repository on GitHub:

https://github.com/SiliconLabs/gecko_sdk/tree/gsdk_4.0/protocol/zigbee

https://github.com/SiliconLabs/gecko_sdk/

https://github.com/SiliconLabs/gecko_sdk/releases

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/demo-applications.zip

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/documentation.zip

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/gecko-sdk.zip

https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.0.0/pintool.zip

https://github.com/SiliconLabs/gecko_sdk/archive/refs/tags/v4.0.0.zip

https://github.com/SiliconLabs/gecko_sdk/archive/refs/tags/v4.0.0.tar.gz

PS: Not sure if Gecko SDK 4.0 was accidentally released or official as of yet as their version history still says that Gecko SDK Suite 3.2 is latest:

https://docs.silabs.com/gecko-platform/latest/version-history

Link to 4.0 seem to work through:

https://docs.silabs.com/gecko-platform/4.0/

grobasoz, 2022-01-28:
@Hedda - I'm not sure GSDKV4.0 is ready for mainstream utilisation just yet. I will continue with testing and if all looks OK, will post some updates to the firmware.

closed 2022-01-28 by grobasoz

Hedda, 2022-01-28:
FYI, Silicon Labs now announced the EFR32MG24 (EFR32 MG24) SoCs and Silabs EFR32xG24 development kits + radio boards:

https://news.silabs.com/2022-01-24-Silicon-Labs-Brings-AI-and-Machine-Learning-to-the-Edge-with-Matter-Ready-Platform

https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-socs

https://www.cnx-software.com/2022/01/24/silicon-labs-bg24-and-mg24-2-4-ghz-wireless-mcus-quadruple-ai-performance-at-a-fraction-of-the-energy/

Looking at specs EFR32MG24 will not make a better Zigbee coordinator than EFR32MG21 but maybe better router and end devices:

* EFR32MG24 come with higher RAM memory, with 128 or 256 KB variants, (compared to 64 or 966 KB variants of EFR32MG21).
* EFR32MG24 flash storage has been increased to 1536 KB or 1024 KB, (compared 512, 768, or 1024 KB in the EFR32MG21).
* Interestingly their similar ARM Cortex-M33 core runs at 78 MHz compared the 80 MHz core in the EFR32MG21
* New feature is an[ integrated AI/ML accelerator NPU for edge interference](http://silabs.com/ai-ml) to potentially lower devices power consumption. 
* _EFR32BG24 and EFR32MG24 SoCs are already shipping today to Alpha customers, with mass deployment slated for April 2022._

New "EFR32xG24 Pro Kit +10 dBm" + "EFR32xG24 Pro Kit +20 dBm" dev boards and EFR32xG24 radio boards coming coon:

[image]

[image]

No news on FCC/CE cerified MGM240 (MGM240P and MGM240L) modules based on EFR32MG24  to replace MGM210.

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules

https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-modules

grobasoz, 2022-01-28:
Saw that - Kits can take a long time to arrive and even longer for the firmware to catch up. MG22 has been out for a while and supporting firmware still isn't complete. 

Sourcing the MG24's should also be an exercise... 

I think I will stick with EFR32MG21A020F1024 as I have radio modules and development kits available as well as "mature" software support for Bluetooth, Zigbee (incl Unify SDK), OpenThread and now Matter.

Hedda, 2022-01-28:
> Sourcing the MG24's should also be an exercise...

One would think Silicon Labs would choose to not launch it now unless they had already secured supply-chain to meet demand?

Think I first read about EFR32MG24 around 2-years ago so could be possible that they delayed the release of it until now just to make sure they first secured supply-chain and made sure they now are finally ready to ramp up manufacturing of it? See news:

I know that Silabs have suffered very hard during the current chip shortage and not even been close to keep up with demand.

grobasoz, 2022-01-28:
One would think... There is stock coming out - Mouser has 187k of the EFR32MG21A020F1024 available... for a price :)