grobasoz/zigbee-firmware #22: Alternative EmberZNet Zigbee "Multi-PAN" firmware for EFR32 Series 1 and/or EFR32 Series 2?

Hedda, 2022-01-31:
FYI, Home Assistant developer agners has begun working on an add-on for Silabs Multiprotocol stack that requires "Multi-PAN" firmware:

https://github.com/zigpy/zigpy/discussions/894

This project does however require that firmware include Multi-PAN Library plugin. That new Multi-PAN Library plugin is new since EmberZNet SDK 6.8.0.2 and new multi-PAN feature can create a host/NCP application that can support up to two coordinator networks:

https://www.silabs.com/documents/public/application-notes/an724-multi-network.pdf

Very early so would not advise any users use "Multi-PAN" firmware testing unless interested in testing and/or contributing to that project.

https://github.com/home-assistant/addons-development/tree/master/silabs-multiprotocol

Concept could in the future even allow running both Zigbee 3.0 and Thread/Matter (OpenThread) stacks applications on a single radio.

It also changes architecture from NCP (Network Co-Processor) based to "DMP RPC" (Dynamic Multiprotocol Radio Co-Processor) based which if I understand correctly offload the network part to the the Zigbee integration application running on system CPU and the adapter becomes slightly more of a "dumb" Zigbee radio (still using EZSP) which for Zigbee removes some limitations on routing tables etc. (meaning that can probably have almost unlimited of devices connected even on radio adapter that has an MCU with limited RAM-memory.

PS: agners is Nabu Casa lead engineer working on Home Assistant Yellow and its EFR32MG21 radio (MGM210P SiP module):

https://www.home-assistant.io/blog/2021/09/13/home-assistant-yellow/

https://www.crowdsupply.com/nabu-casa/home-assistant-yellow

https://www.nabucasa.com/about/

Hedda, 2022-01-31:
Silicon Labs recently released EmberZNet 6.8 (6.8.0.1) SDK with a new feature to concurrent support of multiple PANs (multi-PAN).

_Prior to EmberZNet PRO 6.8.0, the multi-network implementation limited the number of always-on roles that a device could serve on the participating networks. Starting with EmberZNet PRO 6.8.0, the multi-network feature and a new multi-PAN feature allow the device to operate as a coordinator on both Zigbee networks in a host-NCP configuration._

Silicon Labs AN724 application note discusses the design considerations involved in integrating a feature that allows a node with one Zigbee radio to operate on multiple Zigbee networks.

https://www.silabs.com/documents/public/application-notes/an724-multi-network.pdf

"_**Multi-network and multi-PAN features allow a device to operate on two Zigbee networks using the same radio. These Zigbee networks may have different security settings or network parameters, such as short ID, PAN ID, extended PAN ID, or radio power. The only parameter that stays the same on all networks is the node's EUI64. While the multi-network feature allows the two networks to be on different radio channels, the multi-PAN feature requires that this setting matches.**_"

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.8.0.2.pdf

# 1 New Items

## Added in release 6.8.0.2

The feature to concurrently support multiple PANs (multi-PAN) is added in the 6.8.0.1 release. The multi-PAN feature builds upon the existing multi-network feature, however, the multi-network feature limits the number of always-on networks to one, the multi-PAN feature allows for two always-on networks, both of which must be coordinators. The two networks use the same radio to send and receive packets on their own distinctive PAN IDs.

For additional documentation refer to AN724: Designing for Multiple Networks on a Single Zigbee Chip.

## 1.1 New Plugins

### Added in release 6.8.0.2

**Multi-PAN Library**

The new plugin is used by the multi-PAN feature to create a host/NCP application that can support up to two coordinator networks.

### Added in release 6.8.0.2

**Multirail-demo**

A new mutirail-demo plugin has been added. This plugin provides sample code to initialize and interact with a second RAIL handle and is used in the new multi-rail GP sample application.

## 1.2 New APIs

### Added in release 6.8.0.2

**Stack Profile and Security Level**

Introduced **emberSetStackProfile()** together with the following enums: **EMBER_STACK_PROFILE_NONE**, **EMBER_STACK_PROFILE_ZIGBEE_PRO**, **EMBER_SECURITY_LEVEL_NONE**, **EMBER_SECURITY_LEVEL_Z3**.

In addition to the new API, the Zigbee stack now initializes the stack profile and security level based on the security profile of each network so that multi-PAN devices are able to form networks with different stack profiles and security levels.
For additional documentation refer to the Zigbee Framework API Reference Guide.

## 1.3 New Sample Applications

### Added in release 6.8.0.2**

**Multi-PAN**

A new set of Host (MpZ3TcCustomTcHost) and NCP (mp-ncp-spi or mp-ncp-uart) sample applications is added. These sets demonstrate the multi-PAN feature. The host application is a Zigbee 3.0 coordinator on the first network and a coordinator with no security on the second network and is meant to connect to an NCP running one of the multi-PAN NCP applications.

**ZigbeeMinimalHost**

The EmberZNet ZigbeeMinimalHost sample application provides a minimal functional subset to serve as a starting point for users wishing to build their own ZigBee Host applications. The application is configured to operate as a ZigBee Coordinator / Router. No ZigBee Cluster Library (ZCL) application-layer functionality is preconfigured. In the Studio New Project workflow Select Application dialog, it is recommended to select this sample application, rather than the "Start with a blank application" checkbox, to begin development of a new Zigbee Host application.

**Z3GatewayGpComboHost and ncp-uart-hw-gp-multi-rail**

A new set of Host (Z3GatewayGpComboHost) and NCP (ncp-uart-hw-gp-multi-rail) sample applications is added. This set demonstrates the use of the additional rail handle to send application-specific bidirectional GPDF from Combo to GPD.

grobasoz, 2022-02-02:
I have installed EmberZNet 6.8.2.0 and been building and testing various options. Unfortunately I can't get the MP example to work yet.

Hedda, 2022-02-02:
Probably better to try Multi-PAN with a later newer release of EmberZNet like EmberZNet 6.10 / 6.10.x.x rather than 6.8.0.2

I only posted 6.8.0.2 as a reference since it was the first to contain support so its documentation contained the information.

grobasoz, 2022-02-03:
I have added [MultiPan](https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32%20Series%202/EFR32MG21A020F1024) for Series 2 devices and EmberZNet 7.0.1 NCP

Hedda, 2022-02-03:
> I have added [MultiPan](https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32%20Series%202/EFR32MG21A020F1024) for Series 2 devices and EmberZNet 7.0.1 NCP

But that is Multi-PAN Zigbee NCP only (right?) and not also Multi-PAN RCP required by agners addon? -> https://github.com/zigpy/zigpy/discussions/894

grobasoz, 2022-02-03:
@Hedda  - Yes, just NCP so far - working on MP RCP now...

TheDevFreak, 2022-05-18:
Any chance of MultiPan for Series 1 devices? (Specifically interested in the device currently served by `NCP_UHW_MG1B232_650_PA0-PA1-PB11_PA5-PA4.gbl` -- this is used for the Lidl Silvercrest/Tuya Wired Zigbee Gateway)

MattWestb, 2022-05-18:
In GSDK 7 is not having Zoigbee / OT support for EFR32MG1B chips (its deleted) but the EFR32MG1P that IKEA is / was using is still in place and i have one working RCP that is being used for Zigbee EZSP 7.0.x and Open Thread as boarder router.

I dont knowing if Gary can helping cooking one RCP for the EFR32MG1B that is being used in LIDL / tuya ZBGW but hi was doing our original working one for it (the pin layout is different and can using the MG1P for the MG1B then its using different ports for the hardware).

closed 2022-10-20 by grobasoz