grobasoz/zigbee-firmware #23: [REQUEST] EmberZNet Zigbee NCP firmware for Ebyte E104-BT11 modules based on EFR32MG21 with 1024Mb

Hedda, 2022-02-16:
For reference; this question was already discussed a little previously in -> https://github.com/zigpy/zigpy/discussions/727 as well as here -> https://github.com/zha-ng/EZSP-Firmware/issues/11

Availability of Silicon Labs EFR32MG21 based SiP/SiL radio modules like [MGM210 (MGM210P/MGM21L)](https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules) are still scarce so just want to ask:

You think Ebyte E104-BT11G modules which look to be based on Silabs 1024Mb EFR32MG21 can be flashed with Zigbee NCP firmware?

https://www.cnx-software.com/2021/05/04/ebyte-e104-bt11-bluetooth-mesh-module-features-silicon-labs-efr32mg21-mcu/

https://www.cdebyte.net/bluetooth-module/2-4g-ble-mesh-network-bluetooth-module.html

https://www.ebyte.com/en/downpdf.aspx?id=1010

https://www.manualslib.com/manual/1895760/Ebyte-E104-Bt11.html

https://www.ebyte.com/en/downpdf.aspx?id=1010

https://www.ebyte.com/downpdf.aspx?id=990

[image]

E104-BT11 series include "E104-BT11x-IPX" (IPX antenna) and Ebyte "E104-BT11x-PCB" (PCB trace antenna), where the only difference between the "G" and "N" variants looks to be one is pre-configured as Bluetooth Gateway and other as Bluetooth Node for a BT mesh.

Name | Remark | Antenna Type | Size | Distance
-- | -- | -- | -- | --
E104-BT11G-IPX | GATEWAY MODULE | IPX Antenna | 16*13mm | 300m
E104-BT11G-PCB | GATEWAY MODULE | PCB Antenna | 19*13mm | 200m
E104-BT11N-IPX | NODE MODULE | IPX Antenna | 16*13mm | 300m
E104-BT11N-PCB | NODE MODULE | PCB Antenna | 19*13mm | 200m

* https://www.aliexpress.com/wholesale?catId=0&SearchText=E104-BT11G-PCB
  * https://www.aliexpress.com/item/1005001726039426.html
  * https://www.aliexpress.com/item/1005003221082503.html
  * https://www.aliexpress.com/item/1005003676741873.html

* https://www.aliexpress.com/wholesale?catId=0&SearchText=E104-BT11G-IPX
  * https://www.aliexpress.com/item/4001260142336.html
  * https://www.aliexpress.com/item/1005001671008992.html

* https://www.alibaba.com/product-detail/GFSK-BLE-mesh-integrated-circuits-blue_1600200894074.html

* https://www.alibaba.com/product-detail/E104-BT11G-PCB-EFR32-2-4g_1600080880779.html

* https://www.alibaba.com/product-detail/Ebyte-Wireless-Communication-E104-BT11G-IPX_1600080305209.html

[image]

## Pin definition

Pin | Function | Description 
-- | -- | -- | -- 
1 | GND | Ground, connect to power reference ground
13 | NetLink | Network indicator, pin outputs high level after successfully networking
17 | TXD | TX pin of UART，for UART receiving
18 | RDX | RX pin of UART，for UART transmitting
19 | VCC | Power supply 1.9～3.6V（It is recommended to add a ceramic filter capacitor externally)
20 | GND | Ground, connect to power reference ground
22 | PC03 | Low-power role selection pin (internal pull-up). Low power consumption role: pull up or float. Other roles: Go directly to the ground.
23 | PC04 | Reset. Except for hardware information (UART configuration), factory settings are restored.
24 | Wakeup | The wake-up pin is valid in low level under low power mode.
25 | RST | Chip reset trigger input pin, valid in low level
26 | GND | Ground, connect to power reference ground

[image]

PS: The G-version is configured as one "coordinator" and the N as router device all with BT mesh.

grobasoz, 2022-02-17:
Hi @Hedda - I don't see an issue running Silabs NCP firmware on an E104-BT11 - they do use different pin numbers to the "standard?" ones I use so it will be custom firmware.

Hedda, 2022-02-17:
> they do use different pin numbers to the "standard?" ones I use so it will be custom firmware.

Ah OK. FYI, I have not actually bought one myself and am not planning to either as rather now wait for new EFR32MG24 modules.

grobasoz, 2022-02-18:
Agree - will wait for MG24...

Hedda, 2022-02-18:
Heard any news on EFR32MG24 modules from Ebyte or RF-Star/RFStar, or first-party like Silabs MGM210 or MGM220?

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules

https://www.silabs.com/wireless/zigbee/efr32mg22-series-2-modules

Guessing the first-party radio modules will be called MGM240 (MGM240P and MGM24L) with many different specifications?

https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-socs

And sssume webpage when released will be https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-modules then?

grobasoz, 2022-02-19:
I haven't heard anything about module release dates. We are supposed to get chips in March so should have modules completed by end June '22 - just depends on speed of FCC... but having said that it took a year or so to get the MG21 dev parts after their release and another year to get stock! So not holding breath!

Hedda, 2022-09-07:
@grobasoz FYI, MGM240 (MGM240P and MGM240L) radio modules that are meant to be used in commercial products now look to be available:

https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-modules

https://fccid.io/QOQ-GM240P

https://fcc.report/FCC-ID/QOQ-GM240P/

https://www.silabs.com/documents/public/data-sheets/mgm240p-datasheet.pdf

<html>
<body>

Part Number | MCU | Core Frequency (MHz) | Flash | RAM | Cryptography | Secure Vault | Output Power Range (dBm) | GPIO | Temperature Range ºC | Package Type | Antenna | Package Size (mm)
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
MGM240PA22VNA  | ARM Cortex-M33 | 78 | 1536 | 256 | AES-128 AES-256  ECC  SHA-1  SHA-2 | Mid | -24  10 | 26 | -40  105 | PCB | Built-in | 12.9x15.0x2.2
MGM240PA32VNA | ARM Cortex-M33 | 78 | 1536 | 256 | AES-128 AES-256  ECC  SHA-1  SHA-2 | Mid | -33.7  19.9 | 26 | -40  105 | PCB | Built-in | 12.9x15.0x2.2
MGM240PA32VNN | ARM Cortex-M33 | 78 | 1536 | 256 | AES-128 AES-256  ECC  SHA-1  SHA-2 | Mid | -33.7  19.9 | 26 | -40  105 | PCB | RF-Pin | 12.9x15.0x2.2
MGM240PB22VNA| ARM Cortex-M33 | 78 | 1536 | 256 | AES-128 AES-256  ECC  SHA-1  SHA-2 | High | -24  10 | 26 | -40  105 | PCB | Built-in | 12.9x15.0x2.2
MGM240PB32VNA | ARM Cortex-M33 | 78 | 1536 | 256 | AES-128 AES-256  ECC  SHA-1  SHA-2 | High | -33.7  19.9 | 26 | -40  105 | PCB | Built-in | 12.9x15.0x2.2
MGM240PB32VNN | ARM Cortex-M33 | 78 | 1536 | 256 | AES-128 AES-256  ECC  SHA-1  SHA-2 | High | -33.7  19.9 | 26 | -40  105 | PCB | RF-Pin | 12.9x15.0x2.2

</body>
</html>

They have now also released those evaluation radio boards based on them that are meant to be used with Silicon Labs development boards from their starter kits. 

https://www.trustedparts.com/en/part/silicon-labs/XGM240-RB4316A

xGM240P radio module board = XGM240-RB4317A (20dBm).

https://www.silabs.com/development-tools/wireless/xgm240-rb4317a-xgm240p-module-radio-board

xGM240L radio module board = XGM240-RB4316A (10dBm).

https://www.silabs.com/development-tools/wireless/xgm240-rb4316a-xgm240p-module-radio-board

PS: Noted that those new radio module boards were expected to be called MGM240P and MGM240L but look like they will instead be marketed as Silabs xGM240x Series 2 Modules, specifically "xGM240P Broad-based Module" and "xGM240L Lighting Module".

grobasoz, 2022-09-07:
I have ordered some MGM240PA32VNA (1.5Mb, Mid Vault, 20dB, built in antenna) to use for development as Matter devices. Quite expensive (USD13) but necessary until my own production run of ERF32MG24 based modules is complete and FCC certified.

Hedda, 2022-09-07:
So got any plans to make and sell USB dongles based on EFR32MG24? Preferably sold with a high-quality external antenna? 😄 

I seriously think that there is at least a niche market for a "high-end" variant of such EFR32MG24 based radio USB dongle for early adopter people who want the best hardware, especially now that will be able to use the same hardware for Zigbee and Thread (for Matter over Thread). I believe those same people might even want to purchase more than one each if it will be recommended to use separate dedicated radio USB dongles for Zigbee Coordinator and Thread Border Router adapters.

Similar to how Electrolama, Tube Zigbee Gateways, and ZigStar are all selling relatively well via Tindie -> https://www.tindie.com

https://electrolama.com/ -> https://github.com/electrolama/

https://www.tubeszb.com -> https://github.com/tube0013/tube_gateways

https://zig-star.com -> https://github.com/mercenaruss/zigstar_gateways

PS: Maybe even worth building Bluetooth controller firmware for it now that Home Assistant got a native Bluetooth integration?

https://www.home-assistant.io/integrations/bluetooth/

grobasoz, 2022-09-08:
At this stage I sell most of my devices to companies that develop and sell their own Zigbee based products. They generally have great technical expertise (I do very little support) and can handle higher volumes, eg 100 coordinators and 1000 routers per order. That alone justifies the shipping costs ex Australia!
My concern with selling a USB Zigbee dongle is I'm competing with a $7 Sonoff device. I feel end users may not care about or understand RF quality as long as it works and is cheap?

MattWestb, 2022-09-08:
Its the same trying making and selling some product that tuya is spamming the market and can getting free shipment to all the word and only paying 5$ its what you must paying for getting it in stock and then must testing it and doing the shipment = Zero profit and much work and likely also lot of problems with end users.

For the community HA is selling there Sky-Connect that is one MG21 and with good USB chip (i have not open my for SWD flashing it but it coming after holiday in Spain).
Its have NCP and is coming RCP firmware for it so have support for Zigbee and OT in the near future.
And very likely some one like have one router firmware for it (I hope its not being made then baying one good light or outlet instead) and home cooked FW is not possible lading then the boot loader is locked witch app its loading or not = very good.

Great hearing is going well for you Gary and also have time digging in the Silabs messed SDK that is moving forward and also bringing more problems with it.
Still the best EZSP NCP is 6.7.8 and the second is 6.10.3 with more problem paring devices so no need for 7 as long its not working better then our current stable ones (6.7.8).

Hedda, 2022-09-08:
> My concern with selling a USB Zigbee dongle is I'm competing with a $7 Sonoff device. I feel end users may not care about or understand RF quality as long as it works and is cheap?

I think that while that is probably true for a majority of new and casual users, however, I believe that more experienced users and enthusiasts will and do both use and promote high-end products for some core hardware components such as a 20 dBm RF power capable EFR32MG24-based radio USB adapter shipping with a high-quality external antenna + options to choose Zigbee Coordinator and Thread Border Router firmware, thus I think that you could charge $40-60 US-dollars for a such "high-end" product depending how much the shipping costs are.

I do not think that a premium price of $49 or even $59 US-dollar would be way too much to ask for what would be a top-of-the-line product in this niche market.

Shipping cost will of course play a major factor for people who only buy one adapter so if you would prefer to only sell higher volumes then once you have a prototype to show you could perhaps try to reach out to different contacts globally with some of the online home automation enthusiasts resellers around the world in the hope that they are interested in making some deals for bulk orders then I could believe some of them might be interested in buying many USB adapters each to resell to the customers in their own country, and that would at least keep shipping cost down.

Perhaps even check out "[Fulfillment by Amazon (FBA)](https://sell.amazon.com/fulfillment-by-amazon)" which could let you sell your products in Amazon's store and outsources inventory management to Amazon's fulfillment centers through their FBA program, however, I think selling radio frequency devices via Amazon FBA require getting an FCC approval (FCC certification for that product) if sold in North America and CE mark (self-certification mark) if sold in the EU (European Union) :

* https://sell.amazon.com/fulfillment-by-amazon
  * https://sellercentral.amazon.com/gp/help/external/G200141420
  * https://sellercentral.amazon.com/help/hub/reference/external/HXJGN6T5K8HGGWW
  * https://sellercentral.amazon.se/help/hub/reference/external/G3NRFVS4YUYRQ85J

"High-end product" in this scenario == EFR32MG24 based USB adapter with a quality enclosure + high-quality external antenna.

Maybe also add additional GPIO-pads, buttons, and LEDs on the board as added value as a development board for developers(?).

Remember that you would not be competing with ITead/Sonoff with a such niche product but rather compare with these instead:

* Electrolama "ZZH MULTIPROTOCOL RF STICK" = 5 dBm RF power CC2652R1-based sell for $35(US) + shipping from England.
  * https://shop.electrolama.com/collections/usb-rf-sticks/products/zzh-multiprotocol-rf-stick
* Slaesh "CC2652RB development stick" == 5 dBm RF power CC2652RB-based for $30(US) + shipping from Germany.
  * https://slae.sh/projects/cc2652/ 
    * By the way, have another look at cart checkout at https://slae.sh/projects/cc2652/ which offer some options to select Zigbee EZSP or OpenThread firmware pre-flashed,  USB extension cable or not, and an extra "tip" box to donate/sponsor his development.
* Phoscon (Dresden Elektronik) "ConBee II"  == 5 dBm RF power CC2652RB-based for $37-40(US) + shipping from Amazon.
  * https://www.amazon.de/-/en/ConBee-Universal-Zigbee-USB-Gateway-black/dp/B07PZ7ZHG5

Also, could also sell the same "USB stick" as a top-of-the-line Zigbee Signal Repeater device pre-flashed with Zigbee router or Thread router firmware if added a USB extension cable with a foot that acts as a cradle/dock in order to hold up the USB stick. Ex:

https://www.aliexpress.com/item/4000186286169.html
https://www.aliexpress.com/item/1005002641334096.html
https://www.aliexpress.com/item/1005003532032834.html
https://www.aliexpress.com/item/4000024112722.html
https://www.aliexpress.com/item/4000532552991.html
https://www.aliexpress.com/item/32850737048.html
https://www.aliexpress.com/item/1005002470775709.html

[image]

[image]

Hedda, 2022-09-08:
Oh, btw, ITead now sell “Sonoff Zigbee 3.0 USB Dongle Plus V2” model “**ZBDongle-E**” based on EFR32MG21 for $19.99 US-dollar...

...and that means $20 US-dollar + shipping when ship if from China so people in the USA and EU have to pay custom chargers too.

I bet that local resellers around the world will soon be selling that new EFR32MG21-based **“ZBDongle-E”** adapter for $30-40(US).

https://itead.cc/product/zigbee-3-0-usb-dongle/

In addition, ITead has also tried (but failed) to make it clear to buyers will continue to sell their CC2652P-based variant of “Sonoff Zigbee 3.0 USB Dongle Plus” (non-"V2") that has recently been rebranded to "**ZBDongle-P**" side-by-side as an alternative to the new "**ZBDongle-E**" to that buyers have the option to buy both Texas Instruments and Silicon Labs based USB adapters because different home automation applications currently do not offer the same support for all SoC manufacturers.

Anyway, they also raised the price of the CC2652P-based "**ZBDongle-P**" to $19.90 + shipping when buying it direct from China:

https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/

closed 2022-10-20 by grobasoz