grobasoz/zigbee-firmware #24: [REQUEST] EFR32MG13P732 router firmware images for Tuya TYZS1L module and Tuya Zigbee USB Range Extender / USB Dongle Signal Repeater

Hedda, 2022-03-22:
Tuya is apparently making a 10 dBm "Zigbee 3.0 Range Extender" / "Zigbee 3.0 Signal Repeater" (Tuya SKU: "RP280") described as Zigbee Router device in a USB dongle/stick format, which at looks good on paper but has bad firmware so should not be used unless flashed with other firmware. Flashing this with custom Zigbee router firmware could maybe be of interest to the community?

https://zigbee.blakadder.com/Tuya_TS0207.html

Zigbee 3.0 Signal Repeater Model TS0207 manufactured by Tuya

Zigbee ID: "TS0207"; "_TZ3000_m0vaazab"; "FNB54-WTS08ML1.0"

https://github.com/zigpy/zigpy/discussions/736

https://community.home-assistant.io/t/tuya-zigbee-signal-repeater-working-with-zha-and-conbee-2/249329/

[image]

I think it is based Silabs EFR32MG13 / EFR32MG13P732HI chip, the same as the not so common Tuya TYZS1L module, if so should have 512KB Flash storage and 64KB RAM memory? 

https://developer.tuya.com/en/docs/iot/zigbeetyzs1lmodule?id=K989rjcbpt4cx

https://developer.tuya.com/en/docs/iot/light-design?id=K95afopz8waah

https://usermanual.wiki/Hangzhou-Tuya-Information-Technology/TYZS1L-4315271.pdf

Obviously the Tuya USB Signal Repeater can not be flashed with Zigbee Coordinator firmware as it does not have a USB-to-UART bridge chip / USB-to-Serial converter chip, and the Tuya TYZS1L module is probaly only used in router devices and not gateways/bridges?

grobasoz, 2022-03-23:
@Hedda - thanks for the information. I could look into a router firmware for this - but what about programming? You'd need an SWD programmer with pins laid out to connect to the exposed pins on the Tuya board... 10dBm is a little low - probably to keep the price low, but perhaps a 20dBm Series 2 based board might be better?

Hedda, 2022-03-23:
Yeah, I think it is mostly the "USB stick" format and super low price (if order in bulk from China) that make it an attractive product.

IKEA TRÅDFRI Signal Repeater uses a similar chip and comes with a power-supply so it probably a better deal in the end:

https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/

https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

blakadder, 2022-10-13:
I have a [TYZS13](https://developer.tuya.com/en/docs/iot/tyzs13-module-datasheet?id=K98u373pszqe3) module found in Tuya T6E Android panels. I've managed to [reflash it](https://blakadder.com/t6e-z2m) to new firmware via bootloader and J-Tag via the convenient test pads. Since it runs android and adb shell there's even a possibility of direct flashing but I'm not skilled enough to figure it out with the limited tools Android shell has.

It would be beneficial to have router firmware for it since most users have an established network and don't want to run this device as the coordinator.

MattWestb, 2022-10-13:
Interesting project !!

I have see some tuya ZBGW that is wall boxed but without panel but i think its using the same Zigbee module.
I think the most stable NCP firmware is the `efr32mg13p-v8-6780-sw-115200.gbl` and if you is finding one EZSP 6.10.x is shall also being good but the 6.8 and 6.9 is normally having some strange problems that is fixed in the 6.10.

Have you finding some pin / pads that can being used as paring / reset button ?
If the Zigbee module have some connected to the android SOC it can being easy made in Android i think.
Also with com pads its possible doing it by CLI from your Android SOC.

You can also considering installing one Zigbee Controller bridge that have CLI and can sending and receive commands to all devices in the network (its the same way IKEA and HUE is doing in the current bridges / hubs without coordinator).
Then you have one router plus can controlling all devices also the the host system and / or the coordinator is off line from your Android panel with CLI.

grobasoz, 2022-10-13:
@blakadder - Would need some more information on the device... eg basic pin allocation/usage. For routers I'd suggest a simple SoC application, with an API for the Android interface? T6E is nice but US$98?

blakadder, 2022-10-13:
I don't know much about pin allocation except that tx and rx are connected to the Android board and accessible on /dev/ttyS5

There are no buttons or lights connected to it

blakadder, 2022-10-13:
> I have see some tuya ZBGW that is wall boxed but without panel but i think its using the same Zigbee module. I think the most stable NCP firmware is the `efr32mg13p-v8-6780-sw-115200.gbl` and if you [...]

That's good to know, I just picked the latest available! I'll probably change the guide to use that one
 
> Have you finding some pin / pads that can being used as paring / reset button ? If the Zigbee module have some connected to the android SOC it can being easy made in Android i think. Also with com pads its possible doing it by CLI from your Android SOC.

All the pins are available on the module and there are test pads, f.e. RST can be used to reboot it couple times to put into pairing mode but then again you can do that by repowering the entire panel. 

> You can also considering installing one Zigbee Controller bridge that have CLI and can sending and receive commands to all devices in the network (its the same way IKEA and HUE is doing in the [...]

I don't even know what that is, most of my Zigbee knowledge is on consumer level with occasional firmware flash.

MattWestb, 2022-10-13:
Gary from tuya Dev docks:
```
PWM3	UART_CTS	PF4	The coordinator used for the gateway must be connected to hardware flow control by default. The baud rate is 115200. The pin is connected to UART_RTS of the MCU.
PWM2	UART_RTS	PA2	The coordinator used for the gateway must be connected to hardware flow control by default. The baud rate is 115200. The pin is connected to UART_CTS of the MCU.
UART_RX	UART_RX	PA1	The pin is connected to UART_TX of the MCU.
UART_TX	UART_TX	PA0	The pin is connected to UART_RX of the MCU.
```
But we dont use the hardware flow control on this device i think.

From the working firmware flashed site:
```
efr32mg13p-v8-676-sw-115200
EFR32MG13P target
NCP Software flow control, 115200 baudrate
NCP UART TX --> PA0
NCP UART RC <-- PA1
EZSP Version 8
EmberZNet 6.7.6
DCDC bypass
```
I think they is using standard SWD pins for chip flashing (its the no marked TP between GND and RX).

Hedda, 2022-10-14:
> @blakadder - Would need some more information on the device... eg basic pin allocation/usage. For routers I'd suggest a simple SoC application, with an API for the Android interface? T6E is nice but US$98?

FYI, posted some pin info here -> https://github.com/grobasoz/zigbee-firmware/issues/33

blakadder, 2022-10-17:
Yes, they're standard, I flashed via SWD using the pins on the module instead of test pads

grobasoz, 2022-10-17:
So if I understand correctly, if you had router firmware on this device, you could also interface to it via a serial interface (eg /dev/ttyS5) from the Android app to do certain "tasks", and possibly get network information?
One would need an Android App with a serial API?
Also we'd need to have or create an API/CLI for the router?

blakadder, 2022-10-17:
Its possible if there is an app for that but there's also cli interface with adb shell and socat is available

Hedda, 2022-10-17:
> So if I understand correctly, if you had router firmware on this device, you could also interface to it via a serial interface (eg /dev/ttyS5) from the Android app to do certain "tasks", and [...]

@xsp1989 built a preconfigured Zigbee Router firmware mage for some EFR32MG21 dongles as well as Sonoff ZBBridge by ITead (and ZB-GW03 Bridge with same radio module) , if I remember correctly he mentioned somewhere that he mentioned that his router firmware could be accessed through serial API/CLI?

https://github.com/arendst/Tasmota/discussions/11536

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle/README.md#extra-features

 **_Extra features_**
* **_If you send data through the serial port, the router will forward the data to the gateway._**
* **_If data is sent to the router through the coordinator, the serial port will receive the data._**

grobasoz, 2022-10-17:
@Hedda -Thanks for the reference. I see they are using tunneling for the serial transfer. Personally I don't like Zigbee tunneling, and prefer a simple CLI/API not unlike the old AT command set that Telegesis (and Digi with XBee) used.

Hedda, 2022-10-17:
> I see they are using tunneling for the serial transfer. Personally I don't like Zigbee tunneling, and prefer a simple CLI/API not unlike the old AT command set that Telegesis (and Digi with XBee) used.

I believe I read that @xsp1989 also mentioned somewhere that EZSP or some other API/CLI was enabled in his router firmware images as well, so might not just be the serial tunnelling over Zigbee tin his builds?

Hedda, 2022-10-20:
Are there differences between Tuya TYZS1L Module (original request) and Tuya TYZS13 Module (T6E) that affect firmware builds?

Tuya TYZS1L Module that is used in = EFR32MG13P732IM48-C (512-KB flash memory, 64-KB RAM)
https://developer.tuya.com/en/docs/iot/zigbeetyzs1lmodule?id=K989rjcbpt4cx

Tuya TYZS13 Module that is used in T6E smart panels = EFR32MG13P732F512GM48 (512-KB flash memory, 64-KB RAM)
https://developer.tuya.com/en/docs/iot/tyzs13-module-datasheet?id=K98u373pszqe3

blakadder, 2022-10-27:
Got another version of a Tuya panel with the same TYZS13 module

[image]

MattWestb, 2022-10-27:
@blakadder Is the TYZS13 being used as one Zigbee NCP or one Bluetooth NCP then its written `BLE` near the antenna ?
Also with device is it ?

blakadder, 2022-10-27:
It's a Zigbee gateway. Device is ZS-001 or TPP01-Z Tuya control panel

Hedda, 2024-03-18:
Any chance someone here is interested in putting together a generic always-on Zigbee Router (ZR) device application for/with Silicon Labs Gecko SDK (GSDK) that could be reused in the silabs-firmware-builder project for automatically building matching Zigbee Router firmware images to allow users to convert Silabs EFR32 based Zigbee Coordinator USB dongles into dedicated Zigbee signal repeater / Zigbee range extender? If so then please see discussion here -> https://github.com/darkxst/silabs-firmware-builder/issues/15

closed 2024-08-12 by grobasoz