grobasoz/zigbee-firmware #25: [REQUEST] Silicon Labs Zigbee EmberZNet NCP 7.1.0.0 firmware images

Hedda, 2022-06-09:
Silicon Labs Gecko SDK (GSDK) 4.1.0 released with Zigbee EmberZNet SDK 7.1.0.0

https://github.com/SiliconLabs/gecko_sdk/releases

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.0.0.pdf

- 2.4GHz Zigbee Smart Energy support for xG24
- 802.15.4 Signal Identifier and MAC CCA Mode 2 and 3 support for xG24
- Zigbee Green Power Gateway Backup
- Updated GCC and IAR compiler version
- Alpha Concurrent Multiprotocol Zigbee in NCP mode and Open-Thread in RCP mode
- Alpha Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode

grobasoz, 2022-06-09:
Hi @Hedda - good to hear from you again... I'm busy installing the new frameworks and will have a look at the code when time permits, Gary.

Hedda, 2022-06-27:
FYI, depending on Zigbee application implementation the end-user use it will be important for end-users to know EmberZNet SDK 7.1.0 firmware added an update to EZSP v9 (EmberZNet Serial Protocol version 9) which might break some Zigbee application implementation if a user upgrades to so Silabs Zigbee EmberZNet SDK 7.1.0.0 NCP firmware before the Zigbee 
 application developers have added support for  EZSP v9.

In general, the EZSP v9 protocol version largely remains the same as that of  EZSP v8 protocol version, but I understand that there is a breaking change in some responses so that needs implementation to keep compatibility, (like adding sl_Status enum based on sl_Status.h from Silicon Labs Gecko SDK v4.1.0). 

So for now would recommend that end-users do not flash the very latest EmberZNet 7.1.x firmware and instead use no higher than 7.0.x firmware unless they know that the developers of the Zigbee application implementation they use already  added support for EZSP v9 protocol version.

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.0.0.pdf

That is unless and until EZSP adapter implementation in zigbee-herdsman is updated to support additions to EZSP v9. 

See EZSP Reverence guide Rev 4.4 and 4.3

zigbee-herdsman developers please EZSP Reference Guide for 7.0.0 with the changes from 6.10.x highlighted is here:

https://web.archive.org/web/20220401230237/https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

and for 7.1.0 with the changes from 7.0.0 > 7.1.0 hightlighted is here (until the next version):

https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

PS: From the related pull request for the bellows radio library in zigpy I gather that key new feature is adding sl_Status enum based on sl_Status.h from Gecko SDK v4.1.0 -> please see https://github.com/zigpy/bellows/pull/462

MattWestb, 2022-06-27:
If the host system is well implanted EZSP as Bellows have it shall not being any problem then the procedure is first resetting the NCP by software command and asking for EZSP version with one very old EZSP version command.
Then the host system can configure the highest possible protocol that both side is supporting. Example one old EM357 with EZSP 4 bellows is using EZSP 4 and if the same device is using one current firmware its using EZSP 8.
If the NCP / RCP+zbeed is having higher version the host shall using the highest is supporting (V8) and all shall being well.

MattWestb, 2022-06-30:
Was cooking one NCP for Billy with GSDK 4.1 and unpatched ZHA its complaining but looks working OK:
```
Logger: bellows.ezsp
Source: /usr/local/lib/python3.9/site-packages/bellows/ezsp/__init__.py:111
First occurred: 18:11:56 (1 occurrences)
Last logged: 18:11:56

Protocol version 9 is not supported, using version 8 instead
```
And in the log:
```
2022-06-30 18:11:57 INFO (MainThread) [bellows.zigbee.application] EmberZNet version: 7.1.0.0 build 0
```
The patched bellows with V9 support is coming in the next HA release.

Hedda, 2022-08-11:
@grobasoz Would be great if you have the possibility to provide EmberZNet Zigbee NCP 7.1.0.0 images to @kirovilya who is developing the EZSP adapter implementation for [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman) which is used by [Zigbee2MQTT](https://www.zigbee2mqtt.io) and [IoBroker](https://www.npmjs.com/package/iobroker.zigbee) applications, as Zigbee NCP 7.1.0.0 images are currently only wanted for development to ensure compatibility with EZSP 9 (EmberZNet Serial Protocol version 9). See:

https://github.com/Koenkk/zigbee-herdsman/issues/319

FYI, kirovilya has previously mentioned that he owns [Elelabs ELU013 stick](https://elelabs.com/products/elelabs-usb-adapter.html) and [Elelabs ELR023 shield](https://elelabs.com/products/elelabs-zigbee-shield.html) (both based on EFR32MG13P), an [Defaro Sprut Stick 2 Pro](https://spruthub.ru/device/defaro-sprut-stick-2-pro) (based on EFR32MG21), the [ZB-GW04 USB dongle (a.k.a. easyiot stick) which uses EFR32MG21 based "SM-011 v1.0" Zigbee module from CoolKit-Technologies](https://easyiot.aliexpress.com/store/5839056), plus he is also waiting for the arrival of the [new Sonoff-branded “ZBDongle-E” from ITead](https://itead.cc/product/zigbee-3-0-usb-dongle/) (also based on EFR32MG21).

grobasoz, 2022-08-12:
@Hedda  - Sure - will work on them over the weekend... Bearing in mind I'm not too convinced 7.1.0 is fully working. I haven't had time to test on a "live" system... Will try do so over the weekend...

MattWestb, 2022-08-12:
I hope Silabs have fixing MG13 support then MG1B is not working (comport settings looks being broken but the NCP is allocating NVR is its living) but MG1P and MG21 looks working but not all versions like Silabs MG21 Light module IKEA is using.

Hedda, 2022-09-01:
FYI; Silicon Labs has now released Gecko SDK (GSDK) 4.1.1.0 with Zigbee EmberZNet SDK 7.1.1.0 (7.1.1) bug-fix release:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.1.0.pdf

https://github.com/SiliconLabs/gecko_sdk/releases

kirovilya, 2022-09-08:
I have now updated the firmware in ZB-GW04 USB dongle (a.k.a. easyiot stick) to 7.1.1-17 EZSP v9 from [here](https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP).
after some manipulations in code, I managed to run Z2M on this firmware and it works.
devices connect, manage and send data.
By the way, the Touchlink also works.
[image]
[image]
[image]

small code changes will be posted a little later

Loic691, 2022-10-05:
Confirmed that it's working with 1.28.0 Z2M or upper

closed 2022-10-20 by grobasoz

Loic691, 2022-12-15:
Hi all
Where can we find the new ezsp 7.2 firmware ?

MattWestb, 2022-12-15:
https://github.com/SiliconLabs/gecko_sdk is the Silabs GIT for GSDK and EZSP only install Simplicity studio 5 and installing the GSDK and you can build firmware  for your device if you is having enough knowledge.
Or you is making one friendly request from the devs cooking one for you device.

Hedda, 2023-01-23:
FYI, Silicon Labs Gecko SDK (GSDK) Version 4.1.4.0 is a bug-fix release that contains Zigbee EmberZNet SDK 7.1.4.0

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.1.4

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.4.0.pdf

Silicon Labs has now also released Gecko SDK 4.2.0 with Zigbee EmberZNet SDK 7.2.0.0

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.0

Any chance you could release some builds for zigpy and zigbee-herdsman developers to test?

It looks to use EZSP v9 (EmberZNet Serial Protocol version 9) same as Zigbee EmberZNet 7.1

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.0.0.pdf

New features are mostly related to Dynamic Multiprotocol and multi-PAN 802.15.4 in RCP mode:

- Secure key storage support for MG2x parts that support Secure Vault-High

- Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode
- Dynamic Multiprotocol Zigbee-NCP + Bluetooth-NCP (experimental)
- Manufacturing Library (MfgLib) support for Concurrent Multiprotocol RCP
- Zigbee + OpenThread Concurrent Lis-tening on MG24 parts (experimental)
- Zigbee on Host (ZigbeeD) support for 32 bit and 64 bit x86 architecture (experimental)
- MGM240S SiP Module Support

They also added simultaneous Dual-Band and Dual-PHY for Zigbee Smart Energy on MG24/MG12:

- MG24+Si4468 Dual-PHY Zigbee Smart Energy support
- MG12 Dual-Band 2.4GHz + SubGHz Zigbee Smart Energy support

Deprecated in release 7.2.0.0:

- The Secure EZSP feature will be removed in a future relelease.

Loic691, 2023-03-02:
HI
I have now updated the firmware in SONOFF USB Dongle E to 7.1.4 EZSP v9.
With last Z2M all is working as in 7.1.1.
I keep nevertheless (same with 7.1.1) strange behaviour with some devices which leave network randomly and disapear from Z2M GUI.
Here's some description :
- ORVIBO ST30 temperature / humidity sensor
- Which leave randomly network and display the following error
`Failed to configure 'Sonde Roxane', attempt 3 (Error: ConfigureReporting 0xb4e3f9fffe25ad37/2 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Object.batteryVoltage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:66:9) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/orvibo.js:191:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:117:13))
`

- Xiaomo Air monitor VOCKQJK11LM
which leave the network randomly

- Legrand battery device 
Which leave the network randomly
Which on some device (home away) dont' send all the button state randomly
Which randomly have an important latence between push button event and send to mqtt

kirovilya, 2023-03-02:
@Loic691 Hi. do you really have the latest z2m 1.30.2?
I checked my legrand battery remote - it works fine, does not left network.

```Failed to configure 'Sonde Test', attempt 3```
indicates that it cannot configure reporting - this is usually done during pairing, the sensor may have time to fall asleep and therefore does not receive a command. try to wake it up by briefly pressing the pairing button until the configuration is completed.

Loic691, 2023-03-02:
> @Loic691 Hi. do you really have the latest z2m 1.30.2? I checked my legrand battery remote - it works fine, does not left network.
>
> `Failed to configure 'Sonde Test', attempt 3` indicates that it [...]

no 1.30.1

Loic691, 2023-03-02:
I don't saw the last one !
Update to 1.30.2 is done
Waiting so !

Loic691, 2023-03-02:
@kirovilya 
with 1.30.2 I can't publish state on Legrand or Hue device...
`Publish 'set' 'state' to 'Eclairage Piscine' failed: 'Error: Command 0x0004740000a1a14c/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'`

kirovilya, 2023-03-02:
@Loic691 hmmm, try rejoin it. or record herdsman log

Loic691, 2023-03-02:
Humm
I restart Z2M and it's now OK...
Strange

Hedda, 2023-03-02:
@Loic691 this is the wrong repository to post about issues with Zigbee2MQTT.  Please post in zigbee-herdsman repo issue instead -> https://github.com/Koenkk/zigbee-herdsman/issues/319