grobasoz/zigbee-firmware #28: Firmware compatibility with ITead's new “ZBDongle-E” Zigbee 3.0 USB Dongle Plus adapter based on the EFR32MG21?

Hedda, 2022-08-07:
@grobasoz  Can you tell if existing firmware releases are compatible with ITead's new “ZBDongle-E” that is also based on EFR32MG21?

https://itead.cc/product/zigbee-3-0-usb-dongle/

I guess my question is if it this new “ZBDongle-E” dongle board is pin-compatible with the first/old ITead bareboe Zigbee 3.0 USB Dongle (model "9888010100045")  as well as on CoolKit Technologies “SM-011 V1.0” module used in the ITead Sonoff ZBBridge Zigbee Bridge?

[image]

Here is by the way link to FCC filing/report for the new “ZBDongle-E” (a.k.a. "ZBD-E" ) adapter which contains copy of the user manual + a lot more detailed + many more internal and external pictures:

[https://fccid.io/2APN5ZBD-E](https://fccid.io/2APN5ZBD-E)

[https://fccid.io/2APN5ZBD-E/User-Manual/User-Manual-6012461.pdf](https://fccid.io/2APN5ZBD-E/User-Manual/User-Manual-6012461.pdf)

[https://fccid.io/2APN5ZBD-E/Internal-Photos/Internal-Photos-6012455](https://fccid.io/2APN5ZBD-E/Internal-Photos/Internal-Photos-6012455)

[https://fccid.io/2APN5ZBD-E/External-Photos/External-Photos-6012451](https://fccid.io/2APN5ZBD-E/External-Photos/External-Photos-6012451)

It at least looks to use the same [EFR32MG21A020F768IM32 (EFR32 Mighty Gecko Series 2) SoC](https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs/device.efr32mg21a020f768im32?tab=specs) used on the first ITead Zigbee 3.0 USB Dongle, as well as on CoolKit Technologies "SM-011 V1.0" module used in th ITead Sonoff ZBBridge Zigbee Bridge and in Tuya ZS3L and Tuya ZSLC5 SoC modules. It includes a 80 MHz ARM Cortex-M33 core, 768 Flash Storage, 64 RAM, in a QFN32 package, which radio has + 20 dBm maximum power output and receive sensitivity of -104 (250 kbps O-QPSK DSSS) dBm. 

There are further details and more users discussion here:

https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-mcu-now-sold-for-19-99/442695

and here:

https://github.com/Koenkk/zigbee2mqtt/discussions/13373

[image]

By the way, news is that a similar Silabs EFR32MG21 SoC will also be used in the official "Home Assistant SkyConnect USB Stick", see:

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594

You can submit your interest in the official "Home Assistant SkyConnect USB Stick" here if want to get mailed updates on that dongle:

https://docs.google.com/forms/d/e/1FAIpQLScEjHSBJszUZfgO3MIDO51IHr3Oeohcs8BLpRIjY1liJ58IpA/viewform

grobasoz, 2022-08-07:
@Hedda - thank for the information :) 

I can't tell from the photos but if the pinout for the TXD/RXD is the same (PA5, PA6) then it should work with my firmware.

As a side note - the new dongle seems to FCC'd for 5.6mW only... 

Regards,
Gary.

Hedda, 2022-08-09:
FYI, I asked ITead was got the reply that the pinout definition for the EFR32MG21 SoC chip on the new model ZBDongle-E board is the same as the original first-generation barebone Zigbee 3.0 USB Dongle model 9888010100045 (which I understand is also pin-compatible with the "SM-011 v1.0" Zigbee module from CoolKit-Technologies that in use on the board original first-generation Sonoff ZBBridge Zigbee Bridge), but they answered that the new model ZBDongle-E adapter can still not use the exact same firmware images as their old old Model 9888010100045 adapter, so I guess that must mean that they do not use the exact same EFR32MG21 SoC chip or?

I also asked ITead about the default RF output power configuration and was told that the new ZBDongle-E adapter currently ships with Zigbee NCP 6.10.3 firmware that has enabled 20 dBm output RF power default, however is not hardcoded in the firmware so users/applications can choose to lower output RF power if they want.

In addition, got part of Configuration Parameter values Table for Silicon Labs EmberZNet PRO Zigbee networking protocol stack:

Configuration Parameter | Value
-- | --
EMBER_APS_UNICAST_MESSAGE_COUNT | 32
EMBER_PACKET_BUFFER_COUNT | 250
EMBER_NEIGHBOR_TABLE_SIZE | 26
EMBER_SOURCE_ROUTE_TABLE_SIZE | 200
EMBER_ADDRESS_TABLE_SIZE | 32

It does not mention "Child Table Size" but I guess that is probably configured to 32 same as for old model 9888010100045 as per https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/EZSP/README.md or?

grobasoz, 2022-08-09:
Thanks @Hedda - so if they have working NCP firmware do I need to create any? Or do you still want RCP firmware for this board?

Hedda, 2022-08-09:
> if they have working NCP firmware do I need to create any? 

ITead ship it with a working EmberZNet Zigbee NCP 6.10.3 firmware image, (however they not released image file for download).

Though might be nice to also have compatible EmberZNet Zigbee NCP 7.0.x and 7.1.x firmware images for testing as well?

>Or do you still want RCP firmware for this board?

RPC firmware image is still wanted by the community for experimental multiprotocol support as per -> https://github.com/zigpy/zigpy/discussions/894

MattWestb, 2022-08-09:
@Hedda I is missing one very impotent question:
Do the new hardware need the fix for the not calibrated HXO in the firmware or have they calibrating the chip/soc as Silabs papers saying it shall being done ?

Hedda, 2022-09-01:
> I is missing one very impotent question:
> Do the new hardware need the fix for the not calibrated HXO in the firmware or have they calibrating the chip/soc as Silabs papers saying it shall being done ?

@Daniel-zhan-itead Did you manage to get an answer about HFXO Capacitor Bank (CTune) calibration from the engineer?

I know that the first barebone updated firmware had to use a custom CTUNE value of 128 as otherwise the performance was not good at all, as per this discussion here -> https://github.com/xsp1989/zigbeeFirmware/issues/4

More information on that is available on HFXO Capacitor Bank (CTune) calibration here:

https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2019/03/18/hfxo_capacitor_bank-7uRt

https://www.silabs.com/documents/public/application-notes/an0016.2-efr32-series-2-oscillator-design-considerations.pdf

https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2019/01/22/saving_ctune_valuea-qCPl

Hedda, 2022-09-01:
@grobasoz Do you have the possibility to make an EmberZNet Zigbee NCP 7.0.x firmware image for this "ZBDongle-E" adapter?

FYI, EmberZNet EmberZNet SDK 7.0.2.0 GA (7.0.2) have now also been released as a bug-fix release:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.0.2.0.pdf

Daniel-zhan-itead, 2022-09-09:
@Hedda We followed the Silicon Labs official guide for radio antenna calibration tuning for ZBDongle-E, the default CTune value is 133. Compared with the 128 CTune value, it is very close with only a +-2ppm difference, both are within the threshold limit.

MattWestb, 2022-09-09:
@Daniel-zhan-itead Is the CTune sett in manufacture token in the chip flash as shall being made by Silabs paper for manufacturing devices or is the firmware "patched" for fixing it (as the old one) ?

Daniel-zhan-itead, 2022-09-09:
> @Daniel-zhan-itead Is the CTune sett in manufacture token in the chip flash as shall being made by Silabs paper for manufacturing devices or is the firmware "patched" for fixing it (as the old one) ?

In the next firmware update, we will update the CTune value to 128 for the best performance. We are collecting the optimized items and update together.

MattWestb, 2022-09-09:
= the device is **not** calibrated from the factory as shall being made by Silabs manufacturing paper.

God that doing the calibrating by patching the firmware then then first gen was having large problems (before patching CTune) keeping sleeping end device online then it was miss tuning the channel and losing the network.

And for the moment we can expecting user having problem paring new devices and keeping sleepers online until the new firmware is released.

Daniel-zhan-itead, 2022-09-09:
> = the device is **not** calibrated from the factory as shall being made by Silabs manufacturing paper.
>
> God that doing the calibrating by patching the firmware then then first gen was having large [...]

Referring to the Silabs manufacturing paper, do you mean these two docs? or others?

https://www.silabs.com/documents/public/application-notes/an0016.2-efr32-series-2-oscillator-design-considerations.pdf
https://community.silabs.com/s/article/kba-bt-1308-saving-ctune-value-as-manufacturing-token?language=en_US

MattWestb, 2022-09-09:
Nop.
The first is only design and the second is for BT.

In the https://www.silabs.com/documents/public/application-notes/an700-1-mfg-test-guidelines-efr32.pdf and point  `2.4.2 Transmit Frequency Offset Test`.

`Different capacitor values can be written to registers to observe the corresponding frequency offset, and the CTUNE manufacturing token is used to store the value for use by the application.`

KB Using CTUNE token https://community.silabs.com/s/article/ctune-access?language=en_US.

And writing the manufacturing token look in the instruction Silabs commander for flashing it in the chip device property https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf `6.1.6 Flash Tokens`.

And also AN961: Bringing Up Custom Devices for the EFR32MG and EFR32FG Families  https://www.silabs.com/documents/public/application-notes/an961-custom-nodes-efr32.pdf.

Hedda, 2022-09-09:
@Daniel-zhan-itead It would probably also be a good idea for ITead to offer both EmberZNet 6.10.3 and 7.0.2.0 firmware images?

Zigbee EmberZNet 6.10.3.0 GA as a "stable" release.

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.3.0.pdf

and 

Zigbee EmberZNet 7.0.2.0 GA as a "beta" release.

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.0.2.0.pdf

and perhaps even Zigbee EmberZNet 7.1.0.0 GA as an "alpha" release.

https://jp.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.0.0.pdf

PS: Zigbee EmberZNet 7.1.x.x is so far only wanted by developers that want to experiment with EZSP v9 support, see:

https://github.com/grobasoz/zigbee-firmware/issues/25

Hedda, 2022-09-09:
> I guess my question is if this new “ZBDongle-E” dongle board is probably pin-compatible with the first/old ITead barebone Zigbee 3.0 USB Dongle (model "9888010100045"), as well as on CoolKit Technologies “SM-011 V1.0” (ZYZBP008) module used in the "ITead Sonoff ZBBridge Zigbee Bridge" as well as the unbranded ZB-GW04 USB dongle (a.k.a. "easyiot stick") ? -> https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

> By the way, from what read I understand that both the board and the firmware that they built for this new ”ZBDongle-E” adapter were not designed or developed by the same engineer(s) or development team nor firmware developers as ITead’s previous/old Zigbee 3.0 dongle barebone model "9888010100045" adapter from ITead (that was also based on EFR32MG21) -> https://www.facebook.com/watch/?v=262086502015726

@Daniel-zhan-itead Can you by the way confirm if ITead's new "ZBDongle-E" is pin-compatible with ITead’s previous/old Zigbee 3.0 dongle barebone model "9888010100045" USB adapter?

Also maybe confirm if ITead's new "ZBDongle-E" was designed by other engineers/developers than the "9888010100045" adapter?

[kirovilya](https://github.com/kirovilya) mentioned that he managed to successfully flash the 7.1.1  firmware image from https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP to his "ZBDongle-E" adapter from ITead and got it working with the new EZSP v9 protocol version in his develop builds of Zigbee2MQTT -> https://github.com/Koenkk/zigbee-herdsman/issues/319

https://github.com/Koenkk/zigbee-herdsman/issues/319

Daniel-zhan-itead, 2022-09-13:
> > I guess my question is if this new “ZBDongle-E” dongle board is probably pin-compatible with the first/old ITead barebone Zigbee 3.0 USB Dongle (model "9888010100045"), as well as on CoolKit Technologies “SM-011 V1.0” (ZYZBP008) module used in the "ITead Sonoff ZBBridge Zigbee Bridge" as well as the unbranded ZB-GW04 USB dongle (a.k.a. "easyiot stick") ? -> https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP
> 
> > By the way, from what read I understand that both the board and the firmware that they built for this new ”ZBDongle-E” adapter were not designed or developed by the same engineer(s) or development team nor firmware developers as ITead’s previous/old Zigbee 3.0 dongle barebone model "9888010100045" adapter from ITead (that was also based on EFR32MG21) -> https://www.facebook.com/watch/?v=262086502015726
> 
> @Daniel-zhan-itead Can you by the way confirm if ITead's new "ZBDongle-E" is pin-compatible with ITead’s previous/old Zigbee 3.0 dongle barebone model "9888010100045" USB adapter?
> 
> Also maybe confirm if ITead's new "ZBDongle-E" was designed by other engineers/developers than the "9888010100045" adapter?
> 
> [kirovilya](https://github.com/kirovilya) mentioned that he managed to successfully flash the 7.1.1 firmware image from https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP to his "ZBDongle-E" adapter from ITead and got it working with the new EZSP v9 protocol version in his develop builds of Zigbee2MQTT, see his comments here -> [Koenkk/zigbee-herdsman#319 (comment)](https://github.com/Koenkk/zigbee-herdsman/issues/319#issuecomment-1241668554) and here ->[Koenkk/zigbee-herdsman#319 (comment)](https://github.com/Koenkk/zigbee-herdsman/issues/319#issuecomment-1241133124)

Yes, it is compatible and users can flash the firmware as you attached if they want.

Hedda, 2022-09-13:
@grobasoz So then EFR32MG2x-768k (Series 2 768k Flash) firmware images from @grobasoz  should also be compatible? -> https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32MG2x-768k

Daniel-zhan-itead, 2022-09-15:
> @Daniel-zhan-itead then EFR32MG2x-768k (Series 2 768k Flash) firmware images from @grobasoz should also be compatible? -> https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32MG2x-768k

Honestly, we did not verify and test this firmware so I can not give you a certain answer. I will try to contact him and send him a ZBDongle-E for testing or development.:)

Hedda, 2022-09-15:
> Honestly, we did not verify and test this firmware so I can not give you a certain answer. I will try to contact him and send him a ZBDongle-E for testing or development.:)

@Daniel-zhan-itead  Nice! Ah, I see also now that @grobasoz did not list the pads used in the config as in his other readme files:

https://github.com/grobasoz/zigbee-firmware/blob/master/EFR32MG2x-768k/README.md

grobasoz, 2022-09-15:
@Daniel-zhan-itead  - Sue Su from Itead has made contact with me re sending a new dongle for testing... We'll see how it goes :) Thanks Gary.

grobasoz, 2022-10-13:
Have ZBDongle-E from ITead here - will start testing soon.

MattWestb, 2022-10-13:
Sound great !!

User have getting problem with devices that is shipped with 6.10 and being upgraded to 7.x and then downgraded to 6.10 and then the token storage in the NVM3 is mismatching very fundamental and the APP is not starting on the chip.
Trace is NMV3 handling panic.

ITead have one workaround by upgrading to unstable latest 7.1 and the user can using the device and hopefully getting it fixed in near future not braking the NVM3 by firmware app versions.
(can being that  7.x is never back capability with older but i have not seen any Silabs warning in GSDK release.)

If you getting it try erasing the flash pages for the NVM3 with Commander and i think its working.

By the way great see you have time doing little free time work and not 24/7 hard coding and for you beloved customers that demanding more and hard and software for there systems !!

And one thanks to @Daniel-zhan-itead for supporting the community and i think in  the end its good for ITead and the user.

grobasoz, 2022-10-13:
Hello Vienna! I have asked ITead for a schematic so I don't need to reverse engineer the pinouts (still waiting). I tested the ZBDongle-E with latest Zigbee2MQTT and it worked well enough - but didn't do thorough testing with many devices yet. 
I am not sure what firmware is required? Silabs have released SDK4.1.2 so I am updating to that version and testing.
Also trying to finalise new Coordinator hardware so I can get some assembled...

MattWestb, 2022-10-13:
### Hola Gary (still Spanish feeling here in the north) !!

I think you have you have enough info here: https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP.

I dont knowing what its bringing making more firmware then reading the issue https://github.com/xsp1989/zigbeeFirmware/issues/28#issuecomment-1274037357 and they is digging and hopefully getting it fixed but its interesting then very likely other hardware is getting the same problems with 6.10 and 7.x up and downgrading with bootloader.

Z2Ms EZSP Barista (Italian Coffey **Chef**) have getting the last pieces in place and using hashed link keys as ZHA is doing and looks very promised for all EM35X and EFR32 user with real Zigbee 3 paring and install code support !!

Latest GSDK have deleting all NCP examples for MG1P and MG1B but it was never supported and i was not  getting it  working for my Billys (problem getting the com working with  SW flow control). 
RCP is still in but i have not testing coking one in the last update but the earlier is working very good with OTBR and zigbeed but have bugs in zigbeed with SEDs (no firmware bugs but its in the host application).

For My is 6.7.x the golden NCP and the next best is 6.10.x also for MG2X chips!!

New coordinator with MG24 ?? 
(I think MG21) with original Silabs module or self made "chiped" ?
One good USB chip and easy accessed Silabs MINI with PTI ??

You need some up to date for testing and getting one good hardware for you consumer that like getting update to the next gen and having one good working NCP for it. Can also being 6.7.X or 6.10.X if being more stable for you systems.
I think for "normal" devices is MG21 little over kill but in the end its the customer that is paying the end bill.

grobasoz, 2022-10-13:
@MattWestb - My MG24 still being tested for FCC :( Too many people doing USB-Zigbee. My new coordinator is a little different ;)

grobasoz, 2022-10-13:
ITead has given me permission to share the following from the ZBDongle-E.
PA0 = Boot
PC0 = Red LED
PB1 = EFR32-TX 
PB0 = EFR32-RX 

RTS and DTR are used to set boot pin and reset the chip (eg put device into serial bootloader mode).
Will have a play with flashing and testing tomorrow...

Hedda, 2022-10-14:
> I tested the ZBDongle-E with latest Zigbee2MQTT and it worked well enough - but didn't do thorough testing with many devices yet.
> I am not sure what firmware is required? Silabs have released SDK4.1.2 so I am updating to that version and testing.

FYI, EZSP v8 is stable and EZSP has experimental support in edge/dev version, see https://github.com/Koenkk/zigbee-herdsman/issues/319

@grobasoz Maybe you could send @kirovilya a prototype of your adapter?

Carko, 2022-10-21:
@grobasoz any success with flashing yet?

grobasoz, 2022-10-22:
@Carko - Yes, flashing worked OK with both versions (6.10.3 and 7.1.0) and I could do some basic testing using the EZSP release of Zigbee2MQTT. Silabs have released a new version (SDK4.1.3) so I am working on that next.

Hedda, 2022-10-24:
If you have inclination then please also consider building an optional dedicated Zigbee Router firmwire for this device since I know many are intersted in using a few of these as Zigbee network mesh routers (signal repeaters / range extenders), used in USB-chargers for power and then joined/paired, because of their price to performance ratio. 

ITead/Sonoff do even offer such Zigbee Router firmware image build but not sure about its firmware version and default button mappings for putting into join/pairing mode again or resetting to default settings if need to re-pair/re-join it(?).

grobasoz, 2022-10-24:
@Hedda - I can put this request "in the pipeline" - getting rather full now 😮. I would like to make it an SDK4 release as Silabs are throwing many eggs into that basket and I've had no end of trouble with SDK3 (still using SDK2 with my customers!). Maybe even get SLC-CLI working with it?
Regarding pairing/joining etc.

It has one button and one led and the most commonly used "commissioning" I've seen has been...
1. Short press - try join if not joined.
2. Short press - message router with some "alive" message.
3. Long press - factory reset if not joined (5 secs or more).
4. Long press - leave network if joined.

The Led indicated "joined" state.
1. Off - not powered.
2. Alternating/toggling - powered, not joined.
3. Short flash, solid on or showing network traffic - powered, joined.

However you need to open the case to access the button and led? This makes automation necessary... So adding an "auto join" feature - or Bluetooth interface?

Hedda, 2022-10-24:
> However you need to open the case to access the button and led? This makes automation necessary... So adding an "auto join" feature - or Bluetooth interface?

Suggest always putting it into join mode for maybe around 60-seconds when giving it power if not already joined to a network.

Hedda, 2022-10-24:
> Edit>One other "thing". The device would need a "supplier", "name" and "type" and potentially a device converter for the likes of Home Assistant.

Not sure that ITead/Sonoff use now as default in their router images?

https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router

You could use theirs, or pick your own made-up vendor, or use 'Custom devices (DiY)' which multiple different DIY projects use. 

I do not think that Home Assistant ZHA integration will require a custom quirk for ZHA Device Handlers library as it should work out-of-the-box, but as you say that would require a custom converter for zigbee-herdsman (used in ZIgbee2MQTT and IoBroker). 

See this example:

https://github.com/digiblur/Tasmota/blob/development/zigbee_router/silabs.js

* vendor: 'Custom devices (DiY)'
* description: 'Silabs EFR32MG21 Router'

and see other existing converters

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/custom_devices_diy.js

https://digiblur.com/2022/02/20/how-to-convert-the-sonoff-zigbee-bridge-into-a-router-repeater/

https://community.home-assistant.io/t/inofficial-zigbee-router-firmware-released-for-itead-sonoff-zbbridge-and-zb-gw03-zigbee-gateways-bridges-as-alternative-when-used-with-tasmota-and-or-esphome/393733

grobasoz, 2022-10-24:
Sonoff uses "zigbeeModel = DONGLE-E_R" and "vendor = SONOFF" and the device is supported...
[image]

It tries to join a network automatically every 2 seconds (many times - I gave up after a few minutes). Power is set to 20dBm!

Button does nothing. LED exposes a "level" which doesn't seem to change?
[image]

Hedda, 2022-10-28:
> I do not think that Home Assistant ZHA integration will require a custom quirk for ZHA Device Handlers library as it should work out-of-the-box, but as you say that would require a custom converter [...]

FYI, there is a related docs PR for zigbee2mqtt.io related to that mentioned converter example for custom DIY router firmware:

https://github.com/Koenkk/zigbee2mqtt.io/pull/1661

It mentions its pairing setup and some more stuff for the Zigbee Router firmware by xsp1989:

https://github.com/Koenkk/zigbee2mqtt.io/pull/1661/commits/b9f05d18e43234f0bc7bea33d1225cf439a0f09c

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

closed 2024-08-12 by grobasoz