grobasoz/zigbee-firmware #40: [REQUEST] Silabs Zigbee EmberZNet SDK 7.2 NCP experimental builds for testing?

Hedda, 2023-01-23:
I missed this but Silicon Labs have on December 15, 2022 released Gecko SDK  (GSDK) 4.2.0 with Zigbee EmberZNet SDK 7.2.0.0 GA:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.0

Any chance you would consider releasing some experimental builds if [zigpy](https://github.com/zigpy/zigpy/discussions) and [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/issues/319) developers want to test it?

That is,  EmberZNet 7.2.0.0 is probably not ready for general use yet but great if firmware images were available to beta-test for issues.

It looks to use EZSP v9 (EmberZNet Serial Protocol version 9) same as Zigbee EmberZNet 7.1

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.0.0.pdf

New features in it are mostly related to Dynamic Multiprotocol and multi-PAN 802.15.4 in RCP mode:

- Secure key storage support for MG2x parts that support Secure Vault-High
- Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode
- Dynamic Multiprotocol Zigbee-NCP + Bluetooth-NCP (experimental)
- Manufacturing Library (MfgLib) support for Concurrent Multiprotocol RCP
- Zigbee + OpenThread Concurrent Lis-tening on MG24 parts (experimental)
- Zigbee on Host (ZigbeeD) support for 32 bit and 64 bit x86 architecture (experimental)
- MGM240S SiP Module Support

They also added simultaneous Dual-Band and Dual-PHY for Zigbee Smart Energy on MG24/MG12:

- MG24+Si4468 Dual-PHY Zigbee Smart Energy support
- MG12 Dual-Band 2.4GHz + Sub-GHz Zigbee Smart Energy support

Deprecated in release 7.2.0.0:

- The Secure EZSP feature will be removed in a future release.

PS: Silicon Labs has also released Gecko SDK Version 4.1.4.0 as a stable bug-fix release containing Zigbee EmberZNet SDK 7.1.4.0:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.1.4

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.4.0.pdf

grobasoz, 2023-01-24:
Sure, can make a 7.2.0 version...

Hedda, 2023-02-15:
FYI, read someone spotted that EZSP protocol version has been incremented to EZSP v10 in Gecko SDK v4.2 / Zigbee EmberZNet v7.2 and that EZSP v10 was not mentioned in the latest UG100 (EZSP Reference Guide) documentation from Silicon Labs which SiLabs only updated for EZSP v9 and state that no additions was added between EmberZNet PRO Release 7.1.1 and 7.2.0:

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.2/protocol/zigbee/app/util/ezsp/ezsp-protocol.h

https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf)

Gecko SDK v4.2:
* https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.2/protocol/zigbee/app/util/ezsp/ezsp-protocol.h
* EZSP_PROTOCOL_VERSION = 10

Gecko SDK v4.1:
* https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.1/protocol/zigbee/app/util/ezsp/ezsp-protocol.h
* EZSP_PROTOCOL_VERSION = 9

Gecko SDK v4.0:
* https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.0/protocol/zigbee/app/util/ezsp/ezsp-protocol.h
* EZSP_PROTOCOL_VERSION = 8

For reference, Nabu Casa is already officially offering downloads for both "stable" EmberZNet 7.1.x.x (EZSP v9) firmware images as well as "beta" EmberZNet 7.2.x.x (EZSP v10) firmware image for their recently released Home Assistant SkyConnect adapter/dongle/stick:

https://www.home-assistant.io/skyconnect/

https://github.com/NabuCasa/silabs-firmware/tree/main/EmberZNet

https://github.com/NabuCasa/silabs-firmware/tree/main/EmberZNet/beta

xsp1989 is also offering EmberZNet 7.1.x.x (EZSP v9) and EmberZNet 7.2.x.x (EZSP v10) firmware image builds for his ZB-GW04 USB stick (a.k.a. "easyiot Zigbee dongle") that he designed based on "SM-011 V1.x" (ZYZBP008) radio module from CoolKit-Technologies (which in turn has an EFR32MG21 SoC). 

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

https://www.aliexpress.com/item/1005002791666029.html

PS: xsp1989's firmware images for ZB-GW04 are also compatible with Sonoff ZBDongle-E by ITead even though not optimized for it:

https://itead.cc/product/zigbee-3-0-usb-dongle/

Hedda, 2023-04-17:
FYI, latest bug-fix version is now Silicon Labs Zigbee EmberZNet SDK 7.2.2.0 from Gecko SDK version 4.2.2.0 from March 10, 2023:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.2.0.pdf

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.2

grobasoz, 2023-04-17:
@Hedda - Not sure why this would be necessary. Seems like host side of development (Z2M, ZHA, Tasmota etc) is lagging behind EZSP versions?

Hedda, 2023-04-17:
>  Seems like host side of development (Z2M, ZHA, Tasmota etc) is lagging behind EZSP versions?

That is not true any longer.

bellows radio library for zigpy has already been updated to support EZSP v11:

* [Zigbee Home Automation (ZHA) integration for Home Assistant](https://www.home-assistant.io/integrations/zha) (which as you probably know depends on [zigpy](https://github.com/zigpy/zigpy)).
* Domoticz via the [Zigbee for Domoticz Plugin](https://www.domoticz.com/wiki/ZigbeeForDomoticz) (see [Domoticz-Zigbee](https://github.com/zigbeefordomoticz/Domoticz-Zigbee), which is also depends on [zigpy](https://github.com/zigpy/zigpy)).
* Jeedom via [official Zigbee plugin for Jeedom ](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/) (which is also depends [zigpy](https://github.com/zigpy/zigpy) as well).

zigbee-herdsman is not lagging far behind as has already been updated to EZSP v10 (and EZSP v11 is a very minor change):

* [Zigbee2MQTT](https://www.zigbee2mqtt.io/ ) (that is depending on [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/issues/319)).
* [ioBroker](https://github.com/ioBroker/ioBroker.zigbee) (ioBroker adapter for Zigbee devices that also depends on [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/issues/319)).

The only large project that was lagging behind is OpenHAB but its dependency was at least updated to EZSP v10 last month:

* OpenHAB via official [OpenHAB ZigBee Binding](https://www.openhab.org/addons/bindings/zigbee/) (that depends on [com.zsmartsystems.zigbee](https://github.com/zsmartsystems/com.zsmartsystems.zigbee))

PS: Yes I read that the [Zigbee2Tasmota project](https://tasmota.github.io/docs/Zigbee/) is lagging behind those others because their lead Zigbee developer is currently fully focusing on support for the new Matter standard in Tasmota, which in turn might later also lead to a wish for him wanting to add Thread / OpenThread support in Tasmota in order to support Matter over Thread with native Thread Border Router / OpenThread Border Router:

https://github.com/arendst/Tasmota/discussions/17872

Hedda, 2023-05-05:
FYI, Silabs Zigbee EmberZNet SDK	7.2.3.0 has now been released as part of Silicon Labs Gecko SDK (GSDK) Version 4.2.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.3.0.pdf

Silicon Labs Zigbee EmberZNet 7.2.3.0 release states that it is specifically targeting quality improvements and bug fixes.

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.3

grobasoz, 2023-05-05:
@Hedda - thanks - I have installed 4.2.3 now so will do some testing :)

closed 2024-08-12 by grobasoz

Hedda, 2024-08-12:
@grobasoz should this and the others maybe have been closed as ”not planned” instead of as ”completed? Closing as ”completled” mean that they are done while closing as ”not planned” means they will never be done.