grobasoz/zigbee-firmware #41: [REQUEST] Utilize new common "Silicon Labs firmware builder" automated build system from Home Assistant / Nabu Casa developers to create reproducible builds of Zigbee and Thread firmware images

Hedda, 2023-02-10:
Please consider making use of the "Silicon Labs firmware builder" which is a new open-source firmware build system that Home Assistant founders have now among other things announced as a common script tool set for building firmware images for Silabs based hardware.

https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/#building-firmware-for-silicon-labs-chips

(Note that this is only just one section from https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/ )

In summary, they have open-sourced and released the ""Silicon Labs firmware builder" project which is a collection of automated build scripts to help make it easier to create reproducible firmware builds for Zigbee and Thread based SoCs from Silicon Labs. 

Please consider making use of this firmware build system and if possible also contribute any improvements to upstream project:  

https://github.com/NabuCasa/silabs-firmware-builder

As I understand it, right now it allows you to build either Zigbee EmberZNet NCP (Zigbee Coordinator on Silicon Labs Network Co-Processor) firmware images, OpenThread RCP (Radio Co-Processor), or RCP Multi-PAN firmware images for OTBR (multi-protocol Zigbee EmberZNet and OpenThead Border Router using Radio Co-Processor) profiled for their own Home Assistant SkyConnect USB stick/dongle and their EFR32MG21-based radio module integrated into the Home Assistant Yellow appliance, however, the idea is that you can build profiles for other Silicon Labs hardware adapters/dongles/sticks/modules as well. It could of course also be extended to not only support different Silicon Labs based hardware adapters/dongles/sticks/modules but also different type of firmware images, (like maybe OpenThead Border Router on NCP).

(Examples of firmware image builds that it can produce can be found here -> https://github.com/NabuCasa/silabs-firmware ).

Note that was just one of their announcements in that blog post; in related news, they also released "SL Web Tools" which enable flashing firmware to Silabs-based adapters directly from a web page as well as a "SiLabs multi-flasher" as a Python script flashing tool:

https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/#silabs-multi-flasher--sl-web-tools

(Again note that this is only just one section from https://www.home-assistant.io/blog/2023/02/08/state-of-matter-and-thread/ )

https://github.com/NabuCasa/universal-silabs-flasher

https://github.com/NabuCasa/sl-web-tools

https://www.youtube.com/watch?v=-88K23e8XYw&t=2s&ab_channel=HomeAssistant

The "[SL Web Tools](https://github.com/NabuCasa/sl-web-tools)" project is available as open source. It allows users to manage the firmware on Zigbee/Thread sticks using SiLabs chips directly from their browser, with no terminals or compilation necessary. This allows any creator to offer easy firmware updates to their users and make experimental updates available to test bug fixes quickly."

To see it in action, SL Web Tools for flashing firmware has been added to the Home Assistant SkyConnect documentation.

https://skyconnect.home-assistant.io

SL Web Tools is powered by another new open-source Python package “silabs-universal-flasher” → [GitHub - NabuCasa/universal-silabs-flasher: Flashes Silicon Labs radios running EmberZNet or CPC multi-pan firmware](https://github.com/NabuCasa/universal-silabs-flasher)

For reference; the firmware build system for Silicon Labs was made by Home Assistant developer/engineer and Nabu Casa employee [Stefan Agner (a.k.a. agners](https://github.com/agners), who also developed the EFR32MG21-based [Home Assistant SkyConnect USB stick/dongle](https://www.home-assistant.io/skyconnect) as well as the EFR32MG21-based radio module integrated into the Home Assistant Yellow appliance for Rasberry Pi 4 Compute Module). And the "universal-silabs-flasher" firmware tool was made by [puddly](https://github.com/puddly) who is also one of the lead developers of [zigpy ](https://github.com/zigpy/zigpy)and the [radio libraries for zigpy](https://github.com/zigpy) that is used in Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) (among other projects).

PS: They made updates to their documentation as relevant to Silicon Labs based adapters with correct firmware for Zigbee or Thread:
- https://www.home-assistant.io/integrations/thread
- https://www.home-assistant.io/integrations/matter
- https://www.home-assistant.io/integrations/zha
- https://skyconnect.home-assistant.io/

@Daniel-zhan-itead

grobasoz, 2023-02-10:
Very interesting. I'll be sure to have a look when I get back to the office.

Hedda, 2023-03-14:
FYI, ksjh has begun some work on such automated firmware building using Github Actions and started a related discussion on it:

https://github.com/ksjh/silabs-firmware-builder

https://github.com/ksjh/silabs-firmware-builder/discussions/2

Note that most of the automation magic looks to happen in Github Actions workflows:

https://github.com/features/actions

https://docs.github.com/en/actions

Hedda, 2023-08-15:
FYI, [darkxst](https://github.com/darkxst) is not only continuing to improve his fork of silabs-firmware-builder and he has now also forked the Silabs Flasher add-on for Home Assistant and is working on making it compatible with many different Silicon Labs based USB adapters, see:

* https://github.com/darkxst/multipan_flasher (which in turn depends on https://github.com/NabuCasa/universal-silabs-flasher)
* https://github.com/darkxst/silabs-firmware-builder
  * https://github.com/darkxst/silabs-firmware-builder/blob/manifestjs/manifests
* https://github.com/darkxst/silabs-builder-remote
* https://github.com/darkxst/silabs-test
* https://github.com/darkxst/sl-test (fork of https://github.com/NabuCasa/sl-web-tools)
  * https://github.com/darkxst/sl-test/tree/gh-pages
    * https://darkxst.github.io/sl-test/

PS: Another tip is to also check out his blog -> https://dialedin.com.au/blog/

grobasoz, 2023-09-20:
Thanks @Hedda - Will have a look through! Gary.

Hedda, 2024-11-07:
FYI, many other downstream devs are now collaborating on improving upstream, including added Bluetooth and OpenThread RCP:

* https://github.com/NabuCasa/silabs-firmware-builder

PS: Upstream will soon also migrate from Gecko SDK Simplicity SDK which will bring larger changes to dependecies and builds.

* https://github.com/NabuCasa/silabs-firmware-builder/pulls

grobasoz, 2024-11-07:
Thanks @Hedda - Always good to hear from you! 
At this stage I don't see a need for me to continue with Zigbee Firmware releases, there seems to be plenty of solutions available now.