grobasoz/zigbee-firmware #43: [REQUEST] Silabs Zigbee EmberZNet SDK 7.3 NCP experimental builds for testing?

Hedda, 2023-06-19:
Silicon Labs have on the 7th of June 2023 released Gecko SDK (GSDK) v4.3 / 4.3.0 with Zigbee EmberZNet SDK v7.3 / 7.3.0.0 GA:

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.0.0.pdf

https://www.silabs.com/wireless/zigbee?tab=software#software

Any chance would consider releasing some experimental builds if zigpy and zigbee-herdsman developers want to test EmberZNet 7.3?

Reference -> https://github.com/zigpy/bellows/issues/563 and https://github.com/Koenkk/zigbee-herdsman/issues/319

While brand new and not yet supported in any open-source Zigbee gateway solution, early cutting-edge can help pre-release testing.

Normal end-users should be warned against flashing this firmware was is the EZSP protocol version been incremented to EZSP v12 and Zigbee gateway solutions such as Home Asssitant's ZHA integration and Zigbee2MQTT are still only compatible with EZSP v11 and earlier EZSP versions at this time.

EZSP protocol version been incremented to EZSP v12 in Silicon Labs Gecko SDK v4.3 (4.3.0.0) / Zigbee EmberZNet v7.3 (7.3.0.0):

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.3/protocol/zigbee/app/util/ezsp/ezsp-protocol.h#L33

0x0C hexadecimal = 12 in decimal.

### Sidenote - EmberZNet 7.3 is the first Silabs firmware release to be "Zigbee PRO 2023" compatible

Silicon Labs EmberZNet 7.3.0.0 is the first Silabs SDK that offers Zigbee R23 compliance, with it making their Silabs Zigbee stack compatible with Zigbee Protocol Specification Revision 23 / R23 (also known as "Zigbee PRO 2023" and "Zigbee 2023" ).

**Zigbee**
- Zigbee R23 compliance, with these Security enhancements among others:
  - Dynamic link key negotiation
  - Device interview to query devices before they are allowed to join
  - Trust Center Swap Out to replace an existing Trust Center with a new one
  - Frame Counter Synchronization
- Zigbee Direct Device (ZDD) support for:
  - Onboarding/commissioning
  - Communication to all Zigbee devices without a hub (Alpha), using Bluetooth LE
- Zigbee Smart Energy 1.4a compliance (Alpha)
- Enhancements to Zigbee Green Power (ZGP) APIs
- New Zigbee Security upgrade component for moving encryption keys from cleartext NVM3 tokens into secure storage

https://www.silabs.com/blog/zigbee-pro-2023-spec-released-increases-security

https://csa-iot.org/newsroom/zigbee-pro-2023-improves-overall-security-while-simplifying-experience/

https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

Reference:

https://github.com/zigpy/zigpy/issues/1211

Hedda, 2023-06-20:
Heads-up! "**_Default flow control for Openthread changed in v4.3.0 from Hw to HwUart_**" -> https://github.com/NabuCasa/silabs-firmware-builder/pull/18

Hedda, 2023-08-18:
FYI, Zigbee EmberZNet 7.3.1.0 SDK has been released as part of Gecko SDK (GSDK) Version 4.3.1.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.1.0.pdf

Hedda, 2023-10-27:
FYI, Zigbee EmberZNet 7.3.2.0 SDK has been released as part of Gecko SDK (GSDK) Version 4.3.2.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.2.0.pdf

https://github.com/SiliconLabs/gecko_sdk/releases

MattWestb, 2023-10-27:
FYI:
https://github.com/zigpy/zigpy/discussions/894#discussioncomment-7252364
https://github.com/NabuCasa/silabs-firmware/commit/b89f14925ddd83ac62df34cb715afc75c476b644
But the addon is not updated then Puddly is digging very deep for finding some strange problems.

grobasoz, 2023-10-27:
@Hedda - Yes I have 7.3.2 and testing it now. There still seem to be issues present so I've not upgraded any customer firmware. Silabs are investigating some problems with Nabu Casa but I have a few different problems that I'm investigating with my own customer. At this stage my go to releases are still based on 6.7.8 - though 6.10.3 looks promising!

Hedda, 2023-11-23:
> At this stage my go to releases are still based on 6.7.8 - though **6.10.3 looks promising**!

You mean 6.10.9.0 ? ;) - https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.9.0.pdf

closed 2024-08-12 by grobasoz