grobasoz/zigbee-firmware #44: [REQUEST] "Zigbee CLI" and "Zigbee RCP" firmware for ESP32-H2 (similar to ESP32-C6)

Hedda, 2023-08-15:
@grobasoz  just noticed that you added some Zigbee RCP and CLI firmware images for ESP32-C6. 

* https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6
  * https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/ZigbeeCLI
  * https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/Zigbee_RCP

Can you please consider also providing similar Zigbee RCP and CLI firmware images for ESP32-H2?

As I understand the ESP32-H2 is very similar to ESP32-C6, with the major difference being that ESP32-H2 does not have a WiFi radio.  ESP32-H2 has not been made available until now so firmware support for it lags behind the ESP32-C6 even though it is actually newer.

Espressif recently announced their ESP Thread Border Router / Zigbee Gateway development board based on ESP32-H2, see here:

https://www.cnx-software.com/2023/06/20/espressif-esp-thread-border-router-board-combines-esp32-h2-esp32-c3-wireless-chips/

https://www.hackster.io/news/espressif-launches-a-kit-to-turn-its-esp32-modules-into-certified-thread-border-routers-for-matter-c61db0fce050

https://github.com/espressif/esp-thread-br

https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/

https://www.aliexpress.us/item/1005005688193617.html

Als, for reference, an earlier two-board ESP Thread Border Router solution was previously available as example gateway/bridge project:

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br

https://openthread.io/guides/border-router/espressif-esp32

This type of two-SoC setup setup could initially be useful in a proof-of-concept as a remote network-attached Zigbee Coordinator 
where the ESP32-S3 would be flashed with for example ESPHome or Tasmota and the ESP32-C6 would only run Zigbee/OpenThread firmware to simply act as a Zigbee/Thread radio connected via serial using CLI or RCP interface. 

PS: DamKast recently also announced "zigpy-zboss" as an (unofficial and experimental) ZBOSS radio library for zigpy -> https://github.com/kardia-as/zigpy-zboss and that could in theory make zigpy compatible with ESP32-C6/ESP32-H6 if their firmware use the standard ZBOSS API/CLI which if it is the case should make it usable as a remote network-attatched Zigbee Coordinator for Home Assistant's ZHA integration if configure via serial server pass-through as proxy using socat (similar to ser2net) -> https://github.com/zigpy/zigpy/issues/1052 (More zigpy-zboss discussion related to zigpy here -> https://github.com/zigpy/zigpy/issues/394)

grobasoz, 2023-08-15:
Hello @Hedda - Good to hear from you again :) Thanks for all the excellent updates. I must say I missed a few!

I initially only purchased the C6 dev kits (and radio modules) as they do both Zigbee and WiFi (albeit not at the same time) and I thought it would be interesting to see if the RCP firmware would work.
Suffice to say both the Zigbee (Matter -> Zigbee Bridge) and Openthread (Border Router) RCP "hosts" worked very well. That's why I posted the firmware. Espressif Matter over WiFi is excellent as well.

I will have to buy some H2 dev boards for testing but was going to do that anyway - when I had some spare time. I'll put an order in for the kits and have a play when they arrive.
Keep well, Gary.

Hedda, 2023-08-16:
Nice! I do not think that Espressif has yet ramped up production of the ESP32-H2 as they have started to do for the ESP32-C6 so availability will probably be scarce however should hopefully be possible for interested developers to at least get a hold of radio modules if not the development boards. Right now I could only find the ESP Thread Border Router/Zigbee Gateway board (plus optional Ethernet Sub Board for it) and the ESP32-H2-DevKitM-1 dev kit board in stock on Aliexpress but not that two-SoC board or just the radio modules.

ESP Thread Border Router/Zigbee Gateway board
* https://www.aliexpress.com/item/1005005688193617.html
* https://www.aliexpress.us/item/1005005802317569.html
* https://www.aliexpress.us/item/1005005887333709.html
* https://www.aliexpress.us/item/1005005798326291.html
* https://www.aliexpress.us/item/1005005717245286.html

ESP32-H2-DevKitM-1 dev kit board :
* https://www.aliexpress.com/item/1005005252175587.html
* https://www.aliexpress.us/item/1005005468645439.html
* https://www.aliexpress.us/item/1005005802321644.html
* https://www.aliexpress.us/item/1005005844479281.html

I believe that both ESP32-C6 and ESP32-H2 with Ethernet PHY (for wired LAN interface) could be very interesting for many to make Zigbee-to-Ethernet and Thread-to-Ethernet bridge/gateway solutions for stable connections to a user's network. Similar to the open-source LAN-based hardware products from TubesZB https://github.com/tube0013/tube_gateways and ZigStar https://github.com/mercenaruss/zigstar_gateways which are relatively popular in several home automation communities.

CSA (Connectivity Standards Alliance) which governs the new Matter standard l have made it clear that bridges for protocol-XYZ to Matter will play an extremely important role in the adoption of Matter as the default norm as many people will still want to keep using their old devices but bridge them to a Matter fabric.

FYI, read that Tasmota have for example already implemented official initial support of the Matter standard since Tasmota v13 and its developers are currently working on also implementing bridge mode for Matter, though not yet started on Thread support, see https://tasmota.github.io/docs/Matter/ (and https://github.com/arendst/Tasmota/discussions/17872)

That concept would go in line with Tasmota's existing Zigbee2Tasmota (Z2T) support -> https://tasmota.github.io/docs/Zigbee/

Also hoping that ESPHome developers will soon follow with both support Zigbee and Thread on ESP32-C6/ESP32-H2 as well as support of the Matter standard in order to have ESPHome based Zigbee-to-Ethernet and Thread-to-Ethernet bridges too as then they will make such solutions more popular (with ESPHome now being backed by Nabu Casa and the Home Assistant founders).

Hedda, 2023-08-17:
> I initially only purchased the C6 dev kits (and radio modules) **as they do both Zigbee and WiFi (albeit not at the same time)**

I though that ESP32-C6 want ment to support the "**_Coexistence of Wi-Fi and 802.15.4_**" as per?-> https://github.com/espressif/esp-idf/issues/10423

By the way, see in esp-idf update from today that they also added "**_External coexistence - Supported_**" which I assume is Ethernet?

_Originally posted by @jack0c in https://github.com/espressif/esp-idf/issues/10423#issuecomment-1681545160_

>   **_updated:_**
> - [x] **_External coexistence - Supported_**

closed 2024-08-12 by grobasoz