grobasoz/zigbee-firmware #47: Update firmware for EM357 on the Wink Hub 1

RoganDawes, 2023-09-17:
Hi folks,

I am trying to upgrade the OS installed on the Wink Hub 1 to a modern OpenWrt build, and in the process, am trying to make sure that the various radios are up to date and functioning properly. I see that there is currently a file `em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-560.ota` on the device which would appear to be the firmware for the EM357 zigbee peripheral.

Is there a matching updated firmware available, and how would I install it? The debug pins (JTAG/SWD) are broken out, so if needed I can go down to the metal to flash it or recover.

When running /usr/sbin/ZigBeeHACoord, I can see that the conversation with the dongle looks like (very abbreviated log, I can post more if needed), and the device responds (at 57600):

```
> 1970/01/01 04:26:33.961875  length=5 from=0 to=4
 1a c0 38 bc 7e
< 1970/01/01 04:26:34.051594  length=1 from=0 to=0
 11
< 1970/01/01 04:26:35.031594  length=7 from=1 to=7
 1a c1 02 0b 0a 52 7e
> 1970/01/01 04:26:35.041656  length=12 from=5 to=16
 80 70 78 7e 00 42 21 a8 50 ed 2c 7e
< 1970/01/01 04:26:35.051219  length=11 from=8 to=18
 01 42 a1 a8 50 28 15 e4 5c 53 7e
```

However, when trying to expose the tty via socat to the network for e.g. zigbee2mqtt to talk to it, I see:

```
socat -x TCP-L:20108,reuseaddr,keepalive,nodelay,keepintvl=1,keepcnt=5 /dev/ttySP4,b57600
> 1970/01/01 04:00:42.414687  length=1 from=0 to=0
 ef
> 1970/01/01 04:00:43.429031  length=5 from=1 to=5
 fe 00 21 01 20
> 1970/01/01 04:00:49.431749  length=5 from=6 to=10
 fe 00 21 01 20
> 1970/01/01 04:00:55.436844  length=5 from=11 to=15
 fe 00 21 01 20
```

and the dongle doesn't respond.

That suggests that the controlling programs are speaking a different protocol (and zigbee2mqtt is speaking the wrong one).

I am very new to the zigbee world, so would appreciate a pointer in the right direction if possible.

Many thanks!

MattWestb, 2023-09-17:
Old EM35X firmware is using 57600 baud and also normal hardware flow control but some software.
Also i  think (not 100% sure) that Z2M is only supporting EZSP protocol version 8 and newer but i have not testing if older is working with it (ZHA do).
It shall being possible hocking the bootlodare mode with https://github.com/NabuCasa/universal-silabs-flasher (working with ERF32MG1X chips also over Ethernet) shall being possible with all strange modes like RCP with very high baud rate and rebooting the APP to bootloadre with software commands.
If you have one debugger it  can being good reading the internal flash before start testing other firmware so can going back to the original working version.

RoganDawes, 2023-09-17:
Thanks. I'll dig into the universal flasher (can that run via a TCP connection, or does it play games with baud rates?), and see if I can get the JTAG/SWD port working.

I did also try with ZHA, and got the following:
```
socat -x TCP-L:20108,reuseaddr,keepalive,nodelay,keepintv
l=1,keepcnt=5 /dev/ttySP4.real,b57600
> 1970/01/01 06:08:38.618249  length=5 from=0 to=4
 1a c0 38 bc 7e
< 1970/01/01 06:08:38.710875  length=1 from=0 to=0
 11
< 1970/01/01 06:08:39.690811  length=7 from=1 to=7
 1a c1 02 0b 0a 52 7e
> 1970/01/01 06:08:39.711000  length=8 from=5 to=12
 00 42 21 a8 50 ed 2c 7e
< 1970/01/01 06:08:39.720750  length=11 from=8 to=18
 01 42 a1 a8 50 28 15 e4 5c 53 7e
> 1970/01/01 06:08:39.723750  length=4 from=13 to=16
 81 60 59 7e
```
But ZHA reported an unknown error when trying to add the integration via Home Assistant.
Is there a simple command to report the version that I can turn into a /bin/dash script? My interim OpenWrt environment is just a ramdisk, and there is only 64MB total on the board, leaving very little space for more software.

MattWestb, 2023-09-17:
If not getting the flasher working you can trying bellows CLI it can reading the version and also the config of the firmware and also some is possible with zigpy CLI  https://github.com/zigpy/zigpy-cli. best from one different system like desktop Linux or Win.

I think the flasher is not trying the `57600` and need being added in the command line for making it working.

Also ZHA can having problems if the baud rate or flow control is not standard and must using manual configuration if the integration.

grobasoz, 2023-09-18:
@RoganDawes - I have a Wink system somewhere in storage so could have a look. Nice hardware I seem to recall... but I think when the subscription service started most people just tossed theirs in the trash?
I have tried Z2M (latest) with my repository firmware (V6710, 115k2 baud, software flow control) running on EM357 (my hardware) and works fine (so far - with limited testing). 
Fortunately most EM357's use the same serial port pins. I did find hardware flow control was preferred when using 115k2 baud but also found software flow control worked fine with more powerful gateways.

RoganDawes, 2023-09-19:
@grobasoz i did manage to get a basic "info" query to return results, with a socat instance redirecting it to the network.

(On my phone, with no backtick, I'm afraid!)

# socat -x TCP-L:20108,reuseaddr,keepalive,nodelay,keepintvl=1,keepcnt=5 /dev/ttySP4.real,b57600
$ zigpy radio ezsp socket://192.168.1.183:20108/ info
PAN ID:                0x4728
Extended PAN ID:       df:ae:7e:e7:76:89:79:a1
Channel:               24
Channel mask:          [11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
NWK update ID:         0
Device IEEE:           00:0d:6f:00:03:2b:ed:35
Device NWK:            0x0000
Network key:           78:62:92:c4:2f:99:5e:32:d5:74:c8:e9:f2:d3:ef:b7
Network key sequence:  0
Network key counter:   565249

So at least *something* can talk to it remotely.

I'm still working on the GPIO's required to power on the radios in my own kernel, currently I get no response at all.

Would be nice to have someone else able to replicate my results, or interested in getting the Wink 1 (and eventually 2) working.

grobasoz, 2023-09-20:
Seems I have a Wink Hub 2 - beautiful design! 
It runs a few radios, but more importantly, a Silabs EM3587 for Zigbee. The EM3587 also has a PA/LNA so would require "enabling", normally PA3 or PC6.
The question is whether there is enough of a need or interest in spending (precious) time on getting this system to work?

sreknob, 2023-09-21:
I think that there are hundreds of Wink Hub 2s waiting patiently in boxes to be freed! (Including mine).
But time is precious and radios are cheap, so a hard call to make!

RoganDawes, 2023-09-21:
The main problem is that the Wink Hub 2 is afflicted by High Assurance Boot, meaning that running your own code is difficult to impossible.

Nonetheless, I am trying to see if such a thing is possible, as the CPU (depending on silicon revision) has a known bug that allows a buffer overflow while checking the code signature.

I will update here if I manage to get my own code running.

MattWestb, 2023-09-21:
If its the Silicon Labs Security Advisory A-00000476 you thinking of its only have getting fixes for GSDK 4.X and EM35X and MG1X NCP is not supported on this version then need EZSP 6.X (GSDK 3.2 or earlier).
So as long SL is not updating the GSDK 3.2 the possibility is there but is not well documented how to using it.

RoganDawes, 2023-09-21:
No, I'm talking about a hardware vulnerability in the iMX6UL CPU that runs the Linux OS on the Wink Hub v2.

closed 2024-08-12 by grobasoz