grobasoz/zigbee-firmware #48: ESP Zigbee NCP (Espressif ESP NCP host) firmware for ESP32-C6 and/or ESP32-H2?

Hedda, 2024-01-22:
@grobasoz have you considered building ESP Zigbee NCP (Espressif ESP NCP host) firmware for ESP32-C6 and/or ESP32-H2 SoCs?

https://github.com/espressif/esp-zigbee-sdk/issues/124

You already built ESP Zigbee RCP firmware but ESP Zigbee NCP is less complictated for projects like example zigpy/ZHA to make use of:

https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/Zigbee_RCP

ESP32-C6 NCP host can be implemented as a single SoC solution while the ESP32-H2 needs another external chip for interface, like for example either USB-to-UART/Serial converter/bridge chip (e.i. CP210x, FT231X, CH9102F or CH340B) to make a simple USB-adapter or a ESP32-C3 for a network-attached adapter (ie. Zigbee remote adapters similar to https://github.com/tube0013/tube_gateways and https://github.com/mercenaruss/zigstar_gateways ).

Looking at recent commits in their esp-zigbee-sdk SDK looks like they have begun working on implementing an NCP host component:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ncp

https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_ncp/README.md

https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-ncp

https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/application.rst

https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/api-reference/esp_zigbee_ncp.rst

PS:  FYI, it looks like Espressif themselves have one of their developers working on a zigpy radio library for `esp-zigbee-ncp` API/CLI:

https://github.com/zigpy/zigpy-cli/pull/42

https://github.com/zigpy/zigpy/issues/1052

grobasoz, 2024-01-24:
@Hedda - Always good to hear from you :) I started looking into a version of NCP that would work with (Silabs) NCP using a 'C6 but found they were quite different. That was however some time ago so there may have been updates... Will have another look when time permits.

Hedda, 2024-02-06:
> FYI, it looks like Espressif themselves have one of their developers working on a zigpy radio library for `esp-zigbee-ncp` API/CLI:
> 
> [zigpy/zigpy-cli#42](https://github.com/zigpy/zigpy-cli/pull/42)
> 
> [zigpy/zigpy#1052](https://github.com/zigpy/zigpy/issues/1052)

@grobasoz FYI, Espressif has now made an experimental version of zigpy-espzb available in this repo (but it is not yet on PyPI):

https://github.com/zigpy/zigpy-cli/pull/45

https://github.com/lhespress/zigpy-espzb

grobasoz, 2024-02-06:
@Hedda - Thanks for that - I'll see if I can have a look. Gary.

Hedda, 2024-08-12:
@grobasoz closed as completed?

grobasoz, 2024-08-14:
@Hedda - Yes, should be "not planned" but I couldn't find that option when closing?

grobasoz, 2024-08-14:
@Hedda - Options exists on web... will close now, thanks.

closed 2024-08-14 by grobasoz