grobasoz/zigbee-firmware #51: SilverCrest Zigbee Gateway Model TYGWZ-01

Danieldiazi, 2024-05-04:
Hi I've followed this link: [https://zigbee.blakadder.com/Lidl_TYGWZ-01.html](https://zigbee.blakadder.com/Lidl_TYGWZ-01.html) in ordert to use the SilverCrest Zigbee with Home Assistant and I've upgraded the EZSP Version to 6.7.8.0 (firmware: https://github.com/grobasoz/zigbee-firmware/raw/master/EFR32%20Series%201/EFR32MG1B-256k/NCP/NCP_UHW_MG1B232_678_PA0-PA1-PB11_PA5-PA4.gbl)

I wonder if there are any chance to upgrade to a newer version?

Thanks in advance for your support!

joaquinvacas, 2024-05-05:
> Hi I've followed this link: https://zigbee.blakadder.com/Lidl_TYGWZ-01.html in ordert to use the SilverCrest Zigbee with Home Assistant and I've upgraded the EZSP Version to 6.7.8.0 (firmware: [...]

Hey! Interested too.

AFAIK the support for 256k chips was dropped which I think makes difficult to have newer EZSP versions on our coordinators.
But maybe there's hope, at least a more modern version than the 6.7.8.0 FW, I think there is an untested 6.10.x firmware (https://github.com/darkxst/silabs-firmware-builder/issues/22#issuecomment-1797901731 ) but I'm not sure about how it will work.

Anyhow, I've bought a SLZB-06M coordinador to replace mine (So I can use Zigbee2MQTT + new "ember" driver) but I'll keep the LIDL one for testing so if a newer firmware comes, I won't care about trying it without messing with my home's config.

joaquinvacas, 2024-05-08:
> @Danieldiazi @joaquinvacas - Unless you are specifically having problems with 6.7.8 I'd suggest sticking with it. We have noticed quite a few issues with later versions of NCP, especially on the older chips. As you know, the NCP is just a small part of the software, basically a Radio Controller, the "brains" are in the Host application (eg Zigbee2MQTT (experimental still?), ZHA and Tasmota to name a few). Most problems I've encountered have been with the Host Application. If you still want to try a layer version, I could try set some time aside to creating a version for 6.10.9...

It seems on the newer Z2M "ezsp" driver is becoming deprecated in favor of "ember" driver.

Anyhow, it seems that the latest version send some commands that are unknown/incompatible for the older EZSP firmwares and seems that 6.10.x firmwares work fine, at least, at the moment.

https://github.com/Koenkk/zigbee2mqtt/issues/22444#issuecomment-2096504275

Danieldiazi, 2024-05-08:
> @Danieldiazi @joaquinvacas - Unless you are specifically having problems with 6.7.8 I'd suggest sticking with it. We have noticed quite a few issues with later versions of NCP, especially on the [...]

Thanks for your answer and your support.
My concern is that this version could become obsolete, for example, this week we had problems upgrading to version 2024.5.0.  (as you can see in the comments: https://www.home-assistant.io/blog/2024/05/01/release-20245/#backward-incompatible-changes)

Except for this specific case, I haven't any problems and I am very happy with the firmware.

Thanks!

grobasoz, 2024-05-09:
@joaquinvacas - So you would like a V6.10.9 version of NCP_UHW_MG1B232_678_PA0-PA1-PB11_PA5-PA4.gbl? 
I ask as it takes a while to get the hardware setup for testing...

titoum-be, 2024-05-09:
@grobasoz that would be pretty nice as otherwise zigbee2mqtt will kind of drop us behind :-/

MattWestb, 2024-05-09:
@titoum-be Its still V8 protocol so its no difference.

Danieldiazi, 2024-05-11:
> Hey! Interested too.
>
> AFAIK the support for 256k chips was dropped which I think makes difficult to have newer EZSP versions on our coordinators.
> But maybe there's hope, at least a more modern [...]

Boas/Hi:

Can you share your opinion and experience about SLZB-06M device?

joaquinvacas, 2024-05-11:
> @joaquinvacas - So you would like a V6.10.9 version of NCP_UHW_MG1B232_678_PA0-PA1-PB11_PA5-PA4.gbl? I ask as it takes a while to get the hardware setup for testing...

Don't worry, unless somebody else is interested too, don't want to bother you and waste your time with this FW :)

> @titoum-be Its still V8 protocol so its no difference.

Regarding this, seems that some of the commands are accepted on 6.10.x

> > > Hi I've followed this link: https://zigbee.blakadder.com/Lidl_TYGWZ-01.html in ordert to use the SilverCrest Zigbee with Home Assistant and I've upgraded the EZSP Version to 6.7.8.0 (firmware: https://github.com/grobasoz/zigbee-firmware/raw/master/EFR32%20Series%201/EFR32MG1B-256k/NCP/NCP_UHW_MG1B232_678_PA0-PA1-PB11_PA5-PA4.gbl)
> > > I wonder if there are any chance to upgrade to a newer version?
> > > Thanks in advance for your support!
> > 
> > 
> > Hey! Interested too.
> > AFAIK the support for 256k chips was dropped which I think makes difficult to have newer EZSP versions on our coordinators.
> > But maybe there's hope, at least a more modern version than the 6.7.8.0 FW, I think there is an untested 6.10.x firmware ([darkxst/silabs-firmware-builder#22 (comment)](https://github.com/darkxst/silabs-firmware-builder/issues/22#issuecomment-1797901731) ) but I'm not sure about how it will work.
> > Anyhow, I've bought a SLZB-06M coordinador to replace mine (So I can use Zigbee2MQTT + new "ember" driver) but I'll keep the LIDL one for testing so if a newer firmware comes, I won't care about trying it without messing with my home's config.
> > Hola/Hi:
> 
> Can you share your opinion and experience about SLZB-06M device?

Didn't receive it yet!

Gunthervc, 2024-06-04:
Hi,
Also interested here.
Will there be an update? to be possible?
I have 3 coordinators from Lidl. 1 in the house, 1 garage and 1 in the conservatory.
otherwise I have to buy 3 new ones :(

SergioMuinhos, 2024-06-04:
Hi,
Also interested here.
Will there be an update? to be possible?
I have 3 coordinators from Lidl. 1 in the house, 1 garage and 1 in the conservatory.
otherwise I have to buy 3 new ones :(

grobasoz, 2024-06-05:
I spent a few days trying to get the firmware updated to a later version but had no luck. There may be other developers who have managed to generate a suitable application (eg [darkxst](https://github.com/darkxst/silabs-firmware-builder/issues/22#issuecomment-1797901731)) and that's probably the best option. 
I also don't have the hardware to test on, and just generate code "in the blind". Best not use that code for a system that is already working...
If this is still very important, I'll take another run at it but as mentioned above, there are some limitations with the older Series 1 chips and Silabs' support for them.

rubendediegomartinez, 2024-07-22:
Hi,
I'm interested in the new one version 6.10.x.
Any one tried this version?
I can do it. I have got a GW in production and another one to test any change.
But I would thank any support and help
BR

Danieldiazi, 2024-07-25:
I just received SLZB-06M device and I'll change my zigbee system to this new coordinator.

closed 2024-08-12 by grobasoz

mbjd05, 2024-09-27:
I would also be very interested in a new firmware version, especially as LIDL now sells a new version with usb-c (but otherwise unchanged internals) for relatively cheap. To me, it thus seems like a very relevant budget gateway device still. I would love to be able to use it with the new ember driver. Mine will arrive in a week or two, let me now if you reconsider trying to build this firmware and you need someone to help test it.

gpaesano, 2025-02-21:
have a look we have solved the issue and an ember version is now available.
https://github.com/Koenkk/zigbee2mqtt/discussions/22520#discussioncomment-12282505