grobasoz/zigbee-firmware #52: Request for Help with EFR32MG13

maodun96, 2024-06-06:
hello grobas,

I am a new Zigbee user and I have encountered some issues, and I hope to request your assistance.

I have a board that runs the Android system, and its Zigbee chip is the EFR32MG13. 
[image]
I have installed Termux on it. Using the command from the elelabs-zigbee-ezsp-utility, python3 Elelabs_EzspFwUtility.py restart -m btl -p /dev/ttyS5, the feedback I get is:

Generic Zigbee EZSP adapter detected:
Firmware: 6.5.5-176
EZSP v7
Launch in bootloader mode
Generic Zigbee EZSP adapter detected:
Firmware: 6.5.5-176
EZSP v7

python3 Elelabs_EzspFwUtility.py flash -f efr32mg13p-v8-6910-115200.gbl -p /dev/ttyS5,the feedback I get is:

Generic Zigbee EZSP adapter detected:
Firmware: 6.5.5-176
EZSP v7
Launch in bootloader mode
Generic Zigbee EZSP adapter detected:
Firmware: 6.5.5-176
EZSP v7
EZSP adapter not in the bootloader mode. Can't perform update procedure

It seems that the chip did not enter the Bootloader mode as per the command. Does this mean that this chip does not have a Bootloader? Do I have any other methods to write a new firmware?

I have already successfully carried out similar operations on another board that uses the [TYZS13](https://developer.tuya.com/en/docs/iot/tyzs13-module-datasheet?id=K98u373pszqe3) (EFR32MG13P) module, and have flashed the relevant firmware. These two boards seem to only differ slightly in their Zigbee chip.

I am not sure if it is appropriate for me to seek your help here. If you are able to assist me, I would be greatly appreciative. If you are not comfortable with this issue, I will close it.

I apologize for the inconvenience.

Best regards

grobasoz, 2024-06-06:
It is nighttime here so I will try having a look for you tomorrow morning. There are two options to put the device into boot loader mode, via a port pin and via a command over ezsp.

maodun96, 2024-06-06:
> It is nighttime here so I will try having a look for you tomorrow morning. There are two options to put the device into boot loader mode, via a port pin and via a command over ezsp.

Thank you very much, I look forward to your reply.

grobasoz, 2024-06-07:
@maodun96 - I've had a closer look at your post and don't think it would be easy to activate the bootloader by GPIO pin, so that just leaves the software approach. Basically the Elelabs code send a message to the EZSP to enter bootloader mode and my concern is that it is not handling it correctly.

"Normally"  EZSP based Silabs devices have a bootloader installed, even on the MG13's. It is possible to create EZSPs without the bootloader but I doubt anyone does that. Since the EZSP responds to the Elelabs code, you're halfway there...

If we can get the bootloader start, then worst case scenario is you use the SWD pins to flash the code... This is quite straightforward as I see the SWDIO, SWCLK and GND are available as pads. You may need to do some soldering :) You'll also need a SWD programmer but they are cheap enough if you don't have one...

I'll have to set up an old MG13 device (with V6.5.5.EZSP) and test the Elelabs code to see why it's failing... that will take some time unfortunately.

grobasoz, 2024-06-07:
@maodun96 - One other thing, can I have the full part number on the EFR32 please. eg EFR32MG13Px32Fxxx

grobasoz, 2024-06-07:
@maodun96 - I have connected the Elelabs_EzspFwUtility.py to my system here (running EZSP 6.6.3  - earliest I have available), using your command above, and it works as expected. I'll try find (or build) a 6.5.5 version later today.

maodun96, 2024-06-07:
@grobasoz 
I apologize for the delayed response. Thank you for your reply.

1.On the chip, I can only see some printed text: 
EFR32MG13
LEED
2215D00WMU
 I'm unable to determine the exact part number, and perhaps I should use some tool to query it?

2.Since the original Android system has been flashed, I'm unable to know how it was originally controlled. I suspect it was controlled using RTS/CTS, as in the blog post (https://blakadder.com/t6e-z2m/), the author used "115200" firmware to flash the TYZS13 chip.
[image]

The board I have is likely a variant of the one the author had, as they were produced by the same manufacturer and have similar functionality.
One of the boards has the TYZS13 chip, which is the EFR32MG13P, while the others have a red PCB with an EFR32MG13 chip, and I believe their usage and methods should be similar.
Alternatively, you could suggest a tool to help me test the control method.

3.Your mention of "EZSP without a loaded program" seems very plausible. As this product was likely designed to not have firmware upgrades.

I apologize if my response doesn't contain much useful information. This is an unfamiliar domain for my technical level, but I will do my best to try. 
Thank you again for your assistance.

maodun96, 2024-06-07:
> @maodun96 - I have connected the Elelabs_EzspFwUtility.py to my system here (running EZSP 6.6.3 - earliest I have available), using your command above, and it works as expected. I'll try find (or build) a 6.5.5 version later today. ![EZSP_Bootlaunch](https://private-user-images.githubusercontent.com/5369713/337520807-917941ef-0f25-4006-b84f-b5cf5ffadb95.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc3NDg2NTUsIm5iZiI6MTcxNzc0ODM1NSwicGF0aCI6Ii81MzY5NzEzLzMzNzUyMDgwNy05MTc5NDFlZi0wZjI1LTQwMDYtYjg0Zi1iNWNmNWZmYWRiOTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDdUMDgxOTE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTc1MDQzZjdlNzRiMTNkMzZhYmQ3YjUzZTFkMDZmNGRjNDBhYTQ3NTM5NmNlY2FiYTA1MWJhMTE0NjZkNzVjYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.CIYjTOJX-RsmylbPawZvg5Y3kZ5chHcxHS9GRy3HCBY)
> 
> One thing I'm not sure about is whether your system (EZSP) uses software or hardware flow control? Are you able to find out - or try both options??

I don't think the failure should be related to the version, because I have already achieved firmware upgrade on the TYZS13 module, and at that time its version was Firmware: 6.5.0-188.
[image]
~/elelabs-zigbee-ezsp-utility $ python3 Elelabs_EzspFwUtility.py probe -p /dev/ttyS5
2024/06/06 06:53:40 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/06 06:53:40 Elelabs_EzspFwUtility:   Firmware: 6.5.0-188
2024/06/06 06:53:40 Elelabs_EzspFwUtility:   EZSP v7
~/elelabs-zigbee-ezsp-utility $ python3 Elelabs_EzspFwUtility.py restart -m btl -p /dev/ttyS5
2024/06/06 06:54:16 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/06 06:54:16 Elelabs_EzspFwUtility:   Firmware: 6.5.0-188
2024/06/06 06:54:16 Elelabs_EzspFwUtility:   EZSP v7
2024/06/06 06:54:16 Elelabs_EzspFwUtility:   Launch in bootloader mode
2024/06/06 06:54:26 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2024/06/06 06:54:26 Elelabs_EzspFwUtility:   Gecko Bootloader v1.8.0
~/elelabs-zigbee-ezsp-utility $ python3 Elelabs_EzspFwUtility.py flash -f efr32mg13p-v8-6910-115200.gbl -p /dev/ttyS5
2024/06/06 06:58:31 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2024/06/06 06:58:31 Elelabs_EzspFwUtility:   Gecko Bootloader v1.8.0
2024/06/06 06:58:31 Elelabs_EzspFwUtility:   Allready in bootloader mode. No need to restart
2024/06/06 06:58:32 Elelabs_EzspFwUtility:   Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)
.....

2024/06/06 06:59:14 Elelabs_EzspFwUtility:   Firmware upload complete
2024/06/06 06:59:14 Elelabs_EzspFwUtility:   Rebooting NCP...
2024/06/06 06:59:21 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/06 06:59:21 Elelabs_EzspFwUtility:   Firmware: 6.9.1-214
2024/06/06 06:59:21 Elelabs_EzspFwUtility:   EZSP v8

maodun96, 2024-06-07:
Here are photos of my two motherboards
[image]
[image]

grobasoz, 2024-06-07:
@maodun96 - OK, I understand what you have tried now. The MG13P on the red PCB is a different radio on the same (or similar) Android host - and it fails to enter bootload mode.

I actually don't know why the Elelabs utility can probe the firmware/device, but not go into bootload on the "red" radio. It may be some small issue with the communications (flow control)? As mentioned I have no issues with the Elelabs python app here but it is very basic and I haven't tried flow control.
It could be that the Android system allows bootloader entry to be activated using the flow control pins (like ESP32 auto flash using RTS and DTR) - I just don't know. 

If you can't get the command to set bootloader mode (over serial), you could try using the TXD and RXD pins on the PCB (like blakadder seemed to do?). That requires soldering. Failing that, flashing via SWD is the next option. Also soldering.

maodun96, 2024-06-07:
> @maodun96 - OK, I understand what you have tried now. The MG13P on the red PCB is a different radio on the same (or similar) Android host - and it fails to enter bootload mode.
>
> I actually don't [...]

Yes, I am currently using the CH340 to connect the TXD, RXD, and GND pins, but I plan to still use the Elelabs utility. However, I suspect that I will likely still be unable to enter the bootloader.
[image]

maodun96, 2024-06-07:
@grobasoz 
I wired it according to "TX | RX; RX | TX; VCC | 3V3; GND | GND", but I got the same result - I was still unable to enter the bootloader mode using the Elelabs utility.

C:\Users\d4357\Downloads\elelabs-zigbee-ezsp-utility-master>Elelabs_EzspFwUtility.py probe -p COM3
2024/06/07 18:22:28 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/07 18:22:28 Elelabs_EzspFwUtility:   Firmware: 6.5.5-176
2024/06/07 18:22:28 Elelabs_EzspFwUtility:   EZSP v7

C:\Users\d4357\Downloads\elelabs-zigbee-ezsp-utility-master>Elelabs_EzspFwUtility.py restart -m btl -p COM3
2024/06/07 18:23:22 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/07 18:23:22 Elelabs_EzspFwUtility:   Firmware: 6.5.5-176
2024/06/07 18:23:22 Elelabs_EzspFwUtility:   EZSP v7
2024/06/07 18:23:22 Elelabs_EzspFwUtility:   Launch in bootloader mode
2024/06/07 18:23:26 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/07 18:23:26 Elelabs_EzspFwUtility:   Firmware: 6.5.5-176
2024/06/07 18:23:26 Elelabs_EzspFwUtility:   EZSP v7

C:\Users\d4357\Downloads\elelabs-zigbee-ezsp-utility-master>Elelabs_EzspFwUtility.py flash -f efr32mg13p-v8-6910-115200.gbl -p COM3
2024/06/07 18:24:24 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/07 18:24:24 Elelabs_EzspFwUtility:   Firmware: 6.5.5-176
2024/06/07 18:24:24 Elelabs_EzspFwUtility:   EZSP v7
2024/06/07 18:24:24 Elelabs_EzspFwUtility:   Launch in bootloader mode
2024/06/07 18:24:28 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2024/06/07 18:24:28 Elelabs_EzspFwUtility:   Firmware: 6.5.5-176
2024/06/07 18:24:28 Elelabs_EzspFwUtility:   EZSP v7
2024/06/07 18:24:28 Elelabs_EzspFwUtility:   EZSP adapter not in the bootloader mode. Can't perform update procedure

grobasoz, 2024-06-08:
@maodun96 - Nicely done! Thanks for the pic. So it seems like the code on the (Red) board doesn't want to (or support) bootloading. I haven't seen this. 

I wrote a python app (for bootloader entry) for you to try.
Call with...
 **_python boot -pComx_** 
Where Comx is the serial port.
Only works with Xon/XOff and 115k2 but you can change the code to suit anything...
[v7boot.zip](https://github.com/user-attachments/files/15746508/v7boot.zip)

If that fails, you may need to flash with a programmer.

maodun96, 2024-06-08:
> @maodun96 - Nicely done! Thanks for the pic. So it seems like the code on the (Red) board doesn't want to (or support) bootloading. I haven't seen this.
>
> I wrote a python app (for bootloader entry) [...]

@grobasoz 
I've tried running your script. Here are the results:

```
C:\Users\d4357\Downloads\elelabs-zigbee-ezsp-utility-master>v7boot.py boot -p COM3
Try Start Bootloader...
NCP no ZigBee Ack.
```

Maybe I still need to use programmer to write,Can CH340 be used as a programmer?
Can you tell me what tool to use for write? I installed SS5, but it's too complicated for me.

Thanks again for your help!

grobasoz, 2024-06-08:
@maodun96 This is the programming tool that I think you would be better off using [SI-DBG1015A](https://www.mouser.com/ProductDetail/Silicon-Labs/Si-DBG1015A?qs=VJzv269c%252BPbMG6jfKVfGLA%3D%3D)
You may need to buy an additional 10pin cable to use for soldering... not sure of the part number...

maodun96, 2024-06-08:
> @maodun96 This is the programming tool that I think you would be better off using [SI-DBG1015A](https://www.mouser.com/ProductDetail/Silicon-Labs/Si-DBG1015A?qs=VJzv269c%252BPbMG6jfKVfGLA%3D%3D) You may need to buy an additional 10pin cable to use for soldering... not sure of the part number... Have a look at this one...[cable](https://www.mouser.com/ProductDetail/HARTING/33182430200064?qs=YAjHENsuY5EO4A27uWzn7A%3D%3D)

Well, this has already entered another field.
I can foresee the difficulties involved, and I think I should stop here.
Thank you for your help these past few days, even though this time ended with me giving up, but I felt your "open-source spirit" which touched me deeply. These past few days have consumed a considerable amount of your energy, and I am truly sorry for that.
I am from China, and my native language is not English, so our conversation has been through a translator. If I have not been polite enough in our dialogue, that was not my intention. Please forgive me.
Finally, I would like to express my utmost respect and gratitude to you once again.

closed 2024-06-08 by maodun96

grobasoz, 2024-06-08:
@maodun96 Sorry to hear that we could not find a suitable solution. I was happy to try helping you with this problem and even with translation your comments and information was excellent. I have many colleagues and suppliers in China so we often need to work with Chinese to English conversations. Good luck with your projects 👍