grobasoz/zigbee-firmware #53: Help needed - MG21A010HI part of ISP router - Arcadyan

kar200, 2024-12-22:
Hi, I posted this alreay on another repo but it seems it's outside of their scope. If it is for this one please deisregard and close this.

https://github.com/NabuCasa/silabs-firmware-builder/issues/96

It's an ISP router with a MG21A010HI chip. The firmware does have 3 GBL files already. THey are DMP (Dynamic Protocol) Zigbee and BLE. The first 2 provide a cli through /dev/ttyH0.

They also provide a binary to flash the firmware directly from the console. so it's easy to reflash and try again.

I tried a few gbls I found online but none seem to work (Through CLI nor any software). I can run an aarch64 chroot as well which makes testing zigbee2mqtt or other software easier.

I understand that I need to figure out the pinout of the chip before building a firmware. I was wondering if someone has any link on how this is done.

The bootloader is 1.A.3 and by checking the strings of the gbl files teres a reference to gecko_sdk 3.1 . I am guessing the bootloader needs to be updated as well.

grobasoz, 2024-12-23:
@kar200 - I'm not sure what you want me to do here? What make is the router? What model number etc. Have you asked the supplier for information...

kar200, 2024-12-23:
Hi @grobasoz ,

Appologies for not being clear.

I am in the process of getting some smart automation happening at home and still learning on how the whole zigbee network happen. Since I use the same router in different places as my dump AP's I was thinking they would work also as Zigbee devices/routers/coordinators. I might have this concept wrong.

I can only interact with the zigbee chip via the cli on the UART port. when I try other software through guides I find online I always get error messages.

I am under the impression the supplied apps use an older SDK {3.1.1 I beleive). I saw here a few requests for newer firmware for different boards and I thought maybe upgrading the zigbee firmware to a newer SDK would make this possible.

In fact even the included apps  in the router firmware don't work no matter which GBL file I load. Only direct interaction with CLI on the UART port let me input commands. 

The list is :
````
$ cat usr/lib/opkg/info/ble_efr32.list    
/usr/include/gecko/libArcBleHello.h
/usr/include/gecko/bg_types.h
/usr/bin/arc_bleshell
/usr/include/gecko/host_gecko.h
/usr/lib/libArcBleHello.so
/usr/include/gecko/bg_gattdb_def.h
/usr/include/gecko/gecko_bglib.h
/usr/include/gecko/uart.h
/usr/bin/launchbootloader-recovery
/usr/bin/arc_bleempty
/usr/include/gecko/gecko_configuration.h
/usr/include/gecko/bg_errorcodes.h
/usr/bin/arc_blehello
````
````
$ cat usr/lib/opkg/info/zigbee_efr32.list
/usr/bin/zbncpversion
/usr/bin/arc_silabs_programmer
/usr/bin/Z3GatewayHost
/usr/bin/launchbootloader-ncp
/usr/bin/zb-serial-bootloader
/usr/bin/arc_silabs_reset
/usr/bin/XncpLedHost
/usr/bin/soc-uart-commander
/usr/bin/arc_silabs_version
````

grobasoz, 2024-12-23:
@kar200 - Thanks for the information. 
Can you send me any of those .h files? Are they the original files?
Do you have the original (NCP) gbl file?
What is the name, make and model of the router?
If I know some of the original information I can create an "ember" version of the NCP for you to use with Z2M.
Z3GatewayHost can be used on original gbl to run a test "Zigbee Coordinator"

kar200, 2024-12-24:
No, thank you @grobasoz!

the files are all the originals. I tried following some howtos to try and build apps but I didn't manage to get far. I only found 1 device with the EFR32MG21A010 online and it's an Ikea device and even then I didn't get far (I guess the pins are different).

The name of the router is Elife. The model name (from what I could gather on the firmware) is WRV9518BHAX34-ET. It's a custom made router using Broadcom BCMBCA (BCM6856) and made by Arcadyan.  I can share the full firmware if you want (and have time to spend on this).

As I said the chip is there and some software is included but there is nothing on the web interface to interact with the chip or run anything.

The firmware is actually quite limited. I used a very hacky way to change the rootfs to EXT4 from squashfs and then disabled most services and only run a few to get the AP mode going.

Files are attached.

Thanks again[
[gecko.zip](https://github.com/user-attachments/files/18239067/gecko.zip)
](url)

kar200, 2024-12-29:
Just adding some info that I got from Simplicity commander and trying to figure out the connections

````
./commander-cli device info -d EFR32  
Reconfiguring debug connection with detected device part number: EFR32MG21A010F512IM32
WARNING: Could not connect to target device
ERROR: Could not connect debugger.
Part Number    : EFR32MG21A010F512IM32
Die Revision   : A1
Production Ver : 34
Flash Size     : 512 kB
SRAM Size      : 64 kB
Unique ID      : 040d84fffe45a836
DONE
````

PA01 -> SWCLK
PA02 -> SWDIO
PA05-> TX
PA06 -> RX

grobasoz, 2024-12-29:
Ok, that's very useful information. I can build an ember NCP that works with Zigbee2mqtt. That's provided that the TX and RX are for the NCP interface. One way to check would be to see what (hex) characters are displayed when the MG21 is reset... commander cli should have a reset function? Baud may be 115k2 or 57k6...

kar200, 2024-12-30:
I have updated the comment above for the baudrate and the serial interface name.

I am assuming it's the same (unless the firmware does not have the NCP serial).

I could not reset using commander for some reason but I could using a command from the firmware itself (arc_silabs_reset) and also by just shorting the reset pin.

If I connect using the serial interface or from the firware the output is the same (see below). Also if I connect using UART to the zigbee chip I cannot access it from the firmware (I am guessing it's the same one unless I am missing something).

````
3LightSoc_PA_010>Reset info: 0x03 (EXT)
Extended Reset info: 0x0301 (PIN)
EMBER_NETWORK_UP 0xA0D9
NWK Steering stack status 0x90
Find and Bind Target: Start target: 0x00
Find and bind target start: 0x00
````
````
Reset info: 0x03 (EXT)
Extended Reset info: 0x0301 (PIN)
EMBER_NETWORK_UP 0xA0D9
NWK Steering stack status 0x90
Find and Bind Target: Start target: 0x00
Find and bind target start: 0x00
````

grobasoz, 2024-12-30:
It looks like a standard Silabs ZigBee Light firmware has been flashed on the chip. If you use the CLI command "info" you'll get some details. That code won't work with Z2M. If you can run Z2M on the device then you can use Silabs NCP code. I'll create it tomorrow...

kar200, 2024-12-30:
Yes that's the zigbee firmware and as I mentioned earlier (and attached the files) there are 3 separate apps (GBL) provided in the firmware itself. DMP, zigbee and BLE.

I tried the 3 of them and as you are confirming now they do not interact with Z2M and 2 of them have a CLI interface.

Anywah thanks a lot for spending time on this.

kar200, 2024-12-30:
I used the info command (if that if of any relevance) and here is the output:
````
MFG String: 
AppBuilder MFG Code: 0x1002
node [(>)9035EAFFFEA1EE64] chan [20] pwr [3]
panID [0x02E1] nodeID [0x1DE2] xpan [0x(>)F17DBBD8DDDAEDC2]
parentID [0xFFFF] parentRssi [0]
stack ver. [6.7.9 GA build 405]
nodeType [0x02]
Security level [05]
network state [02] Buffs: 75 / 75
Ep cnt: 2
ep 1 [endpoint enabled, device enabled] nwk [0] profile [0x0104] devId [0x0100] ver [0x01]
    in (server) cluster: 0x0000 (Basic)
    in (server) cluster: 0x0003 (Identify)
    in (server) cluster: 0x0004 (Groups)
    in (server) cluster: 0x0005 (Scenes)
    in (server) cluster: 0x0006 (On/off)
    out(client) cluster: 0x1000 (ZLL Commissioning)
    in (server) cluster: 0x1000 (ZLL Commissioning)
ep 242 [endpoint enabled, device enabled] nwk [0] profile [0xA1E0] devId [0x0061] ver [0x01]
    out(client) cluster: 0x0021 (Green Power)
Nwk cnt: 1
nwk 0 [Primary (pro)]
  nodeType [0x01]
  securityProfile [0x01]
CLI Finished
````

grobasoz, 2024-12-31:
I have attached the NCP (Software Flow Control) code and bootloader (if required). 
I tested with Z2M and "ember" adapter.
[NCP_USW_EMBER_512.zip](https://github.com/user-attachments/files/18278609/NCP_USW_EMBER_512.zip)

kar200, 2025-01-01:
Happy new year Garry and Thanks for doing this. Sorry a bit late to reply as I was away.

I have converted the hex file to GBL using simplicity commander and flashed it from terminal using the included software. 

I only tested with elelabs python script (in a chroot env) and the output is promising!

````
bash-5.2# python3 Elelabs_EzspFwUtility.py probe -p /dev/ttyH0 -b 115200
2025/01/01 11:46:07 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2025/01/01 11:46:07 Elelabs_EzspFwUtility:   Firmware: 7.4.3-0
2025/01/01 11:46:07 Elelabs_EzspFwUtility:   EZSP v13
`````

I will have to setup the rest later and see how it performs. Anyway thanks again for all the work you have done for everyone on this page.

grobasoz, 2025-01-01:
Thanks and Happy New Year. Glad you had some success!

kar200, 2025-01-05:
Hi Gary,

Hopefully last question  about this subject. Is it possible to obtain a patch to apply to the silabs-firmware-builder and then try different firwmare (I am thinking router vs coordinator) since I have a few of these routers.

I would also like to confirm that the bootloader works. I even created a single package (btl+app) using Simplicity Commander and it updated without an issue.

One last info (in case someone else comes across this issue): I managed to wipe the bootloader on the device I connected with the stlink/converted Jlink. Simplicity Commander refuses to do much except gives you the device info but when it comes to programming it does not liike the converted Jlink (it requires an official Jlink hardware AFAIK).

After trying for hours with different software I managed to flash the bootloader with pyocd and a cheap stlink clone from aliexpress.

I was getting an error message with pyocd but then found online someone that said he had luck with pyocd 0.34.3 and indeed it worked. I used a pip virtual environment 

The commands are simple: (The hex files are the ones you uploaded here).

````
$ pip install "pyocd==0.34.3"
$ pyocd pack install EFR32MG21A010F512IM3
$ pyocd cmd --target EFR32MG21A010F512IM32

Connected to EFR32MG21A010F512IM32 [Halted]: 3826041C8118303030303032
pyocd> load btl.hex
[==================================================] 100%
pyocd> load zigbee.hex
[==================================================] 100%
pyocd> 

`````

Hopefully that'll help

grobasoz, 2025-01-05:
OK - I think I finally understand what you require, more a case of me not reading your notes carefully or just thinking everyone wants NCP firmware for a new device to run Zigbee2MQTT?

Router firmware can certainly be created and the Z3Light code you have already can be used as a "router", you just need a means to enter CLI commands to connect the device to your (current Zigbee) network.

The beauty of slc-cli is that you can create the necessary project and firmware yourself.

grobasoz, 2025-01-08:
Can I close this or do you need some more help?

kar200, 2025-01-08:
Of course this can be closed and thanks again for all the help. I only realized yesterday that the latest commit dates from 2 years ago which makes me appreciate even more the help you provdied.

Enjoy your evening.

K.

closed 2025-01-08 by kar200