grobasoz/zigbee-firmware #54: [Help] Can't flash firmware for EM3585

firedevel, 2025-01-12:
I obtained an EM3585 module from a smart outlet, and I tried to flash the NCP firmware into it to use it as a Zigbee coordinator. I shorted `nBOOTMODE` to `GND`, and it seems to boot into the Bootloader, but there is no output. When I press Ctrl+C, the UART displays `ERR 18` and starts looping the output CCCC. I tried sending `BTL_STD_EM3585.s37` `NCP_USW_EM3585-LR_678-115k2.s37` and `NCP_USW_EM3585-LR_678-115k2.ebl` using xmodem, but both times it booted back into the original firmware.

grobasoz, 2025-01-13:
You should only be required to flash the NCP code and not the bootloader as the bootloader seemed to start by the nBOOTMODE option?
Just a point to note. The bootloader uses 115200 baud, N, 8, 1 so you would need to use a serial terminal to attach to the serial port with those settings.
I generally use TeraTerm and make sure there is a bootloader prompt when initialised with the nBOOTMODE button.
Eg
[image]
The do a **file->transfer** using the XModem command.

grobasoz, 2025-03-01:
Closed due to inaction.

firedevel, 2025-04-05:
[image]
Seems it's not Gecko Bootloader, only `B:` is output  when nBOOTMODE shorted to GND

grobasoz, 2025-04-05:
Set flow control to 'none'.
Let me know how it goes 👍

firedevel, 2025-04-06:
Where is flow control?
I’ve noticed this [using-legacy-application-bootloader.pdf](https://wwwqa.silabs.com/content/dam/siliconlabs/documents/public/application-notes/an772-using-legacy-application-bootloader.pdf), there is not Gecko Bootloader on EM35x.  

But I don't know why it still goes back to the original firmware after following the instructions to upload ebl.
Perhaps the chip is locked; I should try flashing it via SWD, but I don’t have a programmer.

grobasoz, 2025-04-06:
OK - this is getting tricky... it could be that the bootloader is "secured" by the factory? If that's the case, then a re-flash* is the only option.
Let's assume it is not? You'll need to determine what the module/chip supports in terms of PA/LNA (Power amplifier etc).
If a module, send me the number. If a chip, look for something (chip ~6pins) on the antenna circuit.
The "B:" and going straight into XModem transfer mode is odd - possibly their own bootloader (in which case - re-flash*)?
"Handshake" on the terminal should be "None" but I see in my picture it's set and there didn't seem to be an issue.

Did you use TeraTerm (or Similar) XModem->Send?

[image]

I just tested here and works fine :)

XModem Transfer (Ter Term)

[image]

Coms Port Settings (Tera Term), Bootloading (and running)

[image]

re-flash* - Fortunately on the EM358x the standard JLink's work...

firedevel, 2026-08-15:
[image]  

JLink method worked fine! Thanks for your time!

closed 2026-08-15 by firedevel